fwupd/fwupd #2412: Firmware Update from lvfs-testing not working in Red Hat 8.3 Edge (Beta)

dnoliver, 2020-09-22:
**Describe the bug**
A clear and concise description of what the bug is.

This is more like a heads up than a bug report, because Red Hat 8.3 Edge is kind of under development.

I have 2 CompuLab Fitlet2 devices, fitlet-2 and fitlet-3.
fitlet-2 is running Fedora IoT 32
fitlet-3 is running Red Hat 8.3. Edge (Beta), created following the recipe in https://github.com/osbuild/rhel-for-edge-demo
They are both running the same BIOS version. 
CompuLab is providing firmware updates in lvfs-testing https://fwupd.org/lvfs/devices/com.compulab.flt2.sec.firmware

Made a test to run in both devices:

```shell
#!/bin/bash

set -euxo pipefail

sudo fwupdmgr --version
sudo fwupdmgr enable-remote lvfs-testing --assume-yes
sudo fwupdmgr refresh --force
sudo fwupdmgr get-devices
sudo fwupdmgr get-updates
```

Firmware Update through lvfs-testing works in fitlet-2:

```shell
[test@fitlet-2 ~]$ ./test.sh 
+ sudo fwupdmgr --version
client version: 1.4.5
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.5
+ sudo fwupdmgr enable-remote lvfs-testing --assume-yes
╔══════════════════════════════════════════════════════════════════════════════╗
║ The LVFS is a free service that operates as an independent legal entity and  ║
║ has no connection with Fedora. Your distributor may not have verified any    ║
║ of the firmware updates for compatibility with your system or connected      ║
║ devices. All firmware is provided only by the original equipment             ║
║ manufacturer.                                                                ║
║                                                                              ║
║ This remote contains firmware which is not embargoed, but is still being     ║
║ tested by the hardware vendor. You should ensure you have a way to manually  ║
║ downgrade the firmware if the firmware update fails.                         ║
║                                                                              ║
║ Enabling this functionality is done at your own risk, which means you have   ║
║ to contact your original equipment manufacturer regarding any problems       ║
║ caused by these updates. Only problems with the update process itself        ║
║ should be filed at https://bugzilla.redhat.com/.                             ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Successfully enabled remote
+ sudo fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [***************************************]
Downloading…             [***************************************]
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 1 local device supported
+ sudo fwupdmgr get-devices
fitlet2
│
├─M2SCF-6M 32GB:
│     Device ID:           c46944fe84f2b057692258cc87a812cc86364881
│     Summary:             ATA Drive
│     Current version:     P0810C
│     Serial Number:       20190402017401000493
[... 215 lines cut ...]
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Needs a reboot after installation
?                          ? Cryptographic hash verification is available
?                          ? Device is usable for the duration of the update
?   
??Unifying Receiver:
      Device ID:           f3051add1ff6b15831703b30f1782eff9b2173ee
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        ? Updatable
                           ? Supported on remote server
```

**Additional questions**
- Operating system and version: Red Hat 8.3 Edge
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2020-09-23:
Version 4638 is in stable, not testing, but I think this bug still stands.

In the stable metadata I see only requirement for the 49ff2877-8596-4460-9a97-e70d90609e25 device:

    <requires>
      <id compare="ge" version="0.8.0">org.freedesktop.fwupd</id>
    </requires>

Which of course is fine for RHEL 8.anything. Can you grab me the RHEL 8.3 log from `sudo /usr/libexec/fwupd/fwupd --verbose` please; hopefully that will tell us why the device is not being marked as supported. Thanks.

dnoliver, 2020-09-23:
```shell
[test@fitlet-3 ~]$ sudo /usr/libexec/fwupd/fwupd --verbose
15:48:42:0181 FuDebug              Verbose debugging enabled (on console 1)
15:48:42:0185 FuConfig             loading config values from /etc/fwupd/daemon.conf
15:48:42:0186 FuRemoteList         loading /usr/share/fwupd/metainfo
15:48:42:0189 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:48:42:0190 XbSilo               file: ffc7fb1e-3930-54d4-ed66-dd9a3b857afc, current:ffc7fb1e-3930-54d4-ed66-dd9a3b857afc, cached: (null)
15:48:42:0190 XbSilo               loading silo with file contents
15:48:42:0190 Fwupd                looking for os-release at /etc/os-release
15:48:42:0191 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
15:48:42:0192 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
15:48:42:0193 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
15:48:42:0194 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
15:48:42:0195 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
15:48:42:0197 FuRemoteList         ignoring unfound remote fwupd
15:48:42:0197 FuRemoteList         ordering lvfs-testing=lvfs+1
15:48:42:0197 FuRemoteList         ignoring unfound remote fwupd
15:48:42:0197 FuRemoteList         ignoring unfound remote fwupd
15:48:42:0197 FuRemoteList         ignoring unfound remote fwupd
15:48:42:0197 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
15:48:42:0200 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:48:42:0201 Jcat                 loaded 1 certificates
15:48:42:0201 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:48:42:0201 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
15:48:42:0201 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:48:42:0201 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
15:48:42:0205 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:48:42:0205 Jcat                 loaded 1 certificates
15:48:42:0205 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:48:42:0205 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:48:42:0206 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
15:48:42:0208 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:48:42:0228 FuEngine             client certificate exists and working
15:48:42:0229 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:48:42:0230 FuHistory            got schema version of 5
15:48:42:0230 FuIdle               setting timeout to 7200s
15:48:42:0231 FuHwids              smbios property Manufacturer=Compulab
15:48:42:0231 FuHwids              smbios property EnclosureKind=3
15:48:42:0231 FuHwids              smbios property Family=N/A
15:48:42:0231 FuHwids              smbios property ProductName=fitlet2
15:48:42:0231 FuHwids              smbios property ProductSku=N/A
15:48:42:0231 FuHwids              smbios property BiosVendor=Compulab
15:48:42:0231 FuHwids              smbios property BiosVersion=FLT2.46.37
15:48:42:0231 FuHwids              smbios property BiosMajorRelease=05
15:48:42:0231 FuHwids              smbios property BiosMinorRelease=0c
15:48:42:0231 FuHwids              smbios property BaseboardManufacturer=Compulab
15:48:42:0231 FuHwids              smbios property BaseboardProduct=fitlet2
15:48:42:0253 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
15:48:42:0253 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:48:42:0254 XbSilo               file: 529a9cea-3311-1dde-072c-71a283c658f0, current:529a9cea-3311-1dde-072c-71a283c658f0, cached: (null)
15:48:42:0254 XbSilo               loading silo with file contents
15:48:42:0255 FuEngine             remote lvfs not enabled, so skipping
15:48:42:0256 FuEngine             remote vendor not enabled, so skipping
15:48:42:0256 FuEngine             remote vendor-directory not enabled, so skipping
15:48:42:0256 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
15:48:42:0257 XbSilo               1.46ms      parse blob
15:48:42:0257 XbSilo               1.68ms      loaded file
15:48:42:0257 XbSilo               file: 2dffba5d-f8d2-01cd-07e2-bd97496d44d9, current:2dffba5d-f8d2-01cd-07e2-bd97496d44d9, cached: (null)
15:48:42:0257 XbSilo               loading silo with file contents
15:48:42:0258 XbSilo               0.01ms      parse blob
15:48:42:0258 XbSilo               0.15ms      query on / with `components/component` limit=0 -> 128 results
15:48:42:0258 FuEngine             128 components now in silo
15:48:42:0258 XbSilo               0.15ms      query on / with `components/component/provides/firmware` limit=0 -> 151 results
15:48:42:0258 XbSilo               0.07ms      query on / with `components/component/provides/firmware` limit=0 -> 151 results
15:48:42:0269 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_uefi.so
15:48:42:0295 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_uefi.so
15:48:42:0295 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_modem_manager.so
[... cut ...]
15:48:43:0457 XbSilo               0.03ms      query on component with `custom/value[@key='LVFS::UpdateProtocol']` limit=1 -> 1 results
15:48:43:0457 FuEngine             ignoring RQR12.05_B0028 as older than RQR12.10_B0032
15:48:43:0457 FuCommon             creating path /var/cache/fwupd/motd.d
15:48:43:0465 FuMain               Started with locale en_US.UTF-8
15:48:43:0466 FuMain               Daemon ready for requests
15:48:43:0469 FuMain               FuMain: acquired name: org.freedesktop.fwupd
```

hughsie, 2020-09-23:
Ohh, hang on, did you also enable the `lvfs` repo as well as `lvfs-testing`? On RHEL the former is disabled by default.

dnoliver, 2020-09-23:
:) after enabling the `lvfs` repo, I can see the update:

```shell
[test@fitlet-3 ~]$ sudo fwupdmgr enable-remote lvfs
????????????????????????????????????????????????????????????????????????????????
? The LVFS is a free service that operates as an independent legal entity and  ?
? has no connection with Red Hat Enterprise Linux. Your distributor may not    ?
? have verified any of the firmware updates for compatibility with your        ?
? system or connected devices. All firmware is provided only by the original   ?
? equipment manufacturer.                                                      ?
?                                                                              ?
? Enabling this functionality is done at your own risk, which means you have   ?
? to contact your original equipment manufacturer regarding any problems       ?
? caused by these updates. Only problems with the update process itself        ?
? should be filed at https://bugzilla.redhat.com/.                             ?
?                                                                              ?
????????????????????????????????????????????????????????????????????????????????

Agree and enable the remote? [Y|n]: Y
Successfully enabled remote

[test@fitlet-3 ~]$ fwupdmgr refresh 
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.jcat
Downloading?             [***************************************]
Fetching metadata https://cdn.fwupd.org/downloads/firmware-00591-testing.xml.gz
Downloading?             [***************************************] Less than one minute remaining?
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
Downloading?             [***************************************]
Fetching metadata https://cdn.fwupd.org/downloads/firmware-00940-stable.xml.gz
Downloading?             [***************************************]
Successfully downloaded new metadata: 2 local devices supported

[test@fitlet-3 ~]$ fwupdmgr get-devices
fitlet2
?
??M2SCF-6M 32GB:
?     Device ID:           c46944fe84f2b057692258cc87a812cc86364881
?     Summary:             ATA Drive
?     Current version:     Q0707A
?     Update Error:        No vendor ID set
?     GUIDs:               f8d9d223-2eeb-50a6-89a8-60dbac4cf1fc
?                          323dcfd1-859e-5b6a-936a-30fa898f397d
?                          70e2a776-511f-5c70-86e5-ce2c78ba8387
?     Device Flags:        ? Internal device
?                          ? Requires AC power
?                          ? Needs a reboot after installation
?                          ? Device is usable for the duration of the update
?   
??System Firmware:
?     Device ID:           11d8ed06243abd8103ab13e360847338b4149b1b
?     Current version:     4637
?     Minimum Version:     4637
?     Vendor:              Compulab (DMI:Compulab)
?     GUID:                49ff2877-8596-4460-9a97-e70d90609e25
?     Device Flags:        ? Internal device
?                          ? Updatable
?                          ? Requires AC power
?                          ? Supported on remote server
?                          ? Needs a reboot after installation
?                          ? Cryptographic hash verification is available
?                          ? Device is usable for the duration of the update
?   
??Unifying Receiver:
      Device ID:           f3051add1ff6b15831703b30f1782eff9b2173ee
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        ? Updatable
                           ? Supported on remote server
    
[test@fitlet-3 ~]$ fwupdmgr get-updates
? Unifying Receiver has the latest available firmware version
fitlet2
?
??System Firmware:
  ?   Device ID:           11d8ed06243abd8103ab13e360847338b4149b1b
  ?   Current version:     4637
  ?   Minimum Version:     4637
  ?   Vendor:              Compulab (DMI:Compulab)
  ?   GUID:                49ff2877-8596-4460-9a97-e70d90609e25
  ?   Device Flags:        ? Internal device
  ?                        ? Updatable
  ?                        ? Requires AC power
  ?                        ? Supported on remote server
  ?                        ? Needs a reboot after installation
  ?                        ? Cryptographic hash verification is available
  ?                        ? Device is usable for the duration of the update
  ? 
  ??flt2 System Update:
        New version:       4638
        Remote ID:         lvfs
        Summary:           Firmware for fitlet2 lvfs firmware update testing
        License:           Proprietary
        Size:              16.8 MB
        Created:           2020-06-28
        Urgency:           High
        Vendor:            Compulab
        Flags:             is-upgrade
        Description:       Test release
        
         ? Add Spi Flash Puya p25q128l support
         ? Update Intel GBE UEFI driver to 9.2.06
```

hughsie, 2020-09-23:
> after enabling the lvfs repo, I can see the update

Okay to close this issue? Feel free to open a downstream (Red Hat) bug about the LVFS not being enabled by default; it's politics.

dnoliver, 2020-09-23:
Yes, please close!

Thank you

closed 2020-09-23 by hughsie