fwupd/fwupd #2422: Dell WD15 Dock is not listed

BjoernDaase, 2020-09-28:
**Describe the bug**
`fwupdmgr get-devices` does not list my Dell WD 15 Dock.
This is not https://github.com/fwupd/fwupd/issues/2396, as I do this on a Dell laptop and not https://github.com/fwupd/fwupd/issues/1207 which was reported against 1.2.10.

**Steps to Reproduce**

**Expected behavior**
`fwupdmgr get-devices` lists my Dell WD 15 Dock.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.6
compile-time dependency versions
	gusb:	0.3.5
	efivar:	37
daemon version:	1.4.6
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed via `dnf` on F33.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
XPS 13 9370
│
├─Intel AMT [unprovisioned]:
│     Device ID:          e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:            Hardware and firmware technology for remote out-of-band management
│     Current version:    11.8.79.3722
│     Bootloader Version: 11.8.79.3722
│     Vendor:             Intel Corporation
│     GUID:               2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:       • Internal device
│   
├─Intel(R) Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0xd6
│     Vendor:             GenuineIntel
│     GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:       • Internal device
│   
├─PM981 NVMe Samsung 512GB:
│     Device ID:          f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Summary:            NVM Express Solid State Drive
│     Current version:    EXA73D1Q
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         ad4ad443-6f4d-5735-ba86-17a1583a2403
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:          8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
│     Current version:    1.13.1
│     Minimum Version:    1.13.1
│     Vendor:             Dell Inc. (DMI:Dell Inc.)
│     GUID:               7ceaf7a8-0611-4480-9e30-64d8de420c7c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│   
├─TPM 2.0:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:            Platform TPM device
│ │   Current version:    7.2.0.1
│ │   Vendor:             Dell Inc. (PCI:0x1028)
│ │   GUIDs:              cb8da68d-cd80-5f5b-8fef-038383adbb83
│ │                       ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       7d65b10b-bb24-552d-ade5-590b3b278188
│ │                       6f5ddd3a-8339-5b2a-b9a6-cf3b92f6c86d
│ │                       fe462d4a-e48f-5069-9172-47330fc5e838
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Requires AC power
│ │                       • Needs a reboot after installation
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│     
├─Touchpad:
│     Device ID:          88861d85ec873ffdf484ac0b981c640781bd6937
│     Current version:    1.3.2375007
│     Bootloader Version: 54.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              f4384034-9243-5334-8075-a534be913e46
│                         424bd00e-9789-5cdf-a12a-3c81bc4676d6
│                         140f4458-951b-5bb9-85e2-879bd5b02615
│                         b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─UEFI dbx:
│     Device ID:          362301da643102b9f38477387e2193e57abaa590
│     Summary:            UEFI Revocation Database
│     Current version:    77
│     Minimum Version:    77
│     Vendor:             UEFI:Linux Foundation
│     Install Duration:   1 second
│     GUIDs:              00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│   
├─UHD Graphics 620:
│     Device ID:          bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              fed2efa4-7045-55a1-b4fc-b29283d59fe5
│                         8b72a10c-1279-5f8e-a28a-34fb11a58240
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─Unifying Receiver:
      Device ID:          f2a4275eddaca5782ce76f18b227b11cb965658e
      Summary:            A miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                          279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:       • Updatable
```

**Additional questions**
- Operating system and version:
F33
- Have you tried rebooting?
Yes
- Is this a regression?
I think that I have seen it being listed in the past.

superm1, 2020-09-28:
Can you please share

sudo fwupdtool get-devices --verbose

So we can see where problems occurred?

BjoernDaase, 2020-09-28:
Sure

```
12:27:19:0099 FuDebug              Verbose debugging enabled (on console 1)
12:27:19:0110 FuConfig             loading config values from /etc/fwupd/daemon.conf
12:27:19:0111 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
12:27:19:0111 FuRemoteList         loading /usr/share/fwupd/metainfo
12:27:19:0111 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:27:19:0111 XbSilo               file: 0974c74a-abdd-8e29-535c-beee406e055d, current:0974c74a-abdd-8e29-535c-beee406e055d, cached: (null)
12:27:19:0111 XbSilo               loading silo with file contents
12:27:19:0111 Fwupd                looking for os-release at /etc/os-release
12:27:19:0111 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.rpmsave
12:27:19:0111 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
12:27:19:0112 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
12:27:19:0112 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.rpmsave
12:27:19:0112 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
12:27:19:0112 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
12:27:19:0112 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.rpmnew
12:27:19:0112 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.rpmnew
12:27:19:0112 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
12:27:19:0112 FuRemoteList         ignoring unfound remote fwupd
12:27:19:0112 FuRemoteList         ordering lvfs-testing=lvfs+1
12:27:19:0112 FuRemoteList         ignoring unfound remote fwupd
12:27:19:0112 FuRemoteList         ignoring unfound remote fwupd
12:27:19:0112 FuRemoteList         ignoring unfound remote fwupd
12:27:19:0112 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
12:27:19:0112 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:27:19:0112 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
12:27:19:0112 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:27:19:0112 Jcat                 loaded 1 certificates
12:27:19:0112 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:27:19:0112 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:27:19:0112 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
12:27:19:0112 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:27:19:0112 Jcat                 loaded 1 certificates
12:27:19:0112 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:27:19:0112 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
12:27:19:0112 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:27:19:0118 FuEngine             client certificate exists and working
12:27:19:0118 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:27:19:0118 FuHistory            got schema version of 6
12:27:19:0118 FuHwids              smbios property Manufacturer=Dell Inc.
12:27:19:0118 FuHwids              smbios property EnclosureKind=a
12:27:19:0118 FuHwids              smbios property Family=XPS
12:27:19:0118 FuHwids              smbios property ProductName=XPS 13 9370
12:27:19:0118 FuHwids              smbios property ProductSku=07E6
12:27:19:0118 FuHwids              smbios property BiosVendor=Dell Inc.
12:27:19:0118 FuHwids              smbios property BiosVersion=1.13.1
12:27:19:0118 FuHwids              smbios property BiosMajorRelease=01
12:27:19:0118 FuHwids              smbios property BiosMinorRelease=0d
12:27:19:0118 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
12:27:19:0118 FuHwids              smbios property BaseboardProduct=0W970W
12:27:19:0121 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
12:27:19:0121 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
12:27:19:0121 XbSilo               file: f52701db-fa89-38c7-0c79-74291aa7ad3d, current:f52701db-fa89-38c7-0c79-74291aa7ad3d, cached: (null)
12:27:19:0121 XbSilo               loading silo with file contents
12:27:19:0121 FuEngine             remote lvfs-testing not enabled, so skipping
12:27:19:0121 FuEngine             remote vendor not enabled, so skipping
12:27:19:0121 FuEngine             remote vendor-directory not enabled, so skipping
12:27:19:0121 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
12:27:19:0121 XbSilo               0,01ms      parse blob
12:27:19:0121 XbSilo               0,03ms      loaded file
12:27:19:0121 XbSilo               file: ee0b375a-758f-6a58-c7f8-300a9ff9c1bd, current:ee0b375a-758f-6a58-c7f8-300a9ff9c1bd, cached: (null)
12:27:19:0121 XbSilo               loading silo with file contents
12:27:19:0121 XbSilo               0,00ms      parse blob
12:27:19:0122 XbSilo               0,15ms      query on / with `components/component` limit=0 -> 474 results
12:27:19:0122 FuEngine             474 components now in silo
12:27:19:0122 XbSilo               0,14ms      query on / with `components/component/provides/firmware` limit=0 -> 562 results
[... cut ...]
12:27:21:0676 FuPlugin             performing destroy() on flashrom
12:27:21:0676 FuPlugin             performing destroy() on synaptics_mst
12:27:21:0676 FuPlugin             performing destroy() on modem_manager
12:27:21:0677 FuPlugin             performing destroy() on upower
12:27:21:0677 FuPlugin             performing destroy() on dell
12:27:21:0677 FuPlugin             performing destroy() on redfish
12:27:21:0677 FuPlugin             performing destroy() on uefi
12:27:21:0677 FuPlugin             performing destroy() on logind
```

superm1, 2020-09-28:
>12:27:21:0245 FuPluginSynapticsMST no device found on drm_dp_aux1: invalid chip ID
12:27:21:0246 FuPlugin             performing udev_device_added() on synaptics_mst

Invalid chip ID?  Does this dock work properly?  Do you have a monitor plugged in?

superm1, 2020-09-28:
Double checked on a known good dock on my side and it seems to still be working on 1_4_X and master.

BjoernDaase, 2020-09-28:
Yeah, the dock works perfectly fine.

It has an external monitor, two Logitech receiver, an external HDD and a pair of speakers connected and everything works flawlessly.

superm1, 2020-09-28:
Do you by change have windows set up as a dual boot on the same system?  It would be interesting to know if Windows can probe it properly (I would expect problems there).

BjoernDaase, 2020-09-28:
Nope, (unfortunately) no Windows in this household.

superm1, 2020-09-28:
Oh okay.  Is there anything functionally wrong with the dock that is prompting trying to do an update?  If not, then I'd say there isn't much more we can do here to look into it.

BjoernDaase, 2020-09-28:
No, the dock is alright, I was just surprised to not see it being listed.

Nevertheless, that it reports an invalid ID does not make me happy. Do you know which component is responsible for this? And what the implications of this might be?

superm1, 2020-09-28:
The synaptics mst plugin is what reads the chip ID.  If it's not reading the right ID is why I suspected some sort of firmware issue.  That ID field is how you know which firmware to load onto it.

BjoernDaase, 2020-09-29:
So we are probably having a firmware issue preventing a firmware update? That sounds quite bad...

BjoernDaase, 2020-09-29:
FWIW, https://github.com/fwupd/fwupd/issues/1832 has exactly the same `invalid chip id` entry in its logs.

BjoernDaase, 2020-10-31:
So any idea how we can push that forward?

BjoernDaase, 2020-11-05:
@superm1 Alright, now things get interesting.

I use this dock with a Dell XPS 9370. This model has 3 USB-C ports, two on the left and one on the right. Let's call the ports l2, l3 and r3 according to https://topics-cdn.dell.com/pdf/xps-13-9370-laptop_setup-guide_en-us.pdf. 
The docking station works perfectly fine, no matter which port I connect it to. But it only appears in the output of `fwupd` if it is connect to port l3. If the dock is connected to either l2 or r3, fwupd does not list the devices.

Having it connected to port l3 then gives the following output
```
XPS 13 9370
│
├─Dell WD15:
│ │   Device ID:          ed9027c86c336ba5d3312f83c06f78b2ea6ef915
│ │   Summary:            A USB type-C docking station
│ │   Current version:    0.0.0
│ │   Vendor:             Dell Inc. (PCI:0x1028)
│ │   GUID:               e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:        29bca9f02e33e23330ae801ae7d6c22eb2b758e0
│ │     Summary:          A USB type-C docking station
│ │     Current version:  0.3.18
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ └─Dell WD15 Port Controller 1:
│       Device ID:        27c32815e5fed8f507c0210195bca394b635780f
│       Summary:          A USB type-C docking station
│       Current version:  1.1.8
│       Vendor:           Dell Inc. (PCI:0x1028)
│       GUID:             8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│       Device Flags:     • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│     
├─PM981 NVMe Samsung 512GB:
│     Device ID:          f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Summary:            NVM Express Solid State Drive
│     Current version:    EXA73D1Q
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02
│                         c9d531ea-ee7d-5562-8def-c64d0d144813
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         ad4ad443-6f4d-5735-ba86-17a1583a2403
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
│ │   Current version:    1.13.1
│ │   Minimum Version:    1.13.1
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              7ceaf7a8-0611-4480-9e30-64d8de420c7c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       43ea5588-d9a4-5031-8ad3-308045302d6b
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─Touchpad:
│     Device ID:          88861d85ec873ffdf484ac0b981c640781bd6937
│     Current version:    1.3.2375007
│     Bootloader Version: 54.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              f4384034-9243-5334-8075-a534be913e46
│                         424bd00e-9789-5cdf-a12a-3c81bc4676d6
│                         140f4458-951b-5bb9-85e2-879bd5b02615
│                         b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─Unifying Receiver:
      Device ID:          e3fca09e6a59f114838461a62c3885945f3a5280
      Summary:            A miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                          279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:       • Updatable
```

Does not somehow help?

superm1, 2020-11-06:
I notice your output is still missing the MST version.  
The lack of the MST version makes me suspect that DP aux channel isn't working properly.

I would lean on it's some hardware issue.  The communication with the dock to get FW versions of EC and port controllers happens over a specific pair of type C lines only used for that purpose.  If it's not showing up on certain ports I would suspect either damaged pins, interference on the lines or a damaged trace.

closed 2020-12-13 by stale[bot]

okias, 2021-01-28:
Still valid. Seems that Dell doesn't support WD15 update outside WIndows updater utility. Sadly I'm afraid that 

Version I have has some unpleasant issues, where after Dell XPS 9360 boot I'm often forced to reconnect WD15.
```
Current version:  1.1.8
```

floe, 2021-08-18:
Resurrecting this issue, because it's the exact same thing that happens with my XPS13 + WD15 dock on Ubuntu 20.04, fwupd 1.5.11:

```
XPS 13 9300
│
├─Dell WD15:
│ │   Device ID:          3086ec033384094e82dff9652509f61e12f66da7
│ │   Summary:            A USB type-C docking station
│ │   Current version:    0.0.0
│ │   Vendor:             Dell Inc. (PCI:0x1028)
│ │   GUID:               e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:        09821649fd4c49c71adcaadc9f779db402b46cb2
│ │     Summary:          A USB type-C docking station
│ │     Current version:  0.3.18
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ └─Dell WD15 Port Controller 1:
│       Device ID:        5da6f3f1074a5fd9e04d426b49b301310515e34f
│       Summary:          A USB type-C docking station
│       Current version:  1.1.8
│       Vendor:           Dell Inc. (PCI:0x1028)
│       GUID:             8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│       Device Flags:     • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│     
├─Integrated Webcam HD:
│     Device ID:          4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Current version:    91.58
│     Vendor:             CN0PW36V8LG0099B00DPX02 (USB:0x0BDA)
│     GUIDs:              bba32a58-5e20-5d64-bc15-287159bdaf46
│                         14e94ffe-f85a-5645-9af7-705464a66c8c
│     Device Flags:       • Updatable
│   
├─SSDPEMKF010T8 NVMe INTEL 1024GB:
│     Device ID:          ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:            NVM Express Solid State Drive
│     Current version:    7002
│     Vendor:             Intel Corporation (NVME:0x8086)
│     GUIDs:              f91a5c60-8696-539e-9d0a-57d194f74ac4
│                         396fb4fd-7656-564f-88b5-381e83ca30c0
│                         79517f86-8df8-5d6e-a18b-33f0b36a78e9
│                         68db11e5-b0cf-5bc9-a94e-17e28496e505
│                         8d8244c8-f391-5ef2-83d4-b28b954e4987
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─System Firmware:
  │   Device ID:          f2f44dcc012cdad82b823131f5c85ebfb74d782e
  │   Current version:    1.6.3
  │   Minimum Version:    1.6.3
  │   Vendor:             Dell Inc. (DMI:Dell Inc.)
  │   GUIDs:              e8292593-e66e-4878-b051-f152535ab130
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │                       29a2266e-9abb-57c9-b89e-58f2093f917b
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI Revocation Database
        Current version:  238
        Minimum Version:  238
        Vendor:           UEFI:Linux Foundation
        Install Duration: 1 second
        GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
                          4a6cd2cb-8741-5257-9d1f-89a275dacca7
                          c6682ade-b5ec-57c4-b687-676351208742
                          f8ba2887-9411-5c36-9cee-88995bb39731
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
```

The `synaptics_mst` plugin is installed and enabled. The dock only shows up as an updatable device when plugged into the left USB-C port, not on the right one (even though the dock itself works on both ports). IMHO this makes it less likely to be an issue related to hardware damage?

bdaase, 2021-08-18:
@floe I would like to reopen this issue then but since the bot closed it I seem to have lost power :/

floe, 2021-08-19:
Nevermind. Another interesting update, though: after suspending the laptop overnight with the dock connected, the VMM3332 suddenly _does_ show up in `fwupdmgr get-devices`:

```
XPS 13 9300
│
├─Dell WD15:
│ │   Device ID:          3086ec033384094e82dff9652509f61e12f66da7
│ │   Summary:            A USB type-C docking station
│ │   Current version:    0.0.0
│ │   Vendor:             Dell Inc. (PCI:0x1028)
│ │   GUID:               e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:        09821649fd4c49c71adcaadc9f779db402b46cb2
│ │     Summary:          A USB type-C docking station
│ │     Current version:  0.3.18
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ ├─Dell WD15 Port Controller 1:
│ │     Device ID:        5da6f3f1074a5fd9e04d426b49b301310515e34f
│ │     Summary:          A USB type-C docking station
│ │     Current version:  1.1.8
│ │     Vendor:           Dell Inc. (PCI:0x1028)
│ │     GUID:             8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│ │     Device Flags:     • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ └─VMM3332 inside Dell WD15/TB16/TB18 wired Dock:
│       Device ID:        d762543f8c20f636e6fff031a000078d3e10c600
│       Summary:          Multi-Stream Transport Device
│       Current version:  3.10.02
│       Vendor:           Synaptics (DRM_DP_AUX_DEV:0x06CB)
│       GUIDs:            87f8cd71-1e6b-5622-94e9-0e9d11bdde33
│                         c1e81f96-234e-597d-b9cb-6477d0652868
│                         78c73924-821a-5e6f-af82-86df6f67b182
│                         c1e2bea5-4cf2-5f84-b965-533e8e5fea45
│                         02b77ccb-03ab-521f-b98e-ef05395e9fa2
│                         82460515-30d3-5cdf-be4c-571340842a10
│                         a6f00ca7-54b7-5d51-9d6e-543d8df85376
│                         d6072785-6fc0-5f83-9d49-11376e7f48b1
│                         55d7e858-0ce0-506f-afb3-e8f6d77bf506
│                         4e4e06b6-8217-5cf0-8eff-ce5e987d531a
│       Device Flags:     • Updatable
│     
[...]
```

I think this confirms that it's not a hardware issue. From what I can see on https://fwupd.org/lvfs/devices/com.dell.mst78c73924.firmware there is a newer version available (3.12.02), I'll give it a try and see if this fixes the occasional external display flicker I've been experiencing.

floe, 2021-08-19:
P.S. Update was successful, fingers crossed that it actually fixes something ;-)

superm1, 2021-08-19:
> Nevermind. Another interesting update, though: after suspending the laptop overnight with the dock connected, the VMM3332 suddenly _does_ show up in `fwupdmgr get-devices`:
>
> ```
> XPS 13 9300
> │
> ├─Dell [...]

I think it at least also shows it's not a fwupd issue too.

lkraav, 2024-02-26:
1.9.6: weirdly, on this XPS 7390 2-in-1, I'm not able to get WD15 to show up connected either to left or right USB-C port :man_shrugging: 

It's also not in `lspci` but there are a couple of Microchip items in `lsusb`.

```
θ62° [leho:~] $ sudo lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 041: ID 0424:5537 Microchip Technology, Inc. (formerly SMSC) USB5537B
Bus 002 Device 042: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 036: ID 0424:2137 Microchip Technology, Inc. (formerly SMSC) USB2137B
Bus 003 Device 038: ID 0bda:4014 Realtek Semiconductor Corp. USB Audio
Bus 003 Device 039: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

[leho:~] 1 $ sudo lspci
00:00.0 Host bridge: Intel Corporation Ice Lake-LP Processor Host Bridge/DRAM Registers (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics G7 (rev 07)
00:04.0 Signal processing controller: Intel Corporation Processor Power and Thermal Controller (rev 03)
00:05.0 Multimedia controller: Intel Corporation Image Signal Processor (rev 03)
00:07.0 PCI bridge: Intel Corporation Ice Lake Thunderbolt 3 PCI Express Root Port #0 (rev 03)
00:07.2 PCI bridge: Intel Corporation Ice Lake Thunderbolt 3 PCI Express Root Port #2 (rev 03)
00:0d.0 USB controller: Intel Corporation Ice Lake Thunderbolt 3 USB Controller (rev 03)
00:0d.2 System peripheral: Intel Corporation Ice Lake Thunderbolt 3 NHI #0 (rev 03)
00:0d.3 System peripheral: Intel Corporation Ice Lake Thunderbolt 3 NHI #1 (rev 03)
00:12.0 Serial controller: Intel Corporation Ice Lake-LP Integrated Sensor Solution (rev 30)
00:14.0 USB controller: Intel Corporation Ice Lake-LP USB 3.1 xHCI Host Controller (rev 30)
00:14.2 RAM memory: Intel Corporation Ice Lake-LP DRAM Controller (rev 30)
00:14.3 Network controller: Intel Corporation Ice Lake-LP PCH CNVi WiFi (rev 30)
00:15.0 Serial bus controller: Intel Corporation Ice Lake-LP Serial IO I2C Controller #0 (rev 30)
00:15.1 Serial bus controller: Intel Corporation Ice Lake-LP Serial IO I2C Controller #1 (rev 30)
00:15.3 Serial bus controller: Intel Corporation Ice Lake-LP Serial IO I2C Controller #3 (rev 30)
00:16.0 Communication controller: Intel Corporation Ice Lake-LP Management Engine (rev 30)
00:1d.0 PCI bridge: Intel Corporation Ice Lake-LP PCI Express Root Port #9 (rev 30)
00:1d.7 PCI bridge: Intel Corporation Ice Lake-LP PCI Express Root Port #16 (rev 30)
00:1f.0 ISA bridge: Intel Corporation Ice Lake-LP LPC Controller (rev 30)
00:1f.3 Audio device: Intel Corporation Ice Lake-LP Smart Sound Technology Audio Controller (rev 30)
00:1f.4 SMBus: Intel Corporation Ice Lake-LP SMBus Controller (rev 30)
00:1f.5 Serial bus controller: Intel Corporation Ice Lake-LP SPI Controller (rev 30)
57:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD Controller BG4 (DRAM-less)
58:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
```

On surface, dock functions correctly.

On maybe unrelated note, WD19 dock shows up and updates fine :man_shrugging:

superm1, 2024-02-26:
We've taken out Wd15 support from fwupd now. No updates were ever posted or planned to be posted.