fwupd/fwupd #2429: fwupdmr does not work with pkttyagent

RPigott, 2020-09-29:
fwupdmgr does not appear to work with pkttyagent.

What I mean is for some reason when I don't have graphical polkit helper running, fwupdmgr will exit with "Failed to obtain auth".

This will demonstrate what I mean:

```
$ systemctl enable fwupd-refresh.timer 
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-unit-files ====
Authentication is required to manage system service or unit files.
Authenticating as: ronan
Password: 
==== AUTHENTICATION COMPLETE ====
Created symlink /etc/systemd/system/timers.target.wants/fwupd-refresh.timer → /usr/lib/systemd/system/fwupd-refresh.timer.

$ fwupdmgr get-updates 
[...]
Devices that have been updated successfully:
 • System Firmware (1.19.2 → 1.20.0)
[...]
Upload report now? (Requires internet connection):
[...]
Failed to obtain auth
```

In the above I enable the fwupd-refresh timer which demonstrates pkttyagent in action. Later I try to `fwupdmgr get-updates` which prompts me to consider uploading a firmware report, but when I try to upload one, the password prompt never appears and fwupdmgr fails to authorize. I am using fwupdmgr and systemctl from the same (local) session.

I would expect fwupdmgr to prompt for a password just like systemctl in this case. When I manually run a helper, like `polkit-gnome-authentication-agent-1` it works as expected. I haven't observed other software failing with only with pkttyagent, and based on my understanding of polkit I also don't really know how this is possible :/.

```
$ (. /etc/os-release; echo $NAME)
Arch Linux
$ pacman -Q fwupd polkit systemd polkit-gnome
fwupd 1.4.6-1
polkit 0.118-1
systemd 246.6-1
polkit-gnome 0.105-5
```

I have a little shell script to parse polkit policies for me, and it seems the allow_active policy directives are the same:

```
$ pklist org.freedesktop.fwupd.policy | grep -A4 modify-remote
org.freedesktop.fwupd.modify-remote
allow_any:      auth_admin
allow_inactive:	no
allow_active:   auth_admin_keep
$ pklist org.freedesktop.systemd1.policy | grep -A4 manage-unit-files
org.freedesktop.systemd1.manage-unit-files
allow_any:      auth_admin
allow_inactive: auth_admin
allow_active:   auth_admin_keep
```

superm1, 2020-09-29:
Hmm, I can reproduce the same thing.

One thing I noticed different is that systemd mentions more of it's actions in it's equivalent file for `/var/lib/polkit-1/localauthority/10-vendor.d/fwupd.pkla`.  Can you try to add the rest of the fwupd actions into that file and see if it helps?

RPigott, 2020-09-29:
Well atm I don't know any fwupdmgr actions that provoke a permission check, so I don't know how to reproduce until my firmware vendor releases a new successful update :/. Do you know of a command I can use for testing? Most of them seem to exit immediately when they realize there is nothing to do.

Also, Arch Linux uses polkit not PolicyKit so that won't be applicable, but I can investigate the rules definitions.

superm1, 2020-09-29:
You can use enable-remote lvfs-testing (and conversely disable-remote lvfs-testing)

RPigott, 2020-09-29:
Excellent. That works, thank you.

This is the rules file shipped with Arch matches the one in this repo. As a basic test I tried replacing the rule body with `return polkit.Result.YES` and the enable-remote works without prompt as expected.

I tried both removing and substituting `polkit.Result.AUTH_ADMIN_KEEP` and I get "Failed to obtain auth" in those cases.

I don't think it's related to the rules file, but I also don't know the behavior is different depending on the helper.

superm1, 2020-09-29:
It was entirely a shot in the dark by looking at behaviors. I guess the next thing we can do is check if there is any special code for systemd command or any other way to look for this tool in the session.

hughsie, 2020-09-29:
I actually think we probably want to do something like https://github.com/hughsie/PackageKit/commit/1437130d3ae127e09746436312f1d8240e8d029a although that's going to need some #ifdefs so everything still compiles for win32.

closed 2020-09-29 by hughsie