fwupd/fwupd #2440: HSI should only be shown for some chassis types

hughsie, 2020-10-02:
The different DMI chassis types are:

02h Unknown
03h Desktop
04h Low Profile Desktop
05h Pizza Box
06h Mini Tower
07h Tower
08h Portable
09h LapTop
0Ah Notebook
0Bh Hand Held
0Ch Docking Station
0Dh All in One
0Eh Sub Notebook
0Fh Space-saving
10h Lunch Box
11h Main Server Chassis
12h Expansion Chassis
13h SubChassis
14h Bus Expansion Chassis
15h Peripheral Chassis
16h RAID Chassis
17h Rack Mount Chassis
18h Sealed-case PC
19h Multi-system chassis
1Ah Compact PCI
1Bh Advanced TCA
1Ch Blade
1Dh Blade Enclosure

@superm1 suggested only: 3, 4, 6, 7, 8, 9, A, D should have HSI reported. I assume the others would say something like `HSI Security attributes are not supported for chassis type {}`

superm1, 2020-10-07:
(Since HSI won't be something advertised for 1.5.0, kicking this back to next release for milestone purposes)

superm1, 2020-10-08:
This isn't a complete list of chassis types, SMBIOS 3 added more. Here's the rest missing as of [SMBOIS 3.4](https://www.dmtf.org/sites/default/files/standards/documents/DSP0134_3.4.0.pdf) 

```
1Eh Tablet
1Fh Convertible
20h Detachable
21h IoT Gateway
22h Embedded PC
23h Mini PC
24h Stick PC
```

And I think all those other type are valid for HSI to check against.

closed 2020-10-18 by hughsie