fwupd/fwupd #2443: X1 yoga 2nd get-device does not list system firmware

patricius972, 2020-10-02:
**Describe the question**
Main board was changed and now , get-devices does list system firmware. 
I run arch up to 
[file.txt](https://github.com/fwupd/fwupd/files/5319896/file.txt)
date , with tpm2-tss, encrypt /root and uefi. UEFI update are enable in the bios. 

I also enable the debug mode, so the file is attached. 

Thanks.

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version
```
fwupdmgr --version
client version:	1.4.6
compile-time dependency versions
	gusb:	0.3.5
	efivar:	37
daemon version:	1.4.6

superm1, 2020-10-02:
The UEFI plugin is disabled because the ESP isn't getting found where you specified it.

```
Oct 02 13:56:39 X1Y fwupd[2632]: 18:56:39:0442 FuEngine             disabling plugin because: failed to startup using uefi: invalid OverrideESPMountPoint=/boot/efi specified in config: No ESP with path /boot/efi
```

patricius972, 2020-10-02:
it is mounted and have an EFI folder in it, 
/dev/nvme0n1p5       511M   11M  501M   3% /boot/efi

boot/efi/EFI:
total 12
drwxr-xr-x 2 root root 4096 Mar 29  2020  GRUB
drwxr-xr-x 3 root root 4096 Mar 29  2020  arch
drwxr-xr-x 2 root root 4096 Jan  8  2018 'arch bootloader'

/boot/efi/EFI/GRUB:
total 228
-rwxr-xr-x 1 root root 233472 Oct  2 08:05 grubx64.efi

/boot/efi/EFI/arch:
total 64
drwxr-xr-x 2 root root  4096 Mar 29  2020 fw
-rwxr-xr-x 1 root root 61114 Mar  4  2020 fwupdx64.efi

superm1, 2020-10-02:
It was mounted before or after the daemon was started?

patricius972, 2020-10-02:
before, it is in /fstab, it is mounted at boot, and I just restarted the daemon but still no output. efivars are mounted also.

superm1, 2020-10-02:
Your ESP, is it FAT32?  And if you just comment that line out, same behavior (it's supposed to auto-detect)?

patricius972, 2020-10-02:
yes it is fat32. ok commenting the uefi.conf works. it detected the devices, but still not sure what was the problem, the partition is mounted, location is RW.  

├─System Firmware:
│     Device ID:          2936081f97b1d24b29e7f535b3c82195efb6644c
│     Current version:    0.1.38
│     Minimum Version:    0.1.16
│     Vendor:             LENOVO (DMI:LENOVO)
│     GUID:               9ca89fa5-ce05-47ac-bdf3-5b0b6f21d07a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          53811f636139d09545526fd65f38eae6279852ef
│     Current version:    4653068
│     Minimum Version:    4653068
│     Vendor:             DMI:LENOVO
│     GUID:               22e1e2aa-29e7-4f6f-8bda-4980c14481b7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update

commit 79d34666 (2020-10-07): uefi: Correct a logic error with `FuVolume`

Fixes: #2443
commit 7812f31f (2020-10-07): uefi: Correct a logic error with `FuVolume`

Fixes: #2443
commit 5a835637 (2020-10-08): uefi: Correct a logic error with `FuVolume`

Fixes: #2443
commit 8a641cbd (2020-10-08): uefi: Correct a logic error with `FuVolume`

Fixes: #2443
closed 2020-10-08 by superm1