fwupd/fwupd #2444: System firmware result missing from get-devices output

ericonr, 2020-10-02:
**Describe the bug**

I'm updating from `fwupd` 1.4.2 to 1.4.6, and the output of `fwupdmgr get-devices --show-all-devices` is missing the `System Firmware` entry.

**Steps to Reproduce**

Run `fwupdmgr get-devices --show-all-devices` and check the output. Version 1.4.6 gives me:

```
XPS 15 9570
│
├─GP107M [GeForce GTX 1050 Ti Mobile]:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              dc1e6866-2188-547f-b175-35e07dd7a6af
│                         04232c57-d805-551d-a1de-1bba9350d71f
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel AMT [unprovisioned]:
│     Device ID:          e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:            Hardware and firmware technology for remote out-of-band management
│     Current version:    12.0.47.1524
│     Bootloader Version: 12.0.47.1524
│     Vendor:             Intel Corporation
│     GUID:               2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:       • Internal device
│   
├─Intel(R) Core™ i7-8750H CPU @ 2.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0xd6
│     Vendor:             GenuineIntel
│     GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:       • Internal device
│   
├─PC401 NVMe SK hynix 512GB:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    80007E00
│     Vendor:             SK hynix (NVME:0x1C5C)
│     GUIDs:              12c8974c-99cd-5f9a-b310-c28d565a47f8
│                         2260c743-963b-566e-93f1-8223c3e6b87b
│                         c28fdadd-607d-578c-b6ed-60ac49682904
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:    7.2.0.1
│ │   Vendor:             Nuvoton Technology (TPM:NTC)
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       fac1c8f3-73c8-5cd6-8330-07a3690b5140
│ │                       4521fefa-4a2f-537f-a7f9-d353841eb18e
│ │                       e9ccc1dc-960a-5e09-afe9-e59a904b776d
│ │                       c05f797a-8d97-547f-a7a3-c5c0339748e9
│ │   Device Flags:       • Internal device
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│     
├─Touchpad:
│     Device ID:          881d0bf9902910255d18c15a368c4106b26fee62
│     Current version:    1.7.2378871
│     Bootloader Version: 54.0.0
[... 128 lines cut ...]

**Expected behavior**

I expected to be shown the System Firmware entry.

**fwupd version information**

```
client version:	1.4.6
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.6
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source, because I'm updating the package for Void Linux.

**fwupd device information**

Provided above.

**Additional questions**
- Operating system and version: Linux 5.8.12, musl libc
- Have you tried rebooting? Not yet, will do.
- Is this a regression? Yes

superm1, 2020-10-02:
Are you manually configuring an ESP path?  If so, please try to disable that (see https://github.com/fwupd/fwupd/issues/2443)
Can you please share `fwupdtool get-devices --verbose` output to this bug?

ericonr, 2020-10-02:
My `uefi.conf` file had all the default values. Changing it to have `OverrideESPMountPoint=/boot/efi`, which I have verified is mounted, didn't fix it.

```
(fwupdmgr:6806): GLib-DEBUG: 19:31:40.202: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:6806): Fwupd-DEBUG: 19:31:40.205: looking for os-release at /etc/os-release
(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.206: GP107M [GeForce GTX 1050 Ti Mobile]
  DeviceId:             71b677ca0f1bc2c5b804fa1d59e52064ce589293
  Guid:                 dc1e6866-2188-547f-b175-35e07dd7a6af
  Guid:                 04232c57-d805-551d-a1de-1bba9350d71f
  Plugin:               optionrom
  Flags:                internal|registered|can-verify|can-verify-image
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Version:              a1
  VersionFormat:        plain
  Icon:                 audio-card
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.206: Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              12.0.47.1524
  VersionBootloader:    12.0.47.1524
  VersionFormat:        intel-me
  Icon:                 computer
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.206: Intel(R) Core™ i7-8750H CPU @ 2.20GHz
  DeviceId:             4bde70ba4e39b28f9eab1628f9dd6e6244c03027
  Guid:                 b9a2dd81-159e-5537-a7db-e7101d164d3f
  Plugin:               cpu
  Flags:                internal|registered
  Vendor:               GenuineIntel
  Version:              0xd6
  VersionFormat:        hex
  Icon:                 computer
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.206: PC401 NVMe SK hynix 512GB
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 12c8974c-99cd-5f9a-b310-c28d565a47f8
  Guid:                 2260c743-963b-566e-93f1-8223c3e6b87b
  Guid:                 c28fdadd-607d-578c-b6ed-60ac49682904
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Protocol:             org.nvmexpress
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  Vendor:               SK hynix
  VendorId:             NVME:0x1C5C
  Version:              80007E00
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.206: TPM
  DeviceId:             c6a80ac3a22083423992a3cb15018989f37834d6
  Guid:                 ff71992e-52f7-5eea-94ef-883e56e034c6
  Guid:                 fac1c8f3-73c8-5cd6-8330-07a3690b5140
  Guid:                 4521fefa-4a2f-537f-a7f9-d353841eb18e
  Guid:                 e9ccc1dc-960a-5e09-afe9-e59a904b776d
  Guid:                 c05f797a-8d97-547f-a7a3-c5c0339748e9
  Plugin:               tpm
  Flags:                internal|registered
  Vendor:               Nuvoton Technology
  VendorId:             TPM:NTC
  Version:              7.2.0.1
  VersionFormat:        quad
  VersionRaw:           0x0007000200000001
  Icon:                 computer
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.206: Event Log
  DeviceId:             58bd405f31c48e6eca290b425f530a94c91e955c
  ParentDeviceId:       c6a80ac3a22083423992a3cb15018989f37834d6
  Guid:                 a25657fe-b5dc-5be0-8b78-8b9dfec678ff
  Plugin:               tpm_eventlog
  Flags:                internal|registered
  Checksum:             SHA1(f25d837347e0eaaf069b2edce1869aa2f9875d21)
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.207: Touchpad
  DeviceId:             881d0bf9902910255d18c15a368c4106b26fee62
  Guid:                 1d6705d0-3fa6-58d2-bd0e-806ed6792c18
  Guid:                 2aa17264-7ef0-5cc5-9368-8595974a5e31
  Guid:                 741692a4-ae19-5ecc-9e48-e644604c2345
  Guid:                 a90f8ef5-de8d-5810-a6c2-d0b88b1e0be8
  Plugin:               synaptics_rmi
  Protocol:             com.synaptics.rmi
  Flags:                internal|updatable|registered
  Vendor:               Synaptics
  VendorId:             HIDRAW:0x06CB
  Version:              1.7.2378871
  VersionBootloader:    54.0.0
  VersionFormat:        triplet
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.207: UHD Graphics 630 (Mobile)
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 12c60b98-52cd-500a-b99e-087aa6b47f0b
  Guid:                 db677673-1102-5481-a11e-6f397e123ced
  Plugin:               optionrom
  Flags:                internal|registered|can-verify|can-verify-image
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2020-10-02

(fwupdmgr:6806): FuMain-DEBUG: 19:31:40.207: skipping unreported check
```

superm1, 2020-10-02:
I'm looking for specifically fwupdtool output, or if you can't get that verbose daemon output.

ericonr, 2020-10-02:
Sorry, I got confused!

```
23:53:04:0192 FuDebug              Verbose debugging enabled (on console 0)
23:53:04:0192 FuConfig             loading config values from /etc/fwupd/daemon.conf
23:53:04:0192 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
23:53:04:0193 FuRemoteList         loading /usr/share/fwupd/metainfo
23:53:04:0193 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
23:53:04:0193 XbSilo               file: bcf9ffb5-7114-4290-b7bf-3331072d9ff8, current:bcf9ffb5-7114-4290-b7bf-3331072d9ff8, cached: (null)
23:53:04:0193 XbSilo               loading silo with file contents
23:53:04:0193 Fwupd                looking for os-release at /etc/os-release
23:53:04:0193 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
23:53:04:0193 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
23:53:04:0193 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
23:53:04:0193 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
23:53:04:0194 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
23:53:04:0194 FuRemoteList         ignoring unfound remote fwupd
23:53:04:0194 FuRemoteList         ordering lvfs-testing=lvfs+1
23:53:04:0194 FuRemoteList         ignoring unfound remote fwupd
23:53:04:0194 FuRemoteList         ignoring unfound remote fwupd
23:53:04:0194 FuRemoteList         ignoring unfound remote fwupd
23:53:04:0194 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
23:53:04:0194 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
23:53:04:0194 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
23:53:04:0194 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
23:53:04:0194 Jcat                 loaded 1 certificates
23:53:04:0194 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
23:53:04:0194 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
23:53:04:0194 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
23:53:04:0194 Jcat                 loaded 1 certificates
23:53:04:0194 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
23:53:04:0194 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
23:53:04:0194 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
23:53:04:0194 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
23:53:04:0199 FuEngine             client certificate exists and working
23:53:04:0199 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
23:53:04:0200 FuHistory            got schema version of 6
23:53:04:0200 FuHwids              smbios property Manufacturer=Dell Inc.
23:53:04:0200 FuHwids              smbios property EnclosureKind=a
23:53:04:0200 FuHwids              smbios property Family=XPS
23:53:04:0200 FuHwids              smbios property ProductName=XPS 15 9570
23:53:04:0200 FuHwids              smbios property ProductSku=087C
23:53:04:0200 FuHwids              smbios property BiosVendor=Dell Inc.
23:53:04:0200 FuHwids              smbios property BiosVersion=1.15.0
23:53:04:0200 FuHwids              smbios property BiosMajorRelease=01
23:53:04:0200 FuHwids              smbios property BiosMinorRelease=0f
23:53:04:0200 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
23:53:04:0200 FuHwids              smbios property BaseboardProduct=0D0T05
23:53:04:0203 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
23:53:04:0204 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
23:53:04:0204 XbSilo               file: 5291f479-2d3a-6509-8725-02656b9beab1, current:5291f479-2d3a-6509-8725-02656b9beab1, cached: (null)
23:53:04:0204 XbSilo               loading silo with file contents
23:53:04:0204 FuEngine             remote lvfs-testing not enabled, so skipping
23:53:04:0204 FuEngine             remote vendor not enabled, so skipping
23:53:04:0204 FuEngine             remote vendor-directory not enabled, so skipping
23:53:04:0204 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
23:53:04:0204 XbSilo               0.01ms      parse blob
23:53:04:0204 XbSilo               0.03ms      loaded file
23:53:04:0204 XbSilo               file: 225445b9-5726-216d-3999-41fab1626c5c, current:225445b9-5726-216d-3999-41fab1626c5c, cached: (null)
23:53:04:0204 XbSilo               loading silo with file contents
23:53:04:0204 XbSilo               0.01ms      parse blob
23:53:04:0204 XbSilo               0.16ms      query on / with `components/component` limit=0 -> 474 results
23:53:04:0204 FuEngine             474 components now in silo
23:53:04:0204 XbSilo               0.14ms      query on / with `components/component/provides/firmware` limit=0 -> 562 results
23:53:04:0205 XbSilo               0.16ms      query on / with `components/component/provides/firmware` limit=0 -> 562 results
23:53:04:0207 FuEngine             adding plugin /usr/lib/fwupd-plugins-3/libfu_plugin_rts54hub.so
23:53:04:0207 FuPlugin             performing init() on /usr/lib/fwupd-plugins-3/libfu_plugin_rts54hub.so
[... cut ...]
  VersionBootloader:    54.0.0
  VersionFormat:        triplet
  Created:              2020-10-02

23:53:05:0104 FuPlugin             performing destroy() on tpm_eventlog
23:53:05:0104 FuPlugin             performing destroy() on synaptics_mst
23:53:05:0105 FuPlugin             performing destroy() on upower
23:53:05:0105 FuPlugin             performing destroy() on dell
23:53:05:0105 FuPlugin             performing destroy() on redfish
23:53:05:0105 FuPlugin             performing destroy() on uefi
```

superm1, 2020-10-03:
Ah you're missing udisks2. Install that and it should work.

ericonr, 2020-10-03:
Indeed, thanks!

I see it listed in https://github.com/fwupd/fwupd/wiki/Compilation#runtime-dependencies , at least. Do you think there's anywhere else it could go?

superm1, 2020-10-03:
It became a "harder" dependency fairly recently for the UEFI plugin as part of this commit (in 1_4_X: https://github.com/fwupd/fwupd/commit/25ba41579fb170cf3a29cb2753733172a54984cd).
I think that we probably should message more clearly that's the problem, such as putting a big warning about that that into the log by default rather than just disabling the plugin and keeping it in debugging output.

superm1, 2020-10-03:
And regarding distro; yes debian/ubuntu added a Recommends:
https://github.com/fwupd/fwupd/commit/eb4472a47949875a4d68f02997d32ab1e487ae54

ericonr, 2020-10-03:
> I think that we probably should message more clearly that's the problem, such as putting a big warning about that that into the log by default rather than just disabling the plugin and keeping it in debugging output.

I think that would be nice! I will probably put udisks2 as a dependency on Void, so most people shouldn't even get the warning, but it should help others.

commit c7d93c9f (2020-10-07): uefi: Make udisks2 errors more apparent
When support for dynamically mounting disks was added for https://github.com/fwupd/fwupd/commit/25ba41579fb170cf3a29cb2753733172a54984cd
udisks2 became a harder dependency and it was less obvious to users.

Create devices but show an error in why devices aren't updatable if
it's not found.

Users can still configure ESP manually in `uefi.conf`

Fixes: #2444
PR #2455 (2020-10-08): uefi: Make udisks2 errors more apparent
When support for dynamically mounting disks was added for https://github.com/fwupd/fwupd/commit/25ba41579fb170cf3a29cb2753733172a54984cd
udisks2 became a harder dependency and it was less obvious to users.

Create devices but show an error in why devices aren't updatable if
it's not found.

Users can still configure ESP manually in `uefi.conf`

Fixes: #2444

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit e8a2fc8c (2020-10-08): uefi: Make udisks2 errors more apparent
When support for dynamically mounting disks was added for https://github.com/fwupd/fwupd/commit/25ba41579fb170cf3a29cb2753733172a54984cd
udisks2 became a harder dependency and it was less obvious to users.

Create devices but show an error in why devices aren't updatable if
it's not found.

Users can still configure ESP manually in `uefi.conf`

Fixes: #2444
commit 278ad73e (2020-10-08): uefi: Make udisks2 errors more apparent
When support for dynamically mounting disks was added for https://github.com/fwupd/fwupd/commit/25ba41579fb170cf3a29cb2753733172a54984cd
udisks2 became a harder dependency and it was less obvious to users.

Create devices but show an error in why devices aren't updatable if
it's not found.

Users can still configure ESP manually in `uefi.conf`

Fixes: #2444
closed 2020-10-08 by superm1