fwupd/fwupd #2450: fwupd 1.5.0: Build failure on chromeos/chromiumos due to tss2 dependency

bleungatchromium, 2020-10-06:
**Describe the bug**
A clear and concise description of what the bug is.
Recent versions of fwupd from 1.5.0 branch (master as of this moment) do not build properly in the chromiumos environment (built in a gentoo like environment).

**Steps to Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
Yes, this is a regression. I've done a git bisect of recent commits, and this one came up:
110eb286bcef515ec8b5c4ea80792cf8084cd1bb is the first bad commit
commit 110eb286bcef515ec8b5c4ea80792cf8084cd1bb
Author: Mario Limonciello <mario.limonciello@dell.com>
Date:   Tue Sep 8 12:58:38 2020 -0500

    Make TPM more optional (Fixes: #2360)  

    - Rename the `plugin_tpm` option to `tpm` and when it's disabled
    remove TPM support from all plugins
    - If enabled then require the TSS to be installed

 contrib/ci/build_windows.sh |  2 +-
 contrib/ci/trust.sh         |  2 +-
 contrib/fwupd.spec.in       |  4 ++--
 meson.build                 | 11 +++++------
 meson_options.txt           |  2 +-
 plugins/meson.build         |  4 ++--
 plugins/uefi/fu-self-test.c |  5 +++++
 7 files changed, 17 insertions(+), 13 deletions(-)

Meson log of the failure attached.
[meson-log.txt](https://github.com/fwupd/fwupd/files/5337236/meson-log.txt)

bleungatchromium, 2020-10-06:
Oh i see what happened. I think this just needs an ebuild change on our side since -Dplugin_tpm="false" isn't recognized anymore. i'll take care of it.

superm1, 2020-10-06:
> Oh i see what happened. I think this just needs an ebuild change on our side since -Dplugin_tpm="false" isn't recognized anymore. i'll take care of it.

Yeah and not picked up yet in Gentoo since 1.5.0 isn't out yet.

superm1, 2020-10-06:
(`/me` plugs #2141)

closed 2020-10-06 by superm1