fwupd/fwupd #2454: cros-ec: Read Only/Read Write update with rw_sig checking support (plus honeybuns platform)

bleungatchromium, 2020-10-07:
**Is your feature request related to a problem? Please describe.**

Current cros-ec plugin supports RO and RW update flow for devices that do not check the integrity of the RW partition (a process called rw_sig verification).

Servo Micro falls in that category, since it is a debug dongle, not a production device. In practice the lack of rw_sig check means the jump from RO to RW is not automatic. When it boots, it sits in RO, until something (ie, our updater) tells it to switch to RW.

rw_sig devices behave differently. They will automatically jump from RO to RW upon verifying the RW partition's signature. 
The honeybuns reference (gingerbread and quiche) will rw_sig.

**Describe the solution you'd like**
Our current flow always boots into RO, then updates RW, then updates RO upon jumping to RW.

Since rw_sig devices will start from RW, and since important update-flow updates in RO may be required, update RO first  (this assumes that RO is not actually read-only, as is the current state of these prototypes), then jump to RO.

Once in RO, we need to carefully send the stay-in-ro message, which wasn't used properly in prior patch series. 
The stay-in-ro needs to be sent to cancel the RW_SIG and automatic jump to RW, and needs to happen right before the write phase.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

superm1, 2020-10-07:
@bleungatchromium how soon are you thinking this will go?  We're discussing on a few other PR's about cutting 1.5.0 soon pending a few other changes, so if you want this in it would be good to have that in back of our minds.

bleungatchromium, 2020-10-07:
I have a series of development patches that need to be cleaned up a bunch (lots of debug needs to be removed) but I could have it posted today or tomorrow.

superm1, 2020-10-07:
OK sounds good, let's target to include this for 1.5.0 then.

PR #2459 (2020-10-11): cros-ec: RO/RW update with rw_sig support
Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [X] Feature
- [ ] Documentation

fixes #2454
closed 2020-10-11 by hughsie