fwupd/fwupd #2458: What's the best way of adding an `efi-objcopy` build option?

ericonr, 2020-10-08:
**Describe the question**

As can be seen in https://github.com/fwupd/fwupd/blob/master/meson_options.txt , one can define `efi-cc` and `efi-ld`, which simplify the workarounds needed for cross compiling fwupd. However, the script in https://github.com/fwupd/fwupd/blob/master/plugins/uefi/efi/generate_binary.sh uses `objcopy` directly, which then requires manual patching for cross compilation.

Would it be possible instead to have this as another build option, like `efi-objcopy`? If so, what's the best way of propagating this information to the script?

hughsie, 2020-10-08:
> to have this as another build option, like efi-objcopy

I guess; I'm intrigued what you're cross compiling onto :)

> the best way of propagating this information to the script

You'd use a `configure_file()` in meson and then replace something like `@EFI_OBJCOPY@`

ericonr, 2020-10-08:
> I guess; I'm intrigued what you're cross compiling onto :)

I'm fixing up the package for Void Linux, we use cross compilation for ARM targets. I don't know if any ARM devices have EFI firmware that's updateable (and has been uploaded to LVFS as well), but there's no reason the option shouldn't be available there :)

> You'd use a configure_file() in meson and then replace something like @EFI_OBJCOPY@

I see, thanks! Will try to make a PR for it, then.

PR #2464 (2020-10-10): uefi: add configuration option for objcopy utility.
Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [x] Feature (?)
- [ ] Documentation

Closes #2458
closed 2020-10-10 by hughsie