fwupd/fwupd #2460: Failed to load SMBIOS warning when process does not have access to /sys/firmware/dmi/tables/smbios_entry_point

djcampello, 2020-10-09:
For sandboxed executions when the user running fwupd doe not have access to /sys/firmware/dmi/tables/smbios_entry_point a warning is printed: https://github.com/fwupd/fwupd/blob/6450d0deb446380befa28795c52f5f0b628c4595/src/fu-engine.c#L6081

Why do fwupdtool needs access to this? Can we remove this check if is not needed for proper execution? (This error does not seem to affect the use cases under test)

superm1, 2020-10-09:
The main thing it's used for is for Hardware ID matching, but it's also used for determining whether to run the coreboot, redfish, and uefi plugins.  

The hardware ID matching is used to limit updates to only be available on certain machines.  The metadata on LVFS can specify specific hardware ID's and the client will compare those requirements in the metadata to determine whether an update can be offered.  It makes most sense for internal devices when two physical devices can't be disambiguated by a local GUID.

For what ChromeOS currently uses fwupd for I don't see a reason to need SMBIOS data, but upstream we need leave the warning in place as those other features make sense in the general sense.

hughsie, 2020-10-09:
> For what ChromeOS currently uses fwupd for I don't see a reason to need SMBIOS data

It would be good to support what ChromeOS _does_ use, e.g. we now support Device Tree and SMBIOS and it would be pretty easy to add another way to query things like "which model is this". It's the sort of thing that might be really good to use in the future if there's ever the same USB VID/PID on different baseboard that needs different firmware, for instance.

djcampello, 2020-10-09:
To clarify, this file is generally available in Chrome OS. In our current use case for fwupd we only interact via fwupdtool and for each separate device there is a separate invocation. In order to reduce the risk surface each invocation is launched in a sandboxed environment with only the permissions required to carry the update by the given plugin. While looking at the plugins we have in use, I did not see any use for this file, so the sandbox environment does not have access to it and hence we get the warning message. More details on the sandboxing can be found here: crrev.com/c/2451443

superm1, 2020-10-09:
@djcampello something I was thinking about, are you going to know in advance all the plugins needed from a composite update in between invocations?  For example if you are updating a WD19TB on a system without USB4 you only need the dell_dock plugin.  But if you have usb4 on the system, you'll need the thunderbolt plugin too (and to allow access to all the attributes from it).

I think the same statement should apply to other composite devices too.

djcampello, 2020-10-09:
I plan to address that with crrev.com/c/2460994

hughsie, 2020-10-09:
> all the plugins needed from a composite update in between invocations

Can't ChromeOS just use the _fwupd_ daemon process? Setting it to an auto-quit delay of say 5 seconds... I think it's getting to the point where it's more work using fwupdtool than just doing what everyone else is doing.

djcampello, 2020-10-09:
This would mean that we would need to give broad permissions to the daemon so it can handle all types of updates.( i.e., nvme plugin requires CAP_SYS_ADMIN which is problematic and not necessary for all other plugins in use).

hughsie, 2020-10-09:
> This would mean that we would need to give broad permissions to the daemon

Well, I didn't know this was a concern. Is there a way for fwupd to drop or otherwise limit it's own capabilities depending on the update that's being scheduled? Or even at startup? I'll admit it's frustrating trying to second-guess the reasons that you can't just use the daemon...

djcampello, 2020-10-09:
How would we know ahead of time which update is going to be performed by the daemon? In the current architecture, we use udev to fire fwupdtool for the device. Since the execution is device based we know ahead of time which plugins to use and which permissions to give to the tool. In the daemon case, the daemon would be running expecting any device, right?

hughsie, 2020-10-09:
> In the daemon case, the daemon would be running expecting any device, right?

Yes, the daemon would start up with all capabilities. My proposal would be an extra mode to the daemon that just quits immediately after completing the update -- which allows plugins to drop the capabilities they don't need before starting the update.  

Back a bit; I'm also trying to work out what problem you're trying to fix by dropping capabilities before the update? Are you worried about a malicious .cab file triggering a parsing error in one of the FuFirmware subclassed helpers allowing something like stack smashing? Or one of the FuDevice subclasses doing something insane and allowing the hardware to do actions on the host?

superm1, 2020-10-13:
>  I think it's getting to the point where it's more work using fwupdtool than just doing what everyone else is doing.

There is one notable thing that would need to be considered in that ChromeOS doesn't use PK.  So if all concerns were allayed with the daemon approach we'll still need a compile time option to not use PK in ChromeOS.

>This would mean that we would need to give broad permissions to the daemon so it can handle all types of updates.( i.e., nvme plugin requires CAP_SYS_ADMIN which is problematic and not necessary for all other plugins in use).

It may or may not help, but I think that you can configure on a system by system basis what should be allowed too.  For example some systems won't have NVME disks and so you can just disable NVME plugin on those systems entirely.

hughsie, 2020-10-13:
> we'll still need a compile time option to not use PK in ChromeOS

I think I volunteered to do this yesterday with my monthly sync with Google.

> you can configure on a system by system basis what should be allowed too

We talked briefly about using the existing daemon, but do something like a per-system block of specific plugins. Then the daemon can look at the list of plugins and drop capabilities if none of the enabled ones are requesting CAP_ADMIN for instance.

superm1, 2020-10-13:
>I think I volunteered to do this yesterday with my monthly sync with Google.

So they may have already mentioned this but, if not @djcampello started on it here:
https://chromium-review.googlesource.com/c/chromiumos/third_party/fwupd/+/2354521

Assuming that comes back onto master cleanly adding a meson option to disable PK may work.

> We talked briefly about using the existing daemon, but do something like a per-system block of specific plugins. Then the daemon can look at the list of plugins and drop capabilities if none of the enabled ones are requesting CAP_ADMIN for instance.

Plugins need to then have a handshake to indicate what they actually need that means.

superm1, 2020-10-13:
I'm going to close this thread - the original issue is "expected behavior".  The remaining things that stand out are split off here:

1. Capabilities hand shake: https://github.com/fwupd/fwupd/issues/2471
2. Daemon without PK: hthttps://github.com/fwupd/fwupd/pull/2470