fwupd/fwupd #2463: fwupd.shutdown failed with exit status 1

marek-lach, 2020-10-09:
**Describe the bug**
I am not actually sure what that _status 1_ is, but it seems to be somehow related to power management maybe?

Perhaps someone else can enlighten the issue, though my wifi always gets disconnected on its own, and then I see this error on reboot.

hughsie, 2020-10-09:
What distro? Did you just try to do an ATA firmware update or something?

superm1, 2020-10-09:
I've noticed this lately too with master and hand compilation, but haven't investigated it at all.

marek-lach, 2020-10-09:
> What distro?

Arch in my case.

bulletmark, 2020-10-12:
I get this every shutdown on Arch as well, using `fwupd` 1.4.6-1. I see the last line of `/usr/lib/systemd/system-shutdown/fwupd.shutdown` is `/usr/bin/fwupdtool activate` which is presumably what is returning the exit code. If I run that manually I get:
```
lt:~ sudo fwupdtool activate
Loading…                 [  -                                    ]00:31:18:0312 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0: attribute vendor_name returned no data
00:31:18:0328 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:0d.3/domain1/1-0: attribute vendor_name returned no data
Loading…                 [***************************************]
Activating firmware update Thunderbolt controller in Dell dock…
Device 5e8cd9f97ee07a0af3a5e0c70fffb713aa32362e does not need activation
```

superm1, 2020-10-12:
Any chance you would be able to upgrade to master and see if it's persisting?
This fix went into master: https://github.com/fwupd/fwupd/commit/02085a03719bb603fcb20b7832d14f9399110a65 and I expect should help your described behavior.

Notably we didn't backport it to 1_4_X, I haven't looked to see how complex of a backport it is.

bulletmark, 2020-10-12:
Arch always packages the latest release which is currently 1.4.6. Arch updates their packages within hours/days of a new release. When is the next release due?

bulletmark, 2020-10-12:
I installed current git master and I get the same error message on shutdown except "with error status 2" instead of 1.

commit 9d817e8b (2020-10-12): trivial: fix shutdown error on fwupdtool activate

EXIT_NOTHING_TO_DO needs to be caught by the shutdown script too.
Fixes: #2463
PR #2468 (2020-10-12): trivial: fix shutdown error on fwupdtool activate
EXIT_NOTHING_TO_DO needs to be caught by the shutdown script too.
Fixes: #2463

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 02d9d23f (2020-10-12): trivial: fix shutdown error on fwupdtool activate

EXIT_NOTHING_TO_DO needs to be caught by the shutdown script too.
Fixes: #2463
closed 2020-10-12 by superm1