fwupd/fwupd #2473: Intel AMT missing device name in GUID

mendel5, 2020-10-13:
```
$ sudo fwupdmgr get-devices --show-all-devices
H97-D3H
│
├─Crucial CT275MX300SSD1:
│     Device ID:          f219ac034743df43be49a235e41f128dd47d855f
│     Summary:            ATA Drive
│     Current version:    M0CR070
│     Vendor:             Micron (ATA:0x1344)
│     Serial Number:      [redacted]
│     GUIDs:              2ba20be0-33a6-5e9a-b14a-3eb292a479e3 ← IDE\Crucial_CT275MX300SSD1__________________M0CR070
│                         2774a44c-b02a-5b6d-9b32-c6a749dc104c ← IDE\0Crucial_CT275MX300SSD1__________________
│                         a36974c8-92c2-5472-9e94-628fc111758f ← Crucial_CT275MX300SSD1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel AMT [unprovisioned]:
│     Device ID:          e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:            Hardware and firmware technology for remote out-of-band management
│     Current version:    9.1.25.1005
│     Bootloader Version: 9.1.25.1005
│     Vendor:             Intel Corporation
│     GUID:               2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:       • Internal device
│   
└─Intel(R) Xeon(R) CPU E3-1231 v3 @ 3.40GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x28
      Vendor:             GenuineIntel
      GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
      Device Flags:       • Internal device

```

I'm not sure if this is a bug or an exception that is supposed to be like that.

The device `Intel AMT [unprovisioned]` has a GUID but there is no arrow (`←`) and no device name.

**Expectation**
Similar to this device from above:
```
Intel(R) Xeon(R) CPU E3-1231 v3 @ 3.40GHz:
GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
```

I would expect the Intel AMT to look like this:
```
Intel AMT [unprovisioned]:
GUID:               2800f812-b7b4-2d4b-aca8-46e0ff65814c ← amt
```
However the arrow (←) and device name (e.g. "amt", could be anything else) are missing:
```
Intel AMT [unprovisioned]:
GUID:               2800f812-b7b4-2d4b-aca8-46e0ff65814c
```

superm1, 2020-10-14:
This is the expected behavior.  The GUID is hardcoded in the [code](https://github.com/fwupd/fwupd/blob/master/plugins/amt/fu-plugin-amt.c#L459).  The `README.md` also advertises the same: `These devices use the existing GUID provided by the AMT host interface.`

closed 2020-10-14 by superm1

mendel5, 2020-10-14:
> README.md

https://github.com/fwupd/fwupd/tree/master/plugins/amt

Thanks, got it!