fwupd/fwupd #2475: Dell WD19 detection issue

bryan5989, 2020-10-14:
My WD19TB has deviceId: a455a3a21752e5d31c228f1f5b51761dad88a432 and is correctly listed as supported, but no updates are detected. Dell support site indicates an update is available. Update from support site detects and applies correctly. Support site indicates that fwupdmgr since 1.2.5 should work in place of manual update. I have 1.2.10

```shell
bryan@bryan-1u:~$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
bryan@bryan-1u:~$ fwupdmgr get-devices 
Latitude 3530 Thunderbolt Controller
  DeviceId:             458ffe6d134d86d9544eef2d9ee946c556737a49
  Guid:                 ebdc0ada-9bd5-5020-bb26-d3d216e02e8d
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              41.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-10-14

Thunderbolt controller in Dell dock
  DeviceId:             d30c7550b3f2a8ee2d51837fa61ba2b59a4ad3d1
  ParentDeviceId:       a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 c94770ca-1773-592c-b20a-e87243bc7cd0
  Summary:              Thunderbolt controller
  Plugin:               thunderbolt
  Flags:                updatable|require-ac|supported|registered
  Vendor:               Dell Inc
  VendorId:             TBT:0x00D4
  Version:              43.00
  VersionFormat:        pair
  Icon:                 audio-card
  InstallDuration:      22
  Created:              2020-10-14

RTS5487 in Dell dock
  DeviceId:             acb404019656654d44f80922d94735e831d9bb40
  ParentDeviceId:       a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 707c63d2-e597-5c40-84db-9b1bb4c48d96
  Guid:                 acfcd89b-105d-55b9-b85b-08bf8508f38c
  Guid:                 568ffa1e-a0db-5287-9ea3-872b60f7730b
  Summary:              USB 3.1 Generation 2 Hub
  Plugin:               dell_dock
  Flags:                updatable|require-ac|supported|registered
  Vendor:               Dell Inc
  VendorId:             USB:0x413C
  Version:              01.47
  VersionFormat:        pair
  Icon:                 dock-usb
  InstallDuration:      3
  Created:              2020-10-14

WD19TB
  DeviceId:             a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 cd357cf1-40b2-5d87-b8df-bb2dd82774aa
  Summary:              High performance dock
  Plugin:               dell_dock
  Flags:                updatable|require-ac|supported|registered
  Vendor:               Dell Inc
  Version:              01.00.00.00
  VersionLowest:        01.00.00.00
  VersionFormat:        quad
  Icon:                 dock-usb
  InstallDuration:      80
  Created:              2020-10-14
  UpdateMessage:        Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.

Package level of Dell dock
  DeviceId:             1eca9eabb0c992c136e1deb1f89e3f70c465aa1c
  ParentDeviceId:       a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 8ceeeffd-51b6-580c-9b75-69143227aff8
  Summary:              A representation of dock update status
  Plugin:               dell_dock
  Flags:                updatable|supported|registered
  Vendor:               Dell Inc
  Version:              01.00.14.01
  VersionFormat:        quad
  InstallDuration:      5
  Created:              2020-10-14

VMM5331 in Dell dock
  DeviceId:             58931e8c9ef5eead9c007563814f96fcadb1b993
  ParentDeviceId:       a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 89fec0b6-6b76-5008-b82c-5e5c6c164007
  Summary:              Multi Stream Transport controller
  Plugin:               dell_dock
  Flags:                updatable|require-ac|supported|registered
  Vendor:               Dell Inc
  Version:              05.04.03
  VersionFormat:        triplet
  InstallDuration:      360
  Created:              2020-10-14

RTS5413 in Dell dock
  DeviceId:             ca36279eeac13463fd5d974c88794859294fbfb1
  ParentDeviceId:       a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4
  Guid:                 b27d25f1-019d-5718-b41a-02ddaefe5577
  Guid:                 ac5b774c-b49d-566b-9255-85f0f7f8a4ed
  Summary:              USB 3.1 Generation 1 Hub
  Plugin:               dell_dock
  Flags:                updatable|require-ac|supported|registered
  Vendor:               Dell Inc
  VendorId:             USB:0x413C
  Version:              01.21
  VersionFormat:        pair
  Icon:                 dock-usb
  InstallDuration:      14
  Created:              2020-10-14
```

[out.log](https://github.com/fwupd/fwupd/files/5379128/out.log)

PS I'm already updated in the logs

bryan5989, 2020-10-16:
Ah, I did not. I just downgraded and tried with lvfs-testing enabled, but unable to get catalog:

```shell
bryan@bryan-1u:~/source/repos/reportingmice$ fwupdmgr get-updates 
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…             [***************************************] Less than one minute remaining…
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Failed to update metadata for lvfs-testing: '48A6D80E4538BAC2' is not a valid signature
```

superm1, 2020-10-16:
mmm @hughsie, we have a problem on lvfs-testing again?

hughsie, 2020-10-16:
Well, it's the old problem of the CDN getting out of sync; JCat-enabled fwupd won't have this problem. To work around this anyone can do:

curl -X PURGE https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
curl -X PURGE https://cdn.fwupd.org/downloads/firmware-testing.xml.gz

no auth required!

bryan5989, 2020-10-30:
lvfs-testing worked after your post. Should maybe tell Dell to include this caveat in their update instructions? Or how does this firmware make it into non-testing?

superm1, 2020-10-30:
> Or how does this firmware make it into non-testing?

Dell promotes it to non-testing.  When that happens is up to that team.

closed 2020-10-30 by superm1