fwupd/fwupd #2476: FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES is not set if *only* an alternate branch exists.

meklort, 2020-10-15:
**Describe the bug**
#2420 added support for multiple branches, however with commit f64e7a9cb4b22b83b05246bd70de2eb90b22fd35, at least two different branches must be available from LVFS for branches to be switched to.

With the oss bcm5719 firmware on the Talos II, only one branch is available in lvfs. Even though this branch is "oss-firmware" instead of "default", only one branch is detected and as such FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES is not set.
Since FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES is not set, fwupdmgr does not allow the switch-branch command to be used when the proprietary firmware is installed to switch to the oss firmware.
Note: If both the propretary firmware and the oss-firmware versions were available via LVFS, this would likely pass. The only reason it's failing is since only one firmware variant exists.

**Steps to Reproduce**
1. Install the proprietary firmware on the NIC.
1. Install the lvfs embargo file for the bcm5719 firmware.
1. Run fwupdmgr switch-branch 7de5ffdca08fa52d95fd4bb42aa5d07a4b35d2dd oss-firmware

**Expected behavior**
Since the proprietary firmware is not the "oss-firmware" branch, the command should be accepted.

**fwupd version information**
fwupd build based on master.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
T2P9D01 REV 1.00
│
├─NetXtreme BCM5719 Gigabit Ethernet PCIe:
│     Device ID:          7de5ffdca08fa52d95fd4bb42aa5d07a4b35d2dd
│     Current version:    1.43
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              df899e77-b408-51b3-acd3-e91343344bfb
│                         5d8b12bf-1973-58fc-8b31-3e50fe31954d
│                         cd40a56d-7fbf-57ef-9502-50e9f35d9b1b
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device will backup firmware before installing
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          9e517c3cc54687578aa78274c9644219970a5b5b
│     Current version:    0.4.395
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              cf9a2e58-3d30-5351-b9a3-604d4b3d59c2
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         cd40a56d-7fbf-57ef-9502-50e9f35d9b1b
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device will backup firmware before installing
│   
├─82574L Gigabit Network Connection (Gigabit CT Desktop Adapter):
│     Device ID:          a3f04fb07c9012b396897a081ed0725655b6ba76
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              9ddc6d96-8f9e-51e5-b204-1719d4c8e449
│                         7f5a805d-2c1a-5efa-8cd0-5dd66ebe05e2
│                         6af770c4-de01-5039-8737-1055825b763e
│                         64593cc0-2efd-529d-9bf4-7084542c809d
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Ellesmere [Radeon Pro WX 5100]:
│     Device ID:          47eeafbeb390c1b3e9125143370b09a7e561dd91
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              a40d3841-2720-5889-8a38-59d4413ed178
│                         50cf27ba-0d58-5681-8b13-02631c5aec64
│                         b36f3e1f-f41a-5f49-b04a-250b5cca6cec
│                         881dbcb1-e9e0-53ef-8e07-29cdc1ac8e0e
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NVMe SSD Controller SM951/PM951:
│     Device ID:          7292ecd2edec3b738a86392d3e35087ae7b107b5
│     Current version:    01
│     Vendor:             Samsung Electronics Co Ltd (PCI:0x144D)
│     GUIDs:              81754dda-78ad-52c6-ab2a-a0a8547674af
│                         65c71c21-48e0-579e-9e58-bdb00d03361c
│                         f620cb9c-fedd-5a48-a91b-563c40890059
│                         f8775f26-c040-554b-b27c-bea293913427
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe:
│     Device ID:          48e3a2f898a9e9d83dd82699ca2c453b5f52fc36
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              df899e77-b408-51b3-acd3-e91343344bfb
│                         5d8b12bf-1973-58fc-8b31-3e50fe31954d
│                         cd40a56d-7fbf-57ef-9502-50e9f35d9b1b
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          21b4da64a3cf4e1e36d2dfb471b61f926b2e58a7
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              cf9a2e58-3d30-5351-b9a3-604d4b3d59c2
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         cd40a56d-7fbf-57ef-9502-50e9f35d9b1b
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          61d0e94ced5020fb62205ec9c346a3df76098ac7
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              cf9a2e58-3d30-5351-b9a3-604d4b3d59c2
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         cd40a56d-7fbf-57ef-9502-50e9f35d9b1b
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          7bd2060cbab727fa0015da547d5e890c4421fe17
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              cf9a2e58-3d30-5351-b9a3-604d4b3d59c2
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         cd40a56d-7fbf-57ef-9502-50e9f35d9b1b
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─System Firmware:
      Device ID:          e24dbdf2c162e9342bccb3d8114b2d2fbbd7a5ca
      Current version:    talos-v2.3-rc2-65-g0eaf5d3d
      Vendor:             IBM
      Update Error:       UEFI Capsule updates not available or enabled
      GUID:               230c8b18-8d9b-53ec-838b-6cfc0383493a
      Device Flags:       • Internal device
```

**Additional questions**
- Operating system and version: Fedora 32 ppc64le
- Have you tried rebooting? No
- Is this a regression? N/A

superm1, 2020-10-15:
As a suggestion for this instance (as well as any other future instances like it), what do you think about adding `FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES` to the device directly in the plugin?

meklort, 2020-10-15:
I can see a number of ways to fix this.
* Set `FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES` in the plugin as you mentioend. Note that switch-branch can now be called even when the firmware is already on the right branch. Granted, an error about this will be thrown.
* fu-engine.c could be updated to set `FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES` any time a non-default branch is detected, even if only one is present. This is similar to the first option, except it's automatic (I'd prefer this and can create a PR if you'd like)
* Alternative, fw-util.c could allow switch-branch any time the requested branch does not match the current branch, regardless of the `FWUPD_DEVICE_FLAG_HAS_MULTIPLE_BRANCHES` flag.

hughsie, 2020-10-15:
What about the proposal in https://github.com/fwupd/fwupd/pull/2479 -- that seems logically most correct; there exists a different branch to the one that is installed right now -- the logic being that the device already has a defined branch even if it's not available in the remote.

superm1, 2020-10-15:
that was merged, so closing.  re-open if any other concerns, thanks!

closed 2020-10-15 by superm1