fwupd/fwupd #2487: Inconsistent formatting in fwupdmgr help options

mendel5, 2020-10-17:
When I open the help options in fwupdmgr I get this output:
```
$ fwupdmgr -help
Usage:
  fwupdmgr [OPTION…]

  activate [DEVICE-ID|GUID]         Activate devices
  block-firmware [CHECKSUM]         Blocks a specific firmware from being installed.
  clear-history                     Erase all firmware update history
  clear-offline                     Clears any updates scheduled to be updated offline
  clear-results DEVICE-ID|GUID      Clears the results from the last update
  disable-remote REMOTE-ID          Disables a given remote
  downgrade [DEVICE-ID|GUID]        Downgrades the firmware on a device
  enable-remote REMOTE-ID           Enables a given remote
  get-approved-firmware             Gets the list of approved firmware.
  get-blocked-firmware              Gets the list of blocked firmware.
  get-details FILE                  Gets details about a firmware file
  get-devices                       Get all devices that support firmware updates
  get-history                       Show history of firmware updates
  get-releases [DEVICE-ID|GUID]     Gets the releases for a device
  get-remotes                       Gets the configured remotes
  get-results DEVICE-ID|GUID        Gets the results from the last update
  get-topology                      Alias to get-devices
  get-updates                       Gets the list of updates for connected hardware
  get-upgrades                      Alias to get-updates
  install FILE [DEVICE-ID|GUID]     Install a firmware file on this hardware
  modify-config KEY,VALUE           Modifies a daemon configuration value.
  modify-remote REMOTE-ID KEY VALUE
                                    Modifies a given remote
  refresh [FILE FILE_SIG REMOTE-ID]
                                    Refresh metadata from remote server
  reinstall [DEVICE-ID|GUID]        Reinstall current firmware on the device.
  report-history                    Share firmware history with the developers
  set-approved-firmware CHECKSUM1[,CHECKSUM2][,CHECKSUM3]
                                    Sets the list of approved firmware.
  unblock-firmware [CHECKSUM]       Blocks a specific firmware from being installed.
  unlock DEVICE-ID|GUID             Unlocks the device for firmware access
  update [DEVICE-ID|GUID]           Updates all firmware to latest versions available
  upgrade [DEVICE-ID|GUID]          Alias to update
  verify [DEVICE-ID|GUID]           Checks cryptographic hash matches firmware
  verify-update [DEVICE-ID|GUID]    Update the stored cryptographic hash with current ROM contents

Help Options:
  -h, --help                Show help options

Application Options:
  -v, --verbose             Show extra debugging information
  --version                 Show client and daemon versions
  --offline                 Schedule installation for next reboot when possible
  --allow-reinstall         Allow reinstalling existing firmware versions
  --allow-older             Allow downgrading firmware versions
  --force                   Override warnings and force the action
  -y, --assume-yes          Answer yes to all questions
  --sign                    Sign the uploaded data with the client certificate
  --no-unreported-check     Do not check for unreported history
  --no-metadata-check       Do not check for old metadata
  --no-reboot-check         Do not check for reboot after update
  --no-safety-check         Do not perform device safety checks
  --no-history              Do not write to the history database
  --show-all-devices        Show devices that are not updatable
  --disable-ssl-strict      Ignore SSL strict checks when downloading files
  --filter                  Filter with a set of device flags using a ~ prefix to exclude, e.g. 'internal,~needs-reboot'

This tool allows an administrator to query and control the fwupd daemon, allowing them to perform actions such as installing or downgrading firmware.

```
I noticed two instances of inconsistent formatting. One is about the use of square brackets and the other one about the use of full stops/ periods.

**Square brackets**
Example code from above:
```
  get-releases [DEVICE-ID|GUID]     Gets the releases for a device
  get-results DEVICE-ID|GUID        Gets the results from the last update

```
In the first line there are square brackets before and after the `DEVICE-ID|GUID` while they are missing in the second line. Maybe you can decide whether to consistently use square brackets or not and then we can update it.

**Full stops/ periods**
Example code from above:
```
  get-blocked-firmware              Gets the list of blocked firmware.
  get-details FILE                  Gets details about a firmware file
```
A full stop is being used at the end of the first line, however no full stop is being used at the end of the second line. Maybe you can decide whether to consistently use full stops or not and then we can update it.

**Further info**
```
$ sudo fwupdmgr --version
client version:	1.4.6
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.4.6
```
Install method is snap.

superm1, 2020-10-17:
I'm pretty sure the brackets mean it's "optional"

closed 2020-10-18 by hughsie

mendel5, 2020-10-20:
> I'm pretty sure the brackets mean it's "optional"

Ok