fwupd/fwupd #2489: Where 'set-address' option was gone?

udik-chudik, 2020-10-19:
Hi there!
How can I now set up target address in `fwupd 1.3.11` for DfuSe **.dfu** files?
Earlier, in version **fwupd 1.2.10** all was just fine: `dfu-tool set-address firmware.dfu 0x8008800`. But now a get: **Command not found** :confused:

hughsie, 2020-10-20:
I fear we might have removed it as nobody was using it (it seems incorrectly). We now have a much more powerful builder, if you're trying to assemble a dfuse file from a set of binaries? e.g. would this help: https://github.com/fwupd/fwupd/commit/41400a8cc65c1fef4aa5b72c0d6642220788a482

udik-chudik, 2020-10-21:
Oh no :cry: !
Is there any way to do this in `1.3.11`?

hughsie, 2020-10-21:
> But is there any way to do this in earlier version like 1.3.11

Unfortunately not. If you're using Fedora then there's a COPR with 1.5.0 pre-release which has all the functionality you'd need.

superm1, 2020-10-21:
If you're somewhere you can use snap, you can use the "edge" channel snap too to get the pre-release version.
Follow instructions here but add `--edge` when you install the snap.
https://github.com/fwupd/fwupd/wiki/fwupd-snap

udik-chudik, 2020-10-23:
Okay, thanks guys :+1:

closed 2020-10-23 by udik-chudik