fwupd/fwupd #2494: Auto detect Elan hid-i2c touchpad with VID 04f3 / PID 3XXX

jinglewu, 2020-10-21:
At the "plugins/elantp/elantp.quirk" file : 
[DeviceInstanceId=HIDRAW\VEN_04F3&DEV_314F]
Plugin = elantp
GType = FuElantpHidDevice

Is it possible to include all "DeviceInstanceId = HIDRAW\VEN_04F3&DEV_3XXX"?
-> Vid 04f3 and PID 3XXX

hughsie, 2020-10-21:
@superm1 do you have any good ideas here? I suppose we could support globbing (e.g. simple regex) but that's going to slow down lookups (although I've not profiled how much real CPU time that would be).

@jinglewu how many IDs are we talking about if you listed them individually? 65536 possible devices?

jinglewu, 2020-10-21:
@superm1 PID : 3XXX (3000~3FFF) : 4095 devices.

hughsie, 2020-10-21:
@superm1 I'm kinda erring on just adding an elantp specific extra GUID of `HIDRAW\VEN_04F3&DEV_xXXX` where `x` is 3 in this case

superm1, 2020-10-21:
I guess it's worth asking, does 4xxx or 5xxx use same protocol?

As is does it make sense for plugin to just run all elan VID. If not then I would agree work special guid makes sense to do.

jinglewu, 2020-10-21:
vid 04f3 for all elan device (touchpad, trackpoint, fingerprinter, etc..)
pid 3xxx for elan i2c-hid touchpad device.

superm1, 2020-10-21:
But what happens if i2c hid plugin probes on a non i2c hid device? Will it just be a noop?

superm1, 2020-10-21:
@hughsie another idea is to make it all elan vid but then at elan plugin device setup phase to check if it's actually 3xxx pid or not. Then all other devices won't have to get another guid added.

And if some day it turns out i2c hid touchpad protocol can share some code with other elan device the plugin can be responsible to look at those pid further.

hughsie, 2020-10-21:
> what happens if i2c hid plugin probes on a non i2c hid device

This is my worry. We've seen i2c devices do bad things just from probing, and a new GUID seems the simplest way of not regressing on some random other platform. The problem is that this patch isn't going to work; it's only going to create the extra GUID when the device is being created, when we'd need the FuUdevDevice to do that for us so that it would match the quirk with 3XXX and then use the plugin.

superm1, 2020-10-21:
But making a new GUID implies it's the same firmware for ALL 3XXX devices. That can't be right. What you really want is the VID to let elan plugin probe it, and in probe check lid is 3xxx.

Then in xml you use the actual pid of the device, not 3xxx.

jinglewu, 2020-10-21:
> But what happens if i2c hid plugin probes on a non i2c hid device? Will it just be a noop?

The non i2c-hid touchpad device is cannot probes on i2c-hid driver in Linux kernel.
Elantp plugin have two interface (hid and i2c-recovery).
So if the non i2c-hid touchpad device is probes on i2c-recovery plugin, then the touchpad cannot work on update finish until after reset the computer.

commit b3257503 (2020-10-21): elantp: Match all i2c-hid devices in quirk

In the plugin probe routine, check whether they are actually i2c-hid touchpad
by looking at PID.

Fixes #2494
commit e7540069 (2020-10-21): elantp: Match all i2c-hid devices in quirk

In the plugin probe routine, check whether they are actually i2c-hid touchpad
by looking at PID.

Fixes #2494
PR #2497 (2020-10-21): trivial: match all elan devices in quirk
Here's what I had in mind.  Untested:

In probe routine, check whether they are actually i2c-hid touchpad
by looking at pid.

fixes #2494

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 0ad124a0 (2020-10-21): elantp: Match all i2c-hid devices in quirk

In the plugin probe routine, check whether they are actually i2c-hid touchpad
by looking at PID.

Fixes #2494
closed 2020-10-21 by hughsie