fwupd/fwupd #2499: Fail to reboot into Ubuntu after successful update

motorbreth, 2020-10-21:
triggered the update with Gnome Software on Ubuntu 20.04 Gnome Software asked me to restart, I click restart. During boot, the update starts correctly, and succeeds. After it succeeds, the laptop boots back into fwupd, which fails with the following error:

start_image() returned Invalid Parameter

as a workaround I re added the ubuntu grubx64.efi file to the boot options tried to boot again got the error  
Operating System Loader signature not found in SecureBoot database ('db')

so i turned off secure boot in the bios

**Expected behavior**
the computer to boot to the OS

 fwupdmgr --version
client version: 1.3.11
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.11
``

Latitude E7470
│
├─INTEL SSDSCKKF256H6 SATA 256GB:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:             ATA Drive
│     Current version:     LBFD07N
│     GUIDs:               5384e083-d34c-5028-891f-2454fb503123
│                          f94c2cde-e158-5685-9902-08ec58cd7b35
│                          0d184e64-9412-5d27-9be8-5ba19e879ecb
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.79.3722
│     Bootloader Version:  11.8.79.3722
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│
├─Skylake GT2 [HD Graphics 520] (Latitude E7470):
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     07
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               2f7ef0d3-15f0-571c-b852-de72f1c7c730
│                          bdcb90a0-e7cd-5d77-995d-d79b7183aa29
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│
├─System Firmware:
│     Device ID:           870e59dce3f787904a11aec1bd724f466a573d7d
│     Current version:     1.25.3
│     Minimum Version:     1.25.3
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     GUID:                212026ee-fde4-4d08-ac41-c62cb4036a42
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│
└─Unifying Receiver:
      Device ID:           a04d2149eebdb3eda67a37a2575ec263a8389cef
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.10_B0036
      Bootloader Version:  BOT03.01_B0008
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server

________________________________________________

Devices that have been updated successfully:

 • Unifying Receiver (RQR24.05_B0029 → RQR24.10_B0036)
 • System Firmware (1.23.3 → 1.25.3)

BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0005,0006,0002,0001
Boot0000* Windows Boot Manager  HD(2,GPT,56e603ac-5278-411f-8c5d-313401aee95c,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...t...............
Boot0001* Linux Firmware Updater        HD(1,GPT,d73e493a-f8a7-4f17-aa8d-ea56cd8abb53,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0002* Ubuntu        PciRoot(0x0)/Pci(0x17,0x0)/Sata(2,65535,0)/HD(1,GPT,d73e493a-f8a7-4f17-aa8d-ea56cd8abb53,0x800,0x100000)/File(\EFI\ubuntu\grubx64.efi)
Boot0004* Linux Firmware Updater        HD(1,GPT,d73e493a-f8a7-4f17-aa8d-ea56cd8abb53,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0005  Onboard NIC(IPV4)     PciRoot(0x0)/Pci(0x1f,0x6)/MAC(18dbf235b54c,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot0006  Onboard NIC(IPV6)     PciRoot(0x0)/Pci(0x1f,0x6)/MAC(18dbf235b54c,0)/IPv6([::]:<->[::]:,0,0)..BO

```shell
$ efivar -l | grep fw

0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-212026ee-fde4-4d08-ac41-c62cb4036a42-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule

superm1, 2020-10-21:
It sounds like you don't have a signed fwupd installed.  Can you confirm you have `fwupd-signed` package or not?

motorbreth, 2020-10-21:
fwupd-signed: command not found

superm1, 2020-10-21:
`apt install fwupd-signed`

motorbreth, 2020-10-21:
Reading package lists... Done
Building dependency tree
Reading state information... Done
fwupd-signed is already the newest version (1.27.1ubuntu2+1.3.11-1~focal1).
fwupd-signed set to manually installed.
The following packages were automatically installed and are no longer required:
  binutils-common cpp-7 fakeroot freepats fwupdate gcc-7-base gnome-software-common gnome-user-guide gsfonts guile-2.0-libs ifupdown
  libappstream-glib8 libart-2.0-2 libasan5 libatomic1 libavutil55 libb-hooks-op-check-perl libcapture-tiny-perl libcdio17
  libclass-method-modifiers-perl libclass-xsaccessor-perl libclone-perl libcommon-sense-perl libcrypt-dev libdevel-size-perl
  libdouble-conversion1 libdvdread4 libdynaloader-functions-perl libevent-2.1-6 libexempi3 libexiv2-14 libexporter-tiny-perl libfakeroot
  libfcgi-perl libffi6:i386 libfile-copy-recursive-perl libfuture-perl libgnome-desktop-3-17 libgssdp-1.0-3 libgutenprint2 libhogweed4
  libilmbase12 libindicator3-7 libiptc0 libjson-c3 libjson-xs-perl liblist-moreutils-perl liblouis14 liblqr-1-0 liblsan0 libmicrodns0
  libmozjs-52-0 libmysofa0 libncurses5 libnet-domain-tld-perl libnettle6 libnumber-compare-perl libpackage-stash-xs-perl libperlio-gzip-perl
  libplacebo4 libpoppler73 libqpdf21 libreadline7 libreadonly-perl libref-util-perl libref-util-xs-perl libreoffice-style-galaxy
  libsereal-decoder-perl libstrictures-perl libstruct-dumb-perl libsub-quote-perl libswresample2 libtest-fatal-perl libtest-refcount-perl
  libtinfo5 libtsan0 libtype-tiny-perl libtype-tiny-xs-perl libtypes-serialiser-perl libubsan1 libunicode-utf8-perl libusbmuxd4 libvpx5
  libx264-152 libx265-146 libxml-sax-base-perl libyaml-libyaml-perl linux-headers-5.4.0-48 linux-headers-5.4.0-48-generic
  linux-image-5.4.0-48-generic linux-modules-5.4.0-48-generic linux-modules-extra-5.4.0-48-generic multiarch-support netcat-traditional
  patchutils php7.2-readline php7.2-sqlite3 pppconfig pppoeconf python-crypto python3-zope.interface qpdf xserver-xorg-core-hwe-18.04
  xserver-xorg-hwe-18.04 xserver-xorg-input-all-hwe-18.04 xserver-xorg-input-wacom-hwe-18.04 xserver-xorg-video-nouveau-hwe-18.04
  xserver-xorg-video-vesa-hwe-18.04
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

superm1, 2020-10-21:
Are you missing updated shim or something?

motorbreth, 2020-10-21:
I'm sorry but I don't know what shim is
I installed ubuntu 18.04 on this laptop , and I was updating it regularly , at some point ubuntu offer to upgrade to 20.04 so i did.
On the last update it did the bios update and this is what happened the error and no boot

superm1, 2020-10-21:
Are you fully up to date right now with Ubuntu's OS updates?

motorbreth, 2020-10-21:
this is the result of apt-get upgrade

Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

superm1, 2020-10-22:
>Boot0002* Ubuntu PciRoot(0x0)/Pci(0x17,0x0)/Sata(2,65535,0)/HD(1,GPT,d73e493a-f8a7-4f17-aa8d-ea56cd8abb53,0x800,0x100000)/File(\EFI\ubuntu\grubx64.efi)

So you need to change this to `shimx64.efi`.  I don't know what it was previously though, so it's hard to know how this failure happened unfortunately.

motorbreth, 2020-10-22:
Where do I change it?

superm1, 2020-10-22:
You can change it in BIOS setup, same way you added `grubx64.efi` or you can use `efibootmgr` to change it.

motorbreth, 2020-10-22:
i changed it in the bios now secure boot works
so there no log's we can check to see what went wrong ?

superm1, 2020-10-22:
Did fwupdmgr have you send up any reports?  We can check if anything got captured there.
Also, can you please check in `journalctl -u fwupd.service`

motorbreth, 2020-10-22:
- Logs begin at Wed 2019-09-11 11:22:58 IDT, end at Thu 2020-10-22 21:53:41 IDT. --
Sep 11 17:26:27 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Sep 11 17:26:27 yevgeni-desktop fwupd[2846]: disabling plugin because: failed to startup unifying: no kernel support for CONFIG_HIDRAW
Sep 11 17:26:29 yevgeni-desktop fwupd[2846]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 11 17:26:29 yevgeni-desktop fwupd[2846]: using plugins: amt, csr, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 11 17:26:29 yevgeni-desktop fwupd[2846]: Daemon ready for requests
Sep 11 17:26:29 yevgeni-desktop systemd[1]: Started Firmware update daemon.
Sep 11 11:30:15 yevgeni-desktop systemd[1]: Stopping Firmware update daemon...
Sep 11 11:30:15 yevgeni-desktop systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 12 19:14:23 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Sep 12 19:14:23 yevgeni-desktop fwupd[1702]: disabling plugin because: failed to startup unifying: no kernel support for CONFIG_HIDRAW
Sep 12 19:14:25 yevgeni-desktop fwupd[1702]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 12 19:14:26 yevgeni-desktop fwupd[1702]: using plugins: amt, csr, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 12 19:14:26 yevgeni-desktop fwupd[1702]: Daemon ready for requests
Sep 12 19:14:26 yevgeni-desktop systemd[1]: Started Firmware update daemon.
Sep 19 23:28:54 yevgeni-desktop systemd[1]: Stopping Firmware update daemon...
Sep 19 23:28:54 yevgeni-desktop systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 19 23:31:32 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Sep 19 23:31:34 yevgeni-desktop fwupd[2645]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 19 23:31:35 yevgeni-desktop fwupd[2645]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 19 23:31:35 yevgeni-desktop fwupd[2645]: Daemon ready for requests
Sep 19 23:31:35 yevgeni-desktop systemd[1]: Started Firmware update daemon.
Sep 22 21:20:01 yevgeni-desktop systemd[1]: Stopping Firmware update daemon...
Sep 22 21:20:01 yevgeni-desktop systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 27 00:26:15 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Sep 27 00:26:17 yevgeni-desktop fwupd[3448]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 27 00:26:18 yevgeni-desktop fwupd[3448]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 27 00:26:18 yevgeni-desktop fwupd[3448]: Daemon ready for requests
Sep 27 00:26:18 yevgeni-desktop systemd[1]: Started Firmware update daemon.
Sep 28 11:31:20 yevgeni-desktop systemd[1]: Stopping Firmware update daemon...
Sep 28 11:31:20 yevgeni-desktop systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 28 11:33:30 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Sep 28 11:33:32 yevgeni-desktop fwupd[2120]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 28 11:33:33 yevgeni-desktop fwupd[2120]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 28 11:33:33 yevgeni-desktop fwupd[2120]: Daemon ready for requests
Sep 28 11:33:33 yevgeni-desktop systemd[1]: Started Firmware update daemon.
Oct 08 14:45:11 yevgeni-desktop systemd[1]: Stopping Firmware update daemon...
Oct 08 14:45:11 yevgeni-desktop systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Oct 08 14:46:57 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Oct 08 14:46:59 yevgeni-desktop fwupd[2413]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Oct 08 14:47:00 yevgeni-desktop fwupd[2413]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Oct 08 14:47:00 yevgeni-desktop fwupd[2413]: Daemon ready for requests
Oct 08 14:47:00 yevgeni-desktop systemd[1]: Started Firmware update daemon.
Oct 16 19:28:55 yevgeni-desktop systemd[1]: Stopping Firmware update daemon...
Oct 16 19:28:55 yevgeni-desktop systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Oct 16 19:31:12 yevgeni-desktop systemd[1]: Starting Firmware update daemon...
Oct 16 19:31:15 yevgeni-desktop fwupd[2508]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Oct 16 19:31:15 yevgeni-desktop fwupd[2508]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Oct 16 19:31:15 yevgeni-desktop fwupd[2508]: Daemon ready for requests

motorbreth, 2020-10-22:
- Logs begin at Wed 2019-09-11 11:22:58 IDT, end at Thu 2020-10-22 21:53:41 ID>
Sep 11 17:26:27 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 11 17:26:27 ubuntu-lap fwupd[2846]: disabling plugin because: failed t>
Sep 11 17:26:29 ubuntu-lap fwupd[2846]: disabling plugin because: failed t>
Sep 11 17:26:29 ubuntu-lap fwupd[2846]: using plugins: amt, csr, ebitdo, n>
Sep 11 17:26:29 ubuntu-lap fwupd[2846]: Daemon ready for requests
Sep 11 17:26:29 ubuntu-lap systemd[1]: Started Firmware update daemon.
Sep 11 11:30:15 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Sep 11 11:30:15 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 12 19:14:23 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 12 19:14:23 ubuntu-lap fwupd[1702]: disabling plugin because: failed t>
Sep 12 19:14:25 ubuntu-lap fwupd[1702]: disabling plugin because: failed t>
Sep 12 19:14:26 ubuntu-lap fwupd[1702]: using plugins: amt, csr, ebitdo, n>
Sep 12 19:14:26 ubuntu-lap fwupd[1702]: Daemon ready for requests
Sep 12 19:14:26 ubuntu-lap systemd[1]: Started Firmware update daemon.
Sep 19 23:28:54 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Sep 19 23:28:54 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 19 23:31:32 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 19 23:31:34 ubuntu-lap fwupd[2645]: disabling plugin because: failed t>
Sep 19 23:31:35 ubuntu-lap fwupd[2645]: using plugins: amt, csr, unifying,>
Sep 19 23:31:35 ubuntu-lap fwupd[2645]: Daemon ready for requests
lines 1-23...skipping...
-- Logs begin at Wed 2019-09-11 11:22:58 IDT, end at Thu 2020-10-22 21:53:41 IDT. --
Sep 11 17:26:27 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 11 17:26:27 ubuntu-lap fwupd[2846]: disabling plugin because: failed to startup unifying: no kernel support for CONFIG_HIDRAW
Sep 11 17:26:29 ubuntu-lap fwupd[2846]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 11 17:26:29 ubuntu-lap fwupd[2846]: using plugins: amt, csr, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 11 17:26:29 ubuntu-lap fwupd[2846]: Daemon ready for requests
Sep 11 17:26:29 ubuntu-lap systemd[1]: Started Firmware update daemon.
Sep 11 11:30:15 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Sep 11 11:30:15 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 12 19:14:23 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 12 19:14:23 ubuntu-lap fwupd[1702]: disabling plugin because: failed to startup unifying: no kernel support for CONFIG_HIDRAW
Sep 12 19:14:25 ubuntu-lap fwupd[1702]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 12 19:14:26 ubuntu-lap fwupd[1702]: using plugins: amt, csr, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 12 19:14:26 ubuntu-lap fwupd[1702]: Daemon ready for requests
Sep 12 19:14:26 ubuntu-lap systemd[1]: Started Firmware update daemon.
Sep 19 23:28:54 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Sep 19 23:28:54 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 19 23:31:32 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 19 23:31:34 ubuntu-lap fwupd[2645]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 19 23:31:35 ubuntu-lap fwupd[2645]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 19 23:31:35 ubuntu-lap fwupd[2645]: Daemon ready for requests
Sep 19 23:31:35 ubuntu-lap systemd[1]: Started Firmware update daemon.
Sep 22 21:20:01 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Sep 22 21:20:01 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 27 00:26:15 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 27 00:26:17 ubuntu-lap fwupd[3448]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 27 00:26:18 ubuntu-lap fwupd[3448]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 27 00:26:18 ubuntu-lap fwupd[3448]: Daemon ready for requests
Sep 27 00:26:18 ubuntu-lap systemd[1]: Started Firmware update daemon.
Sep 28 11:31:20 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Sep 28 11:31:20 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Sep 28 11:33:30 ubuntu-lap systemd[1]: Starting Firmware update daemon...
Sep 28 11:33:32 ubuntu-lap fwupd[2120]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Sep 28 11:33:33 ubuntu-lap fwupd[2120]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Sep 28 11:33:33 ubuntu-lap fwupd[2120]: Daemon ready for requests
Sep 28 11:33:33 ubuntu-lap systemd[1]: Started Firmware update daemon.
Oct 08 14:45:11 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Oct 08 14:45:11 ubuntu-lap systemd[1]: Stopped Firmware update daemon.
-- Reboot --
[... cut ...]
Oct 16 19:31:15 ubuntu-lap fwupd[2508]: disabling plugin because: failed to coldplug thunderbolt_power: No support for force power via kernel or bolt
Oct 16 19:31:15 ubuntu-lap fwupd[2508]: using plugins: amt, csr, unifying, ebitdo, nitrokey, thunderbolt, wacomhid, dell, udev, upower, colorhug, dfu, steelseries, altos, synapticsmst, uefi
Oct 16 19:31:15 ubuntu-lap fwupd[2508]: Daemon ready for requests
Oct 16 19:31:15 ubuntu-lap systemd[1]: Started Firmware update daemon.
Oct 23 23:05:31 ubuntu-lap systemd[1]: Stopping Firmware update daemon...
Oct 23 23:05:31 ubuntu-lap systemd[1]: Stopped Firmware update daemon.

superm1, 2020-10-22:
Looks like nothing in the logs here for us to use.  Can you send up your `fwupdmgr report-history`

motorbreth, 2020-10-22:
No reports require uploading
i actually ran it yesterday by mistake and it uploaded something i dont know where

superm1, 2020-10-22:
I found your report in LVFS from 22 hours ago, but unfortunately nothing useful to be found out of it.  I would suggest the following.
1. Next time a firmware update is available collect the following:
`efibootmgr -v`
2. Install the update
3. If there is a problem report, sort it like you did this time around.
4. After you've resolved it, capture `efibootmgr -v` output again

Bring those to this bug report.

superm1, 2020-10-22:
For now I'll close it since there isn't anything we can do without more information, and you're back up and running.

closed 2020-10-22 by superm1