fwupd/fwupd #2507: No updates to process, called in error, start_image: invalid parameter

lzap, 2020-10-23:
Hello,

my Fedora X1 laptop suffers from an incorrect behavior for some time. I have experienced this in Fedora 31, 30 (I believe) and now after I upgraded to 32 I still see the same.

I download updates via Gnome Software, reboot, enter boot entry selection (Enter + F12), pick Linux Update Firmware and I am getting this error. The laptop is connected to AC.

I do see four CAB files downloaded, however it is not starting up.

I had one year old firmware, Richard told me to upgrade one by one up until the latest stable firmware to see if it helps. I've done this however there is no change in behavior. My ThinkPad X1, 6th gen simply does not work. I must have some EFI setting that prevents this.

How can I troubleshoot this?

hughsie, 2020-10-23:
> start_image: invalid parameter

I fear your currently running BIOS is too old to accept the capsule update. Would you be able to update to **one** newer than your current firmware using the CDROM .sio update method and then try again for a capsule? And then try the "next" one if that fails? I appreciate it might take some time, apologies. If that works then I can ping Lenovo to mark the new firmware as requiring at least this newer version.

lzap, 2020-10-23:
> I fear your currently running BIOS is too old to accept the capsule update

Richard, you asked me the same when we discussed this on the internal email thread, I did update from very old firmware about 8 times one by one up until the last version (May 2020). Let me pull the thread up here:

I am trying to update my X1 firmware using our awesome fwupdate from
the day one when I received it. It was Fedora 30, it did not work. I
thought I'd wait. Upgraded to F31 and now F32 but it still does not
work.

So I started digging. Looks like I need to interrupt the boot process
and load Linux Firmware Upgrade boot entry manually via Enter + F12.
Is this expected behavior?

Anyway, now I can see the error! I see four updates scheduled (some
UUID files), it lists them but then it fails with:

No updates to process
Called in error, start_image: invalid parameter

I tried to clean up updates using the CLI, redownload them. No luck.

Granted I have slightly old firmware which is probably 10-15 minor
releases older, maybe it's not compatible or capable of doing that.
Although I would not expect that, this should be driven by the
updater. Any ideas? I've reported this at

https://github.com/rhboot/fwupdate/issues/137

but searching the web does not give me any results with this error.
Something is really wrong.

This is a standard X1 setup provided by Red Hat IT. Maybe there is
some firmware setting I switched, luckily I made some notes when I was
configuring it:

Config > Power > Sleep State = Linux
Config > Network > Wake on LAN = Disabled
Config > Network > Wake on LAN from Dock = Disabled
Config > Thunderbolt (TM) 3 > Thunderbolt BIOS Assist Mode = Enabled
Config > Thunderbolt (TM) 3 > Wake by Thunderbolt (TM) 3 = Disabled
Config > Thunderbolt (TM) 3 > Security Level = No Security
Config > Intel AMT = Disabled
Security > Computrace > Current Setting = Disabled
Security > I/O Port Access > Fingerprint Reader = Disabled
Security > I/O Port Access > Wireless WAN = Disabled
Security > I/O Port Access > Memory Card Slot = Disabled

Also, I did my initial BIOS update before I installed Fedora 30, so I
used the ISO method:

https://lukas.zapletalovi.com/2019/07/thinkpad-x1-carbon-6th-gen-on-fedora.html

```
# fwupdmgr get-devices
20KGS23S08
│
├─Thunderbolt Controller:
│     Device ID:           6773845b50ab2da5e8c752197c87b70c642d6b9c
│     Souhrn:              Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc ← TBT-01091706
│                          b8360cb5-b152-506a-9990-49d7bbbc3b88 ←
TBT-01091706-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│
[... 49 lines cut ...]
      Current version:     0.1.38
      Minimum Version:     0.0.1
      Vendor:              LENOVO (DMI:LENOVO)
      Update State:        success
      GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update

# fwupdmgr update 1c53551e7da69d896138fac1ae131c83ad46d923
20KGS23S08 must remain plugged into a power source for the duration of
the update to avoid damage. Continue with update? [Y|n]:
Downloading 0.1.46 for System Firmware...
Rozbaluje se…            [***************************************]
Autentizuje se…          [***************************************]
Updating System Firmware…[     \                                 ]
Plánuje se…              [***************************************]
Successfully installed firmware
```

Yet during update I am getting the very same error.

lzap, 2020-10-23:
Note now I do have a version from May 2020, the paste above is from May.

hughsie, 2020-10-23:
Ahh, you're the same Lukáš -- sorry!