fwupd/fwupd #2508: fwupd does not start at all (permission denied on smbios_entry_point)

bluppfisk, 2020-10-23:
**Describe the bug**
I noticed Ubuntu 20.04 not pulling any firmware updates on Dell XPS 13 9300.

Starting `sudo service fwupd start` would result in a timed out message.

Running the libexec executable yielded the following errors:

```
**@**:~$ /usr/libexec/fwupd/fwupd
10:44:22:0197 FuEngine             Failed to load SMBIOS: Failed to open file “/sys/firmware/dmi/tables/smbios_entry_point”: Permission denied
10:44:22:0235 FuPluginUefi         SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: SMBIOS not supported
10:44:22:0244 FuPluginUefi         failed to add /sys/firmware/efi/esrt/entries/entry0: ESRT GUID '(null)' was not valid
10:44:22:0265 FuPluginDfu          0bda:5511 is missing download capability
10:44:22:0265 FuEngine             failed to add USB device 0bda:5511: USB error on device 0bda:5511 : Access denied (insufficient permissions) [-3]
10:44:22:0300 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/drm_dp_aux1: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux1: Permission denied
10:44:22:0300 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/drm_dp_aux2: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux2: Permission denied
10:44:22:0301 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-3/drm_dp_aux3: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux3: Permission denied
10:44:22:0301 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/drm_dp_aux4: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux4: Permission denied
10:44:22:0302 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/drm_dp_aux0: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux0: Permission denied
10:44:22:0304 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-DLL096D:01/0018:06CB:CDE6.0001/hidraw/hidraw0: failed to open /dev/hidraw0: Permission denied
ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 
10:44:22:0306 FuEngine             failed to add udev device /sys/devices/platform/STM7408:00/tpm/tpm0: failed to initialize TPM library
10:44:22:0332 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:1d.0/0000:57:00.0/nvme/nvme0: failed to open /dev/nvme0: Permission denied

```

**Steps to Reproduce**
Described above. Basically just start the service.

**Expected behavior**
I expected not even needing to start the fwupd service, but also for it to start on demand

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
... it hangs there, presumably because it fails to contact the daemon

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
it came pre-installed with Ubuntu but I re-instelled with apt.
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2020-10-23:
> Permission denied

Are you running it as root?

bluppfisk, 2020-10-23:
> > Permission denied
> 
> Are you running it as root?

@hughsie I am indeed daft. I was trying to reproduce the error so hard that I forgot sudo. I've updated the original report with verbose output from the libexec executable.

superm1, 2020-10-23:
Any chance you can install the debug symbols and attach a debugger to see where it's spinning?

bluppfisk, 2020-10-23:
@superm1 it would be my pleasure. However, I'm not sure what or how. Should I just follow the instructions here : https://wiki.ubuntu.com/Debug%20Symbol%20Packages, then run it again? Where should I find the output/debug information?

bluppfisk, 2020-10-23:
installed the debug symbols as above, then `apt-install fwupd-dbgsym`, then `sudo strace  /usr/libexec/fwupd/fwupd -v` yields the following (only the last few lines; full file here: [fwupd.trace.txt](https://github.com/fwupd/fwupd/files/5429221/fwupd.trace.txt))

```
write(2, "\33[32m11:46:45:0543 ", 1911:46:45:0543 )   = 19
write(2, "FuHistory            ", 21FuHistory            )   = 21
write(2, "\33[34mget device\n\33[0m", 20get device
) = 20
fcntl(6, F_SETLK, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
fcntl(6, F_SETLK, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
fcntl(6, F_SETLK, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
stat("/var/lib/fwupd/pending.db-journal", 0x7ffd4cfe5240) = -1 ENOENT (No such file or directory)
pread64(6, "\0\0\0\2\0\0\0\4\0\0\0\0\0\0\0\0", 16, 24) = 16
stat("/var/lib/fwupd/pending.db-wal", 0x7ffd4cfe5240) = -1 ENOENT (No such file or directory)
fstat(6, {st_mode=S_IFREG|0644, st_size=16384, ...}) = 0
fcntl(6, F_SETLK, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
write(2, "\33[32m11:46:45:0543 ", 1911:46:45:0543 )   = 19
write(2, "FuIdle               ", 21FuIdle               )   = 21
write(2, "\33[34minhibiting: thunderbolt req"..., 56inhibiting: thunderbolt requires device wakeup
) = 56
write(2, "\33[32m11:46:45:0543 ", 1911:46:45:0543 )   = 19
write(2, "FuIdle               ", 21FuIdle               )   = 21
write(2, "\33[34minhibiting: thunderbolt req"..., 56inhibiting: thunderbolt requires device wakeup
) = 56
write(2, "\33[32m11:46:45:0543 ", 1911:46:45:0543 )   = 19
write(2, "FuPlugin             ", 21FuPlugin             )   = 21
write(2, "\33[34mperforming coldplug() on tp"..., 47performing coldplug() on tpm_eventlog
) = 47
openat(AT_FDCWD, "/sys/kernel/security/tpm0/binary_bios_measurements", O_RDONLY) = 18
fstat(18, {st_mode=S_IFREG|0440, st_size=0, ...}) = 0
fcntl(18, F_GETFL)                      = 0x8000 (flags O_RDONLY|O_LARGEFILE)
fstat(18, {st_mode=S_IFREG|0440, st_size=0, ...}) = 0
read(18,  <unfinished ...>)             = ?
```

$ ls -l /sys/kernel/security/tpm0/binary_bios_measurements 
-r--r----- 1 root root 0 okt 23 11:37 /sys/kernel/security/tpm0/binary_bios_measurements

hughsie, 2020-10-23:
can you, as root, `cat /sys/kernel/security/tpm0/binary_bios_measurements` ?

bluppfisk, 2020-10-23:
Yes:

```shell
$ sudo cat /sys/kernel/security/tpm0/binary_bios_measurements 
Killed
```

superm1, 2020-10-23:
Can you look at dmesg after that?

bluppfisk, 2020-10-23:
Thanks for looking at this. I've attached the whole output because I'm not sure what to look for.

[fwupd.dmesg.txt](https://github.com/fwupd/fwupd/files/5429620/fwupd.dmesg.txt)

hughsie, 2020-10-23:
Is that the stock Ubuntu kernel?

superm1, 2020-10-23:
`5.9.1-050901-generic` : no that's not a stock kernel . It looks like it's probably a mainline kernel from the kernel PPA if I was gonna guess.  The particular crash is:
```
[ 2712.709336] BUG: kernel NULL pointer dereference, address: 000000000000002c
[ 2712.709341] #PF: supervisor read access in kernel mode
[ 2712.709343] #PF: error_code(0x0000) - not-present page
[ 2712.709344] PGD 0 P4D 0 
[ 2712.709347] Oops: 0000 [#1] SMP NOPTI
[ 2712.709350] CPU: 6 PID: 14562 Comm: fwupd Not tainted 5.9.1-050901-generic #202010170731
[ 2712.709352] Hardware name: Dell Inc. XPS 13 9300/0Y4GNJ, BIOS 1.2.0 08/13/2020
[ 2712.709358] RIP: 0010:tpm2_bios_measurements_start+0x38/0x1f0
[ 2712.709360] Code: 55 41 54 53 48 83 ec 30 4c 8b 16 65 48 8b 04 25 28 00 00 00 48 89 45 d0 48 8b 47 70 4c 8b a0 d0 06 00 00 48 8b 88 d8 06 00 00 <41> 8b 5c 24 1c 48 89 4d b0 48 89 d8 48 83 c3 20 4d 85 d2 75 31 4c
[ 2712.709362] RSP: 0018:ffffbc0ac424bde8 EFLAGS: 00010286
[ 2712.709364] RAX: ffff9efbf25bd000 RBX: 0000000000000000 RCX: 0000000000000010
[ 2712.709366] RDX: 0000000000005fb8 RSI: ffff9efbfd25bbe0 RDI: ffff9efbfd25bbb8
[ 2712.709367] RBP: ffffbc0ac424be40 R08: 0000000000005fb8 R09: 0000000000000000
[ 2712.709368] R10: 0000000000000000 R11: 0000000000000002 R12: 0000000000000010
[ 2712.709370] R13: ffff9efbfd25bbb8 R14: ffff9efbfd25bbe0 R15: ffff9efbfd25bbf0
[ 2712.709372] FS:  00007fd79b6e8b00(0000) GS:ffff9efbff980000(0000) knlGS:0000000000000000
[ 2712.709373] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2712.709375] CR2: 000000000000002c CR3: 00000002e6fc8003 CR4: 0000000000770ee0
[ 2712.709376] PKRU: 55555554
[ 2712.709377] Call Trace:
[ 2712.709383]  seq_read+0x95/0x470
[ 2712.709387]  vfs_read+0xa6/0x190
[ 2712.709390]  ksys_read+0x67/0xe0
[ 2712.709392]  __x64_sys_read+0x1a/0x20
[ 2712.709396]  do_syscall_64+0x38/0x90
[ 2712.709400]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
```

I would suggest to try to isolate where it's coming from.  If you Use a distro kernel does it happen to?  If you use a newer kernel does it happen?

bluppfisk, 2020-10-23:
No (that is 5.6 I believe); but I've seen this behaviour on most recent kernels (5.8, 5.9 and 5.9.1). Since I've been evolving Ubuntu 20.04 since kernel 5.4 and it has never pulled a BIOS update for me, I believe this exists in all kernels. That said, I will try using different kernels that are installed and report back.

superm1, 2020-10-23:
The top of the `dmesg` is where I got that version from.  I think for now until we have this figured out you can block the 
`tpm_eventlog` plugin from running by modifying `/etc/fwupd/daemon.conf` and disabling the `tpm_eventlog` plugin in there.

bluppfisk, 2020-10-23:
Yet the `tpm_eventlog` plugin does not seem to be active or at least I don't see it being activated in `daemon.conf`. Perhaps I need to explicitly disable it? What would be the config line?

```ini
[fwupd]

# Allow blacklisting specific devices by their GUID
# Uses semicolons as delimiter
BlacklistDevices=

# Allow blacklisting specific plugins
# Uses semicolons as delimiter
BlacklistPlugins=test;invalid

# Maximum archive size that can be loaded in Mb, with 0 for the default
ArchiveSizeMax=0

# Idle time in seconds to shut down the daemon -- note some plugins might
# inhibit the auto-shutdown, for instance thunderbolt.
#
# A value of 0 specifies 'never'
IdleTimeout=7200

# Comma separated list of domains to log in verbose mode
# If unset, no domains
# If set to FuValue, FuValue domain (same as --domain-verbose=FuValue)
# If set to *, all domains (same as --verbose)
VerboseDomains=

# Update the message of the day (MOTD) on device and metadata changes
UpdateMotd=true
```

superm1, 2020-10-23:
Add to BlacklistPlugins=test;invalid

bluppfisk, 2020-10-23:
@superm1 @hughsie indeed; disabling the tpm_eventlog plugin by adding `tpm_eventlog` to the line `BlacklistPlugins` works. After this I can start the daemon using `sudo service fwupd start` and then `sudo fwupdmgr refresh` works just right.

Much obliged.

superm1, 2020-10-23:
@bluppfisk glad to hear that workaround worked for you.  I would expect you can still reproduce the kernel problem by using `cat` on that file.  I would suggest either reproducing with the ubuntu kernel and filing a kernel bug with `ubuntu-bug linux` or reproducing with linus' tree and then filing a kernel bugzilla.

superm1, 2020-10-23:
Since this isn't a fwupd bug (per-say) i'm going to close this report.  However please report back if you file a bug, I'd definitely like to follow along on any findings.

closed 2020-10-23 by superm1

bluppfisk, 2020-10-24:
@superm1 That's cool. I thought I'd just report that it also happens with the stock kernel (5.4 - updated to minor revision 52)