fwupd/fwupd #2513: Raise a reconsideration to incorporate Ubuntu Core as part of design

woodrow-shen, 2020-10-26:
**Is your feature request related to a problem? Please describe.**
I knew it's a known discussion in the past and @timchen119 had some patches carried in the confined fwupd (aka uefi-fw-tools) for ubuntu core. We realized that the request of using fwupd in core is sustainable from the customers, and more importantly we also understand the limitations that ubuntu core didn't support yet until now. For example, the policykit in core is one of major feature requests that need to struggle to overcome it from Canonical. We'll try to review all dependencies what fwupd needs for core, so I'd like to create this issue to discuss something further, e.g. how to deal with snapd reseal during fw update, especially when Ubuntu Core 20 is coming soon with secured FDE and TPM measurement.

Currently, I rebase the upstream to make uefi-fw-tools work since 1.4.5, and first thing I don't want it to be patched is about the detection of ubuntu core system. In classic, in the function `fu_uefi_get_esp_path_for_os` in plugins/uefi/fu-uefi-common.c, the fwupd parses the "ID" field from /etc/os-release and ESP metadata to combine the full path that will be used to copy the signed fwupd.efi and a capsule file under /boot/efi/EFI/ubuntu/fw/. In case of ubuntu-core, ID from /etc/os-release will be "ubuntu-core" instead of "ubuntu", so that I had to add a condition to weave this case. 

**Describe the solution you'd like**
By our internal discussion, the ESP layout ubuntu core used is /EFI/BOOT/BOOT*.EFI path, and in this case shim binary path is /EFI/boot/bootx64.efi, and fwupd binary and cap file are also under /EFI/boot/ and /EFI/boot/fw/ respectively. Not sure if you have any concerns about the proposal.

**Describe alternatives you've considered**
I also had a PR of fwupd interface regarding Ubuntu Core ESP layout:
https://github.com/snapcore/snapd/pull/9514

**Additional context**
There is current development for uefi-fw-tools:
https://github.com/Lyoncore/fwupd-snapcraft/tree/fwupd-upstream

Not sure if what I provided here is clear to all maintainers, but I think we can have a checklist after having some conversations.

Please review it and any opinions or recommendations are welcome, thanks.

Woodrow

hughsie, 2020-10-26:
> the policykit in core

The soon-to-be-released 1.5.0 in master drops polkit as a hard dep, although you obviously don't get any configuration of authorisation actions.

> how to deal with snapd reseal during fw update

We're open for ideas here, I know there are at least 3 other projects that want to deal with the PCR0 changing when firmware is updated.

> fwupd parses the "ID" field from /etc/os-release and ESP metadata to combine the full path

Can't you set EFI_OS_DIR when configuring the system?

superm1, 2020-10-26:
>Can't you set EFI_OS_DIR when configuring the system?

It seems they don't want to enable dual boot scenarios to me, and choose to use UEFI removable path rather than their own path in ESP.
 
> the ESP layout ubuntu core used is /EFI/BOOT/BOOT*.EFI path,

I think this is _potentially_ asking for problems with interactions with the UEFI boot manager depending upon the vendor's implementation.  If you're not using/creating `Boot####` NVRAM entries during your OS installer, you may find that changing `BootNext` or `BootOrder` don't work well.  It's up to vendor how to prioritize "removable" path of `/efi/boot` in their BDS implementation.  If you can prove this worry to be wrong (perhaps by UEFI BDS spec quotes), then I think you can submit PR to special case code handling when using `/efi/boot` instead of regular path.  I think this could be perhaps a configuration option in `uefi.conf` to use removable path instead of `/etc/os-release` to build the path.

> The soon-to-be-released 1.5.0 in master drops polkit as a hard dep, although you obviously don't get any configuration of authorisation actions.

It's also currently implemented to only allow installation of CAB file that is signed by trusted authority installed into `/etc/pki/fwupd`.  By default that's LVFS.

> We're open for ideas here, I know there are at least 3 other projects that want to deal with the PCR0 changing when firmware is updated.

We had talked previously about using hooks of some sort.  Perhaps we can call into a service or hook directory.

> secured FDE and TPM measurement.

What PCR values are sealed against?  Does user get a choice?  How to handle that you won't know new final PCR value in advance?

superm1, 2020-10-26:
> Not sure if what I provided here is clear to all maintainers, but I think we can have a checklist after having some conversations

You need to add confinement rules for many other plugins that are important today. Nvme needs `cap_sys_admin`. Many plugins need `/dev/bus/usb`. MST plugin needs `/dev/drm_dp_aux#`. There are probably others, but those are the ones that come to mind.

hughsie, 2020-10-26:
> Nvme needs cap_sys_admin. Many plugins need /dev/bus/usb

Should we put some of this stuff into each plugins README.md file?

superm1, 2020-10-26:
> Should we put some of this stuff into each plugins README.md file?

Sure

superm1, 2020-10-26:
@woodrow-shen we documented all the interfaces directly in each plugin's `README.md` as part of https://github.com/fwupd/fwupd/pull/2516.  You can reference them to find all the interfaces you need for all plugins.

woodrow-shen, 2020-10-27:
> EFI_OS_DIR

I'm not aware of this option, yes, it should be good enough to apply the correct one with it. But, there is a small problem here. I check the function fu_uefi_bootmgr_bootnext in plugins/uefi/fu-uefi-bootmgr.c and it shows that fu_uefi_get_esp_app_path (esp_path, "shim", error) gave the fixed string "shim" to compose the shim's path. As mentioned above, the filename of shim in core is bootx64.efi. Any ideas of how to handle this in order to give a right place upon the basis of OS?

woodrow-shen, 2020-10-27:
> @woodrow-shen we documented all the interfaces directly in each plugin's `README.md` as part of #2516. You can reference them to find all the interfaces you need for all plugins.

@superm1 Thanks for providing the comprehensive docs! I'll take some time to review them.

superm1, 2020-10-27:
> Any ideas of how to handle this in order to give a right place upon the basis of OS?

As I mentioned above I think that a PR for special casing handling for UEFI removable path might make sense if it's deemed safe.
Maybe `EnableUefiRemovablePath=false` in `uefi.conf`.  If it is set to true then don't use `/etc/os-release` and use `bootx64.efi` instead of `shimx64.efi`

>@superm1 Thanks for providing the comprehensive docs! I'll take some time to review them.

Sure.  Some of the plugins might not make too much sense in your context (such as superio only works if kernel lockdown is turned off and flashrom might not matter for some devices you support initially since you're using UEFI).

woodrow-shen, 2020-10-27:
> It seems they don't want to enable dual boot scenarios to me, and choose to use UEFI removable path rather than their own path in ESP.
>
>
> I think this is _potentially_ asking for problems with [...]

I didn't dig into our FDE implementation spec., but from my understanding we measured PCR7/8/11 for various purposes:

- PCR7: SB configuration, certificate for shim, grub, and kernel
- PCR8: for grub static configuration and kernel cmdline
- PCR11: public key use for internal assertion and recovery payload

As far as I know Ubuntu Core 20 can be installed automatically without interaction. After that, the user can have a choice to enter recovery mode for particular reasons, but I'm not sure how it impacts the PCR changes. I'd like to invite our product engineering to join this discussion as well for better explanations.

woodrow-shen, 2020-10-27:
> As I mentioned above I think that a PR for special casing handling for UEFI removable path might make sense if it's deemed safe.
> Maybe `EnableUefiRemovablePath=false` in `uefi.conf`. If it is set to [...]

Great! Thanks for hints. Let me try this way and send a PR afterwards.
> Sure. Some of the plugins might not make too much sense in your context (such as superio only works if kernel lockdown is turned off and flashrom might not matter for some devices you support [...]

superm1, 2020-10-27:
> As far as I know Ubuntu Core 20 can be installed automatically without interaction

How is that possible? Shouldn't you need to have a pin to seal a secret for decryption to a heirarchy? I would think such an object will be needed to reseal as well.

Maybe can you share full spec so we can understand better how you do it?

woodrow-shen, 2020-10-29:
> > Great! Thanks for hints. Let me try this way and send a PR afterwards
> 
> One more thought. Maybe this new option should enable using the extra path as a fallback if the regular paths don't work. I feel like then you have the potential that this confined snap can actually be used outside of Ubuntu core as well at some point.

@superm1 What extra path can be used? What you mean is to use EFI_OS_DIR or just the original way (/etc/os-release) as a fallback?

The another thing I would like to ask is about ESP path for UC20. It's actually mounted to /run/mnt/ubuntu-seed by snapd in current design, so initially I thought we should have a bind mount from /run/mnt/ubuntu-seed to /boot/efi. However, I check that we can use OverrideESPMountPoint to overwrite this path. If this option can be applied for UC20, I think we're fine without above PR sent to snapcore/core-initrd @xnox pointed out.

Thanks.

superm1, 2020-10-29:
> @superm1 What extra path can be used? What you mean is to use EFI_OS_DIR or just the original way (/etc/os-release) as a fallback?

It depends if you:
1. want the confined snap to work outside of UC20
and 
2. your design will continue to use UEFI removable path.  

If you will use UEFI removable path and you don't care about it working outside of UC20, then go ahead and set `EFI_OS_DIR` as `boot`.

If you want it to work outside of UC20 my suggestion was going to be for the OS distributor to be pulled from `/etc/os-release` in host environment and set it as a runtime variable that can be accessed here: https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-common.c#L222.
Then snap can request an interface to access ESP and can know which directory to use based on what host it's running on.  And for UC20, add new code to use UEFI removable path instead of the OS vendor's path when a new a fallback option when `uefi.conf` had option set which would be part of your confined snap build.

> The another thing I would like to ask is about ESP path for UC20. It's actually mounted to /run/mnt/ubuntu-seed by snapd in current design, so initially I thought we should have a bind mount from [...]

This sounds reasonable to me, but again I would caution whether you want this confined snap to work outside of UC20.  If so, will `snapd` always have `/run/mnt/ubuntu-seed` at that path?

woodrow-shen, 2020-10-29:
> It depends if you:
>
>     1. want the confined snap to work outside of UC20
>        and
>
>     2. your design will continue to use UEFI removable path.
>
>
> If you will use UEFI removable path and you don't [...]
Yes, I think more generic way will be best for the upstream, so essentially that's my purpose.

> This sounds reasonable to me, but again I would caution whether you want this confined snap to work outside of UC20. If so, will `snapd` always have `/run/mnt/ubuntu-seed` at that path?

Yes, it does, and UC20 won't have /boot/efi to be mounted.

woodrow-shen, 2020-10-29:
@hughsie @superm1 I'm working on adding `EnableUefiRemovablePath` to uefi.conf but I got a little obstacle that using `fu_plugin_get_config_value` needs FuPlugin object which is not easy to be refered in the function https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-common.c#L222. The work around I can figure out is to parse uefi.conf directly as static function insdie fu-uefi-common.c. as this is for uefi plugin indeed. Any ideas?

hughsie, 2020-10-29:
> Any ideas?

Couldn't we set the UEFI path using a quirk? Then Ubuntu core would just ship one extra quirk file...

superm1, 2020-10-29:
>Couldn't we set the UEFI path using a quirk? Then Ubuntu core would just ship one extra quirk file...

If they're going to only ever support the confined snap on Ubuntu core, they could just compile with `EFI_OS_DIR` and don't need a quirk even.

>The work around I can figure out is to parse uefi.conf directly as static function inside fu-uefi-common.c. as this is for uefi plugin indeed. Any ideas?

Parse here: https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-plugin-uefi.c#L425
Set it as metadata on the device.
Change `fu_uefi_get_esp_app_path` to take a `FuUefiDevice` as argument, and look up ESP path from metadata directly.  Also look up this new boolean option from device. 
Look for case of `unknown` (assuming that's what normally happens).  If Something else happens, look for that case.

woodrow-shen, 2020-10-30:
That PR is just verified on Dell edge 5000, and of course I need your feedback to revise accordingly. Thanks!

woodrow-shen, 2020-11-03:
As fwupd quickly released the 1.5.1, I have few questions after checking the release notes:

1. When 1.5.0 claimed that it dropped efivar dependency, I still see the `efi_generate_file_device_path` in the fu-uefi-device.c/fu-uefi-bootmgr.c that use open syscall to access disks, and it was unacceptable for the confined snap. Will this function be replaced with the another way in future?

2. Above comment mentioned that policykit will be dropped, but I don't see any commits related to this. Still WIP?

hughsie, 2020-11-03:
> I still see the efi_generate_file_device_path in the fu-uefi-device.c/fu-uefi-bootmgr.c

Ya, we're still using efibootmgr which in turn uses efivar.

> Will this function be replaced with the another way in future

I wasn't planning to.

> Above comment mentioned that policykit will be dropped

The polkit dep is optional since 1.5.0 -- although disabling it is little-tested.

superm1, 2020-11-03:
> Ya, we're still using efibootmgr which in turn uses efivar.

Specifically `libefiboot1` which is what `efibootmgr` uses too.

> uses open syscall to access disks

I believe you'll need to propose a change here: https://github.com/rhboot/efivar should you want to hcange this.

>The polkit dep is optional since 1.5.0 -- although disabling it is little-tested.

Keep in mind it will only allow installation of content trusted by the keys in `/etc/pki/fwupd`.  By default this is LVFS.

woodrow-shen, 2020-11-04:
> > I still see the efi_generate_file_device_path in the fu-uefi-device.c/fu-uefi-bootmgr.c
> 
> Ya, we're still using efibootmgr which in turn uses efivar.
> 
> > Will this function be replaced with the another way in future
> 
> I wasn't planning to.

I see. That's fine.
> 
> > Above comment mentioned that policykit will be dropped
> 
> The polkit dep is optional since 1.5.0 -- although disabling it is little-tested.

Oh, never notice this. Thanks for pointing out.

woodrow-shen, 2020-11-04:
> > Ya, we're still using efibootmgr which in turn uses efivar.
> 
> Specifically `libefiboot1` which is what `efibootmgr` uses too.
> 
> > uses open syscall to access disks
> 
> I believe you'll need to propose a change here: https://github.com/rhboot/efivar should you want to hcange this.

Yes, you're right. Interestingly, it doesn't update since the end of 2018 :|
> 
> > The polkit dep is optional since 1.5.0 -- although disabling it is little-tested.
> 
> Keep in mind it will only allow installation of content trusted by the keys in `/etc/pki/fwupd`. By default this is LVFS.

superm1, 2020-11-04:
> Yes, you're right. Interestingly, it doesn't update since the end of 2018 :|

It's changing regularly, I see commit 19 days ago. But yes no release in a long time.

hughsie, 2020-11-04:
> Specifically libefiboot1

I'm not against dropping this as a dep, I'm just pretty sure I don't want to be the one to do it. At the moment having the code shared makes sense, although I agree if nothing else uses the API that we're using we should ask Peter if it should be moved.

xnox, 2020-11-04:
We don't seal to PCR8, instead we seal to PCR12 where we have chosen to measure sd-boot stub kernel cmdline & "model assertion" (Ubuntu Core specific definition of the system).

Re:PIN at the moment at install time we seal against PCR measurements which remain the same without user interactions or input (no pin / password / biometrics). The design is extensible and they can be mixed-in/added later. The PCR measurements however do take the firmware state into account, variables, certs and hashes of boot artefacts, etc. Some of which can be affected by fwupdate.

For Ubuntu Core itself, we currently do not create BootEntry nor set the BootOrder. I guess we could create one for our bootflow, and then have another one for the fwupdate. Such that we can flip flop between then. However, care would have to be taken in resealing to them. I don't think we'd want to release the full disk encryption secret when booted into fwupdate.

What happens at the end of fwupdate in EFI envrionment? Are Boot entries removed, BootNext cleared to return the system to pristine state?

superm1, 2020-11-04:
> We don't seal to PCR8, instead we seal to PCR12 where we have chosen to measure sd-boot stub kernel cmdline & "model assertion" (Ubuntu Core specific definition of the system).

> Re:PIN at the moment at install time we seal against PCR measurements which remain the same without user interactions or input (no pin / password / biometrics). The design is extensible and they can [...]

Do you have some further documentation or links to code that we can see on this to better understand everything about firmware being measured?

> For Ubuntu Core itself, we currently do not create BootEntry nor set the BootOrder. I guess we could create one for our bootflow, and then have another one for the fwupdate. Such that we can flip [...]

Yes, I agree you wouldn't want to be releasing the FDE secret to FW updates.  I think your stage 1 bootloader would either need to be extending PCR12 with what the second stage bootloader selected is if you don't already or the BootEntry selected.

 >What happens at the end of fwupdate in EFI envrionment? Are Boot entries removed, BootNext cleared to return the system to pristine state?

* BootNext is cleared by the firmware automatically, it's a one time only boot option.
* BootXXXX which was created remains
* BootOrder remains

We _used to_ remove the BootXXXX entry and get it back to clean state, but it had problems on some UEFI implementations.  Stuff like this is why I think it's very dangerous to rely upon how NVRAM variables do (or don't) work on some systems and why I really don't think you should use UEFI removable path.
https://github.com/fwupd/fwupd/commit/60373e03fd0668d379ba0d7b90a6f02ba7d87478

xnox, 2020-11-04:
PCR12 is extended by us, by things that are themselves signed-verified-measured and cover all the bootloaders, kernel commandlines, kernel, initrd, model. I don't think we have an uptodate public documents around that. There is code in public in snapd / secboot which is the most up to date thing.

We also have issues with creating BootXXX variables and BootOrder on some devices. Which is less than ideal. I don't think any of them publish fwupdates however =)

woodrow-shen, 2020-11-24:
Just for record.

Recently we just found that gpg package is not landed in core18, and it causes gpg related errors when using `fwupdmgr refresh`

1. signature failure: failed to set protocol: Invalid crypto engine
2. signature failure: failed to import GPG-KEY-Linux-Vendor-Firmware-Service: Unsupported protocol

However, it's weird that we didn't hit this case before, until these days. But anyway, currently we just make gpg installed under snap, and ensure that gpgme can operate correctly.

superm1, 2020-11-28:
So one possibility for you - just compile libjcat without GPG support.  LVFS remotes will have both PKCS7 and GPG support so you don't really _need_ to compile both in.

woodrow-shen, 2020-12-08:
@superm1 I've pushed a PR about the fwupd interface to support any distros on classic system:

https://github.com/snapcore/snapd/pull/9766

fyi

closed 2020-12-08 by superm1