fwupd/fwupd #2514: LVFS returning Bad Gateway for system firmware download with broken contact us mailto link

thimslugga, 2020-10-26:
**Describe the bug**
A clear and concise description of what the bug is.

I wasn't 100% sure where to report LVFS website issues as the repo for it appears to be an archive with PRs only allowed. Anyways attempting to download the below files to update the system firmware for a Dell XPS 13 9360 results in receiving a bad gateway error. I also double checked by visiting the returned URL which returns 'LVFS Something went wrong!' which has a contact us mailto that isn't rendering correctly e.g. mailto:{{admin_email}}.

* https://fwupd.org/lvfs/devices/com.dell.uefi5ffdbc0d.firmware
* https://fwupd.org/downloads/945a07c7e55207d87aab392cf55d008bf85cae4154d9a11ade522841e93327d2-Dell%20XPS%2013%209360%20System%20BIOS_Ver.2.14.2.cab

```
LVFS

Something went wrong!

Try refreshing the page, or going back and attempting the action again. If this problem persists, contact us.
```
```
mailto:{{admin_email}}
```
```shell
curl -I https://fwupd.org/downloads/945a07c7e55207d87aab392cf55d008bf85cae4154d9a11ade522841e93327d2-Dell%20XPS%2013%209360%20System%20BIOS_Ver.2.14.2.cab
HTTP/1.1 502 Bad Gateway
Server: nginx/1.16.1
Date: Mon, 26 Oct 2020 13:37:06 GMT
Content-Type: text/html
Content-Length: 1451
Connection: keep-alive
ETag: "5f7f3815-5ab"
```

**Steps to Reproduce**
Steps to reproduce the behavior.

```shell
$ fwupdmgr refresh --force
$ fwupdmgr update
• INTEL SSDPEKNW010T8 has no available firmware updates
Upgrade available for System Firmware from 2.13.0 to 2.14.2
XPS 13 9360 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Unknown                  [***************************************]
Failed to download https://fwupd.org/downloads/945a07c7e55207d87aab392cf55d008bf85cae4154d9a11ade522841e93327d2-Dell%20XPS%2013%209360%20System%20BIOS_Ver.2.14.2.cab: Bad Gateway
```

OR

* https://fwupd.org/downloads/945a07c7e55207d87aab392cf55d008bf85cae4154d9a11ade522841e93327d2-Dell%20XPS%2013%209360%20System%20BIOS_Ver.2.14.2.cab

**Expected behavior**
A clear and concise description of what you expected to happen.

The system firmware is updated.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-updates
• INTEL SSDPEKNW010T8 has no available firmware updates
• TPM 2.0 has no available firmware updates
• Touchpad has no available firmware updates
XPS 13 9360
│
└─System Firmware:
  │   Device ID:           redacted
  │   Current version:     2.13.0
  │   Minimum Version:     2.13.0
  │   Vendor:              Dell Inc. (DMI:Dell Inc.)
  │   GUID:                redacted
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─XPS 13 9360 System Update:
        New version:       2.14.2
        Remote ID:         lvfs
        Summary:           Firmware for the Dell XPS 13 9360
        License:           Proprietary
        Size:              11.5 MB
        Created:           2020-07-14
        Urgency:           Critical
        Vendor:            Dell Inc.
        Flags:             is-upgrade
        Description:       This stable release fixes the following issues:
        
         • Firmware updates to address the Intel Security Advisories
         • Firmware updates to address the Intel Technical Advisory
         • Fixed the issue where the SMM Security Mitigation gets disabled while loading BIOS defaults.
         • Fixed an issue with the hard drive master password reset not functioning.
```

hughsie, 2020-10-26:
Eek! Fixed in https://gitlab.com/fwupd/lvfs-website/-/commit/970d06e08f379e6b6479c3c940574217bd754ffb thanks for telling us. I've deployed this, please confirm it now works.

closed 2020-10-26 by hughsie

thimslugga, 2020-10-26:
Looks good 👍