fwupd/fwupd #2523: vli: Share SPI Update

memily, 2020-10-28:
**Describe the bug**
For all share spi devices (Mini Dock, TR_Gen2, Hyper), only VL103 PD is updating/downgrading suddenly. 

**Steps to Reproduce**
local fwupd+fwupdmgr install corresponding *.cab on end product device.

**Expected behavior**
Both hub+pd to update. We see that update_v2 is being called from fu-vli-usbhub-device.c and that erase, write_H1+write_H2 execute with no error. However, upon reading back the contents of spi flash, we see one H1, unchanged hub fw (no H2), and the expected updated/downgraded pd. 

We've attached ./fwupd --verbose, however there's less information? Do we have to set flags or some different procedure?
[verbose.zip](https://github.com/fwupd/fwupd/files/5450806/verbose.zip)

**fwupd version information**
```
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.5
	efivar:	37
daemon version:	1.5.0
```

joakim-tjernlund, 2020-10-28:
What about the Ultra Docks ? Do they they have the same problem?

hughsie, 2020-10-28:
> only VL103 PD is updating/downgrading

What hardware? Can you provide .cab files showing the problem please?

memily, 2020-10-29:
Sorry, I tested with the same Mini Dock *.cab files as the #2376 thread. Ultra Dock is not affected. Here are the *.cab files for Lenovo Mini Dock, the behavior is the same across all VL817+VL103 share spi devices.

[Lenovo_Mini_Dock.zip](https://github.com/fwupd/fwupd/files/5456751/Lenovo_Mini_Dock.zip)

joakim-tjernlund, 2020-10-29:
>  Ultra Dock is not affected. 
Then could you please release Ultra Dock to LVFS ?

hughsie, 2020-10-29:
@memily I might have broken something in 1.5.0 -- can you try with master please? Using master I see VL211 being updated, then VL103 and then VL817. If that works for you I'll do a 1.5.1 on Monday for you.

memily, 2020-10-30:
>can you try with master please? 

Status update from the master branch: problem unborked! Update/downgrade success on all share spi devices (Mini Dock/TR_Gen2/Hyper). You can close issue now, thanks! I wonder what happened with 1.5.0?

>Then could you please release Ultra Dock to LVFS ?

As a test engineer in a different company from the ODMs that maintain Lenovo products; I am not familiar with LVFS ODMs' release schedule for firmware updates. Sorry.

hughsie, 2020-10-30:
> problem unborked

This was my fault, sorry. It would be good to get a device test (e.g. like https://github.com/fwupd/fwupd/blob/master/data/device-tests/devices/logitech-rqr12.json ) so that we don't accidentally cause a regression in the future -- all those tests are run on all devices before we tag a release. For this to be possible we need to public versions of firmware on the LVFS of different versions, so this might be one to think about later.

I'll tag 1.5.1 on Monday as there's one more regression we have to fix today.

closed 2020-10-30 by hughsie

memily, 2020-10-30:
> For this to be possible we need to public versions of firmware on the LVFS of different versions

Sure, how can we help? We can send you end product TR_Gen2 but we're not sure about firmware. These public versions are for test purposes correct? No end user will use them accidentally?

hughsie, 2020-10-30:
> These public versions are for test purposes correct

Yes, they'd be uploaded to the LVFS but remain in "private" state -- in theory users could download them as the checksum is "leaked" in the device test file, but in practice that's not super interesting. If they remain in private then they'll never be offered to users automatically.

hughsie, 2020-10-30:
@memily if you can upload both versions of the mini dock then I'd be happy to make you a template you can use for the other one.

memily, 2020-11-09:
>if you can upload both versions of the mini dock then I'd be happy to make you a template you can use for the other one. 

Thank you for waiting, we've sent the TR_Gen2 dock and I've uploaded *.cab files for Mini Dock and TR_Gen2 Dock onto LVFS. In regards to the template are there instructions on "lvfs readthedocs"?

We would also like to request opening a new VID 0x2D01 for HYPER on our lvfs account. Thanks

hughsie, 2020-11-09:
> Thank you for waiting, we've sent the TR_Gen2 dock and I've uploaded *.cab files for Mini Dock and TR_Gen2 Dock onto LVFS.

Great, thanks.

>  In regards to the template are there instructions on "lvfs readthedocs"?

No, although there probably should be some documentation somewhere... The easiest thing to do is look at the 40au0065 files in https://github.com/fwupd/fwupd/tree/master/data/device-tests/devices and copy them with the new SKU number. Then change the old and new filenames (not URLs!) to the Gen2 uploads, and then change the GUIDs and versions to thier expected versions. Then you can run `hardware.py devices/lenovo-WHATEVER*` and check all tiers pass.

> We would also like to request opening a new VID 0x2D01 for HYPER on our lvfs account. Thanks

That was easy! Done. :)

memily, 2020-11-10:
>although there probably should be some documentation somewhere... 

*Should be*, you say? Why let me get out my discerning eye, and my manager a pair from among his pile of glasses, to take a gander. Meanwhile, we've submitted tests for your pr approval; they pass on our local fwupd for the tr_gen2 and hyper demo devices.  The naming is odd, we're not sure when to include [VL817]...

hughsie, 2020-11-10:
>  The naming is odd, we're not sure when to include [VL817

It's up to you; it's whatever you'd like it to show on the console.