fwupd/fwupd #2526: ThinkPad Thunderbolt 3 Dock listed as "not authorized"

evgeni, 2020-10-29:
**Describe the bug**
context: https://twitter.com/zhenech/status/1321764709465493504

My Thinkpad Thunderbolt 3 Dock is listed as not authorized.

```
% boltctl 
 ● Lenovo ThinkPad Thunderbolt 3 Dock
   ├─ type:          peripheral
   ├─ name:          ThinkPad Thunderbolt 3 Dock
   ├─ vendor:        Lenovo
   ├─ uuid:          00682028-ca87-0801-ffff-ffffffffffff
   ├─ generation:    Thunderbolt 3
   ├─ status:        authorized
   │  ├─ domain:     c8010000-0082-8098-a045-660c30237120
   │  ├─ rx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  ├─ tx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  └─ authflags:  secure
   ├─ authorized:    Thu 29 Oct 2020 09:50:23 AM UTC
   ├─ connected:     Thu 29 Oct 2020 09:50:18 AM UTC
   └─ stored:        Wed 19 Sep 2018 08:55:34 AM UTC
      ├─ policy:     auto
      └─ key:        yes

% boltctl domains
 ● domain0 c8010000-0082-8098-a045-660c30237120
   ├─ bootacl:  1/16
   │   └─[0] 00682028-ca87-0801-ffff-ffffffffffff
   └─ security: secure
```

**Steps to Reproduce**
run `fwupdmgr get-devices`

**Expected behavior**
not have it list "Not authorized"

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.0

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf, Fedora 33

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20L8S2N80P
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:          973d32c4d1aa7e36b11d41702e8d6776ae13d633
│     Current version:    15.00
│     Vendor:             Lenovo (TBT:0x0108)
│     Update Error:       Not authorized
│     GUIDs:              fab08d10-6524-5cb6-ab20-6d365c812f9e
│                         0f511908-5c9f-511e-9099-06823ec33614
│                         df675f10-53ac-59ca-bddd-0a86ee492920
│     Device Flags:       • System requires external power source
│                         • Device stages updates
│   
[... 138 lines cut ...]
│     Bootloader Version: BOT03.01_B0008
│     Vendor:             USB:0x046D
│     Install Duration:   30 seconds
│     GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                         fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                         279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:       • Updatable
│                         • Supported on remote server
│   
└─VMM3320:
      Device ID:          d762543f8c20f636e6fff031a000078d3e10c600
      Summary:            Multi-Stream Transport Device
      Current version:    3.12.00
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:              5a9c2eec-d04c-5cec-aaaf-2043b6ea8605
                          e4acf3d9-790d-51a7-8284-b38a0ccb0e91
                          d6072785-6fc0-5f83-9d49-11376e7f48b1
      Device Flags:       • Updatable

```

**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? no
- Is this a regression? no

superm1, 2020-10-29:
I suspect what's happening here is that it wasn't authorized when fwupd started up `fu_thunderbolt_device_check_authorized` (returned false), and then later it got authorized but `fu_thunderbolt_device_check_authorized` didn't run again.  It's supposed to: https://github.com/fwupd/fwupd/blob/6450d0deb446380befa28795c52f5f0b628c4595/plugins/thunderbolt/fu-thunderbolt-device.c#L565 which is called from https://github.com/fwupd/fwupd/blob/0651aeef24a7330659febbf5e8fbc1b78283e7a2/libfwupdplugin/fu-udev-device.c#L79 on change events.

superm1, 2020-10-29:
If you `systemctl restart fwupd` presumably it sorts itself out?

evgeni, 2020-10-29:
Nope, remains unauthorized.

superm1, 2020-10-29:
Can you please check the sysfs file directly?  It should be `/sys/bus/thunderbolt/devices/*/authorized`.

evgeni, 2020-10-29:
```
/sys/bus/thunderbolt/devices/0-1/authorized:2
/sys/bus/thunderbolt/devices/0-1/device_name:ThinkPad Thunderbolt 3 Dock
```

superm1, 2020-10-29:
Ah, did you use a challenge for authorization?  I didn't realize that actually showed a different value for authorized attribute until now.
https://git.kernel.org/pub/scm/linux/kernel/git/westeri/thunderbolt.git/tree/drivers/thunderbolt/switch.c#n1374

evgeni, 2020-10-29:
*I* did not. At least not explicitly. It's just whatever the firmware and bolt agreed on based on the fact that the device is set to "secure".

PR #2527 (2020-10-29): thunderbolt: recognize authorized value of '2' as well
This means that the user performed a challenge to authorize the
device.
Fixes: #2526

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 3c4e5746 (2020-10-29): thunderbolt: recognize authorized value of '2' as well

This means that the user performed a challenge to authorize the
device.
Fixes: #2526
closed 2020-10-29 by superm1