fwupd/fwupd #2528: fwup should reside in it's own EFI directory on the efi partition

johannbg, 2020-10-29:
**Describe the bug**
Could be a strictly downstream issue anyway in Fedora the "Linux-Firmware-Updater" boot loader is placed under /EFI/fedora/fwupdx64.efi .

Given that fwupd is distribution agnostic it arguably should reside in it's own efi directory as in /efi/EFI/fwup/ which included the fw/ directory and relevant cab files ( /efi/EFI/fwup/fw/fwupd-$FOO.cab ) and the loader fwupdx64.efi ( /efi/EFI/fwup/fwupdx64.efi ) .

**Steps to Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

superm1, 2020-10-29:
A few problems with it going in it's own directory:
1. It needs shim to chainload in UEFI secure boot.  shim is signed by Microsoft and distributed by OS vendors and includes their certificate in the binary.
1. fwupd needs to be signed by a trusted authority for the OS.  As the fwupd binary distributed with fedora is signed by a different authority than the one distributed with Ubuntu this is not possible.

The part of the argument I can possibly see making sense is to put the cap files in their own directory, but even then they're cleaned up after a successful update.

johannbg, 2020-10-29:
Well downstream could just ship it with the package or signing it I mean fwup should work with the preloader, shim as well as those that want to manually sign the UEFI executable as in these three usecase right?

superm1, 2020-10-29:
But then what happens when you install Ubuntu and Fedora as a dual boot?  Who gets to own `fwupdx64.efi`?

Does shim allow using relative paths to load another binary?  I don't even know if it can do `../fwupd/fwupdx64.efi`.  Maybe it could.

superm1, 2020-10-29:
Now the thing that could flip all this on it's head is if we got _Microsoft_ to sign `fwupdx64.efi`.  And then all the distros distributed that instead of relying upon shim to load fwupd.

hughsie, 2020-10-30:
> is if we got Microsoft to sign fwupdx64.efi

I'd heavily NAK that, given the problems we have inside Red Hat for this kind of "just ship this unaltered binary" situation.  It's **super** painful for shim already.

superm1, 2020-10-30:
It would have to be some entity like LF doing it, and yeah if it's sure to be a painful process for distro pushing on it doesn't make sense to me either.

closed 2020-10-30 by superm1

johannbg, 2020-10-30:
@hughsie RH literally controls the shim-review board [1] for the UEFI signing process so is it not just painful because RH has made more painful then it has to be? 

@superm1 to me it makes very much sense that fwup as a project gets officially signed since it is an cross manufacture/distribution project and in the end of the day if the DE's are ever going to succeed in the linux ecosystem they will have to become a full blown OS as in Gnome will have to become GnomeOS and control it's development process from ground up ( from it's community vision to the hand of the end users ) and maintain an entire OS if they are ever going to succeed whether the Gnome community has realised that and admitted that fact or not. For example the fragmentation that has happened in Fedora will only result in conflict of interests which inevitably will hamper any progress that the "desktop" team/wg tries to do since the desktop team does not own the entire stack from ground up so it's decisions are bound and thus affected by action made by other WG right.

Anyway I digress from my pov it would be wise for the project to go through that painful process [2] of getting Microsoft to sign the fwupdx64.efi since in the long run it will pay off but granted I dont have any insight on how ye see the project in 5 - 10 years time and what it's aiming for in the long run.

1. https://github.com/rhboot/shim-review
2. https://techcommunity.microsoft.com/t5/hardware-dev-center/updated-uefi-signing-requirements/ba-p/1062916

superm1, 2020-10-30:
> to me it makes very much sense that fwup as a project gets officially signed since it is an cross manufacture/distribution project

I mean I don't disagree, from an upstream perspective we try to be as distro agnostic as possible.  You'll see tons of changes or in some places consolations recently in the codebase to make Linux based OSes able to use the project without forking for their needs.  We want to see the exact same core code running on Fedora, CoreOS, Ubuntu Core, ChromeOS and Ubuntu desktop and it's best for the ecosystem if we can keep that guarantee in place.  

If we're going to ever get to a place that the project distributes a signed binary, we need to have the major distro folks that sign it onboard with it.  And to onboard them I think we need to do soul searching.
Really what will it buy us that we can't accomplish with distros signing the EFI binary and shim loading it?  
Keep in mind we only install binary to the ESP when there is an update available.  So to me the only thing we get is that someone who dual boots multiple Linux OSes doesn't end up with a few extra megs on their ESP when they install a firmware update from one and then a firmware update from the other.

johannbg, 2020-10-30:
Well I think it also would play a big role establishing trust from hw vendors both to the project as well as any distro that backs it up so I would surprised if distribution would not jump on the chance of getting on board. 

Regarding the soul searching, providing end users with the ability of upgrading the firmware on their hardware in user friendly manner is one of the greater accomplishments in the linux ecosystem so dont sell yourselves and your hardwork short. And keep in mind downstream is profiting from your hardwork and your projects success so companies need you more than you ( should ) need them. Whoever is the employer, is lucky to have you.