fwupd/fwupd #2531: WARNING: UEFI ESP partition not detected or configured

kaihendry, 2020-10-30:
**Describe the bug**
WARNING: UEFI ESP partition not detected or configured

**Steps to Reproduce**

```
[root@t14s ~]# fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
```

**Expected behavior**

No warning

**fwupd version information**
Please provide the version of the daemon and client.
```shell
root@t14s ~]# fwupdmgr --version
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.0
```

Installed via yay upon Archlinux

https://s.natalian.org/2020-10-30/devices.txt

**System UEFI configuration**

https://s.natalian.org/2020-10-30/efibootmgr.txt

```shell
[root@t14s ~]# efivar -l | grep fw
[root@t14s ~]#
```

```shell
[root@t14s ~]# tree /boot/
/boot/
├── 6a097f4a4732426788a1061d488275a0
├── amd-ucode.img
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   ├── loader.conf
│   └── random-seed
└── vmlinuz-linux

7 directories, 9 files
```

**Additional questions**
- Operating system and version: Archlinux
- Have you tried rebooting? yes
- Is this a regression? don't know 
- Are you using an NVMe disk? yes 
- Is secure boot enabled? no

https://twitter.com/kaihendry/status/1322000584304586752

hughsie, 2020-10-30:
Hey @kaihendry -- sorry about that. Can you get us the output of `sudo /usr/bin/fwupdtool esp-list --verbose` please.

kaihendry, 2020-10-30:
```
[root@t14s ~]# /usr/bin/fwupdtool esp-list --verbose
13:34:50:0880 FuDebug              Verbose debugging enabled (on console 1)
GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UDisks2 was not provided by any .service files
[root@t14s ~]# logout
[hendry@t14s ~]$ sudo /usr/bin/fwupdtool esp-list --verbose
13:35:01:0611 FuDebug              Verbose debugging enabled (on console 1)
GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UDisks2 was not provided by any .service files
```

superm1, 2020-10-30:
I guess we need a new error condition specifically for udisks.  Since it's going to be used by swap plugin too soon, it should be a core requirement.

hughsie, 2020-10-30:
@kaihendry you need to install udisks as a dep to fwupd now. If you do that, does the ESP detection work?

> I guess we need a new error condition specifically for udisks

The alternate is we have a URL to a wiki page for each existing failure, e.g. https://github.com/fwupd/fwupd/wiki/FWUPD_PLUGIN_FLAG_ESP_NOT_FOUND and then list all the things that could cause the failure there?

superm1, 2020-10-30:
> @kaihendry you need to install udisks as a dep to fwupd now. If you do that, does the ESP detection work?
>
>
> The alternate is we have a URL to a wiki page for each existing failure, e.g. [...]

Sounds good to me.

toto-dev, 2020-11-01:
After upgrading form `1.4.6` to `1.5.0` I'm experiencing the same problem. I have udisks version `2.9.1` correctly installed so I don't think that is the issue.
The output that I get for `sudo /usr/bin/fwupdtool esp-list --verbose`:
```
14:51:19:0293 FuDebug              Verbose debugging enabled (on console 1)
14:51:19:0306 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p6, type: 0fc63daf-8483-4772-8e79-3d69d8477de4
14:51:19:0310 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: e3c9e316-0b5c-4db8-817d-f92df00215ae
14:51:19:0322 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
14:51:19:0326 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: de94bba4-06d1-4d40-a16a-bfd50179d6ac
14:51:19:0331 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p5, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b
14:51:19:0347 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b
Choose a volume:
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p5
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
```

The first one is the boot partition and the second one is the EFI partition (Shared with windows)

kaihendry, 2020-11-01:
extra/udisks2 right? Appears to have a crazy number of dependencies https://s.natalian.org/2020-11-02/1604247124_1918x1047.png

But indeed the warning goes away. I think I can live with the warning if I can avoid a bazillion dependencies.

```
[root@t14s ~]# /usr/bin/fwupdtool esp-list --verbose
16:12:42:0566 FuDebug              Verbose debugging enabled (on console 1)
16:12:42:0620 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4
16:12:42:0640 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/amd-ucode.img
/boot/vmlinuz-linux
/boot/initramfs-linux.img
/boot/EFI/systemd/systemd-bootx64.efi
/boot/EFI/BOOT/BOOTX64.EFI
/boot/loader/entries/arch.conf
/boot/loader/loader.conf
/boot/loader/random-seed
/boot/initramfs-linux-fallback.img
[root@t14s ~]# fwupdmgr update
Devices with no available firmware updates:
 • USB3.1 Hub
 • USB3.1 Hub
 • System Firmware
 • ThinkPad USB-C Dock Gen2 USB Audio
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB2.0 Hub
 • USB2.0 Hub
 • WDC PC SN730 SDBQNTY-512G-1001
No updatable devices
```

superm1, 2020-11-02:
If you want to avoid the dependencies, put your ESP into `/etc/fwupd/uefi.conf`.  But IIRC fwupd won't be able to mount it for your automatically anymore without udisks2.

kaihendry, 2020-11-02:
IIUC, I'm a little surprised it's so hard given the options above to figure out the EFI disk automatically. Shouldn't a Linux kernel or systemd bug be opened or something?

hughsie, 2020-11-02:
> Appears to have a crazy number of dependencies

Wow, which desktop distro *isn't* installing udisks by default? Does KDE or GNOME even work without UDisks?

> surprised it's so hard given the options above to figure out the EFI disk automatically

This is super complicated to get right. In some cases the user has multiple ESPs, and the ESP doesn't even have to be FAT32. Some distros mount it at startup, other don't. If we want to support the ones that don't (which seems to be ClearLinux, Fedora SB and CoreOS) then we need to be able to mount it at runtime; hence udisks.

I think it's right that we use udisks to discover information about disks; on normal general purpose "just works" desktop system it's being used already my nautilus etc, and in the minimal case or unusual (e.g. ChromeOS, or other embedded devices) it's perfectly correct to expect the user to set a line of configuration. Perhaps 1.5.0 was too unhelpful, but we merged a patch just now to show the user a wiki page which explains how to debug and set the configuration option in the gentoo/embedded case.

kaihendry, 2020-11-02:
https://dwm.suckless.org/ and anti-bloat campaigner here, so I do not agree on this decision _just to discover the ESP partition_.. 

Don't worry, I can live with the warning.

hughsie, 2020-11-02:
> and anti-bloat campaigner here

I don't think you can have "anti bloat" and "just works" at the same time.

> Don't worry, I can live with the warning.

You'll get no UEFI firmware updates...

kaihendry, 2020-11-02:
> You'll get no UEFI firmware updates...

Wouldn't

```
$ cat /etc/fwupd/uefi.conf
[uefi]
OverrideESPMountPoint=/boot
```

be sufficient?

hughsie, 2020-11-02:
> be sufficient?

If that's your ESP, yes, that would work too :)

kaihendry, 2020-11-02:
/boot is my EFI indeed, automounted IIUC by systemd. 
```

[hendry@t14s ~]$ mount | grep boot
systemd-1 on /boot type autofs (rw,relatime,fd=51,pgrp=1,timeout=120,minproto=5,maxproto=5,direct,pipe_ino=18214)
```

I would always assume /boot to be EFI, but I concede I'm not an expert.

hughsie, 2020-11-02:
> I would always assume /boot to be EFI

superm1, 2020-11-02:
OK so as concluded for anyone else who comes to this bug:

If you see this error follow the wiki page: https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found to come to next steps.

closed 2020-11-02 by superm1