fwupd/fwupd #2532: Failed to start Refresh fwupd metadata and update motd.

tsarajar, 2020-10-30:
Saw this in our syslog this morning on a host deployed by MAAS. It's a Ubuntu 20.04.
>Oct 30 06:43:12 ready-koi fwupdmgr[1110349]: Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Oct 30 06:43:12 ready-koi fwupdmgr[1110349]: Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
Oct 30 06:43:12 ready-koi systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 06:43:12 ready-koi systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
Oct 30 06:43:12 ready-koi systemd[1]: Failed to start Refresh fwupd metadata and update motd.
Oct 30 06:43:13 ready-koi systemd[1]: apt-daily-upgrade.service: Succeeded.

Alas, the info I can get comes here:

>root@ready-koi:/var/log/libvirt/qemu# service fwupd-refresh status
● fwupd-refresh.service - Refresh fwupd metadata and update motd
Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static; vendor preset: disabled)
Active: failed (Result: exit-code) since Fri 2020-10-30 06:43:12 UTC; 29min ago
TriggeredBy: ● fwupd-refresh.timer
Docs: man:fwupdmgr(1)
Process: 1110349 ExecStart=/usr/bin/fwupdmgr refresh --no-metadata-check (code=exited, status=1/FAILURE)
Main PID: 1110349 (code=exited, status=1/FAILURE)
>
>Oct 30 06:43:10 ready-koi systemd[1]: Starting Refresh fwupd metadata and update motd...
Oct 30 06:43:12 ready-koi fwupdmgr[1110349]: Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Oct 30 06:43:12 ready-koi fwupdmgr[1110349]: Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
Oct 30 06:43:12 ready-koi systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 06:43:12 ready-koi systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
Oct 30 06:43:12 ready-koi systemd[1]: Failed to start Refresh fwupd metadata and update motd.
root@ready-koi:/var/log/libvirt/qemu# fwupdmgr --version
client version: 1.3.11
compile-time dependency versions
gusb: 0.3.4
efivar: 37
daemon version: 1.3.11

List of devices:
>root@ready-koi:/var/log/libvirt/qemu# fwupdmgr get-devices --show-all-devices
PowerEdge R430
│
├─G200eR2:
│     Device ID:           d62bbbab5045e98c4f26c1dcac8b2ee96a3d90e4
│     Current version:     01
│     Vendor:              Matrox Electronics Systems Ltd. (PCI:0x102B)
│     GUIDs:               eea12a9d-1364-52a3-bd65-185aced495a3 ← PCI\VEN_102B&DEV_0534&REV_01
│                          831840e6-6886-5ef7-88cf-3f5db20e5e4c ← PCI\VEN_102B&DEV_0534
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│
├─NetXtreme II BCM57810 10 Gigabit Ethernet:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Current version:     10
│     Vendor:              Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:               e52860f5-7656-5c6d-8ffb-dc0fc963abe1 ← PCI\VEN_14E4&DEV_168E&REV_10
│                          5e757d7e-57f0-50f0-97d8-341257823517 ← PCI\VEN_14E4&DEV_168E
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│
├─NetXtreme II BCM57810 10 Gigabit Ethernet:
│     Device ID:           ed6eceeea5a37ef9446ef275eff306d30e0401bd
│     Current version:     10
│     Vendor:              Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:               e52860f5-7656-5c6d-8ffb-dc0fc963abe1 ← PCI\VEN_14E4&DEV_168E&REV_10
│                          5e757d7e-57f0-50f0-97d8-341257823517 ← PCI\VEN_14E4&DEV_168E
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│
└─System Firmware:
      Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
      Current version:     2.11.0
      Vendor:              Dell Inc.
      Update Error:        UEFI Capsule updates not available or enabled
      GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
      Device Flags:        • Internal device
                           • Requires AC power
                           • Needs a reboot after installation

SeaMonkey82, 2020-11-29:
I'm also seeing this error in Linux Mint 20.  I believe that I previously installed fwupd using snap prior to upgrading from Linux Mint 19.3.  On another machine running Linux Mint 20 where fwupd was never manually installed, the service completes successfully.

Perkolator, 2020-12-28:
Also seeing this on Mint 20. Clean Mint install, no upgrades. Just starting out my linux journey and still learning. How severe is this problem?

superm1, 2021-01-11:
There's also a report at https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1909603 showing the same thing on Ubuntu 20.04.

hughsie, 2021-01-11:
A network error perhaps? Can we make fwupd-refresh.service return the verbose output somehow?

saschaarthur, 2021-01-12:
No thats the same issue as in #391 and in #2708 

Heres also a screenshot as proof: 

[image]

Im fighting this issue now on a huge amount of ubuntu 20.04 LTS servers. 

To be honest i really dont understand as discussed in #391 why exactly the daemon should have a huge update (1.4 vs 1.3) with a bunchload of new libs when the CDN settings for this files firmware / signature could just be adjusted. A simple payload in the url would also prob. do the trick.. 

Also I dont get exactly why the developers here, close/mute topics about it. Even if theres a huge amount of affected LTS systems (mint, debian, ubuntu) and give answers like "theres no workaround, go for an non LTS system, or retry in a couple of hours..".

Its crashing systemd units in LTS versions, and its a shame that an LTS system is not getting a proper update for this. 

I dont want to deal further with it, Im just removing this daemon on all my servers.

superm1, 2021-01-12:
>Also I dont get exactly why the developers here, close/mute topics about it. Even if theres a huge amount of affected LTS systems (mint, debian, ubuntu) and give answers like "theres no workaround, go for an non LTS system, or retry in a couple of hours..".

FYI it's in the order of minutes to retry, not hours.
The reason for locking discussion on those two issues was that they were very noisy and the root cause was getting lost.  The fix for them is well understood, needing to update to the 1.4.x release.  As for the reason such a big update is needed is because the root cause of the issue is that CDN mirroring order can't be controlled.

> To be honest i really dont understand as discussed in #391 why exactly the daemon should have a huge update (1.4 vs 1.3) with a bunchload of new libs when the CDN settings for this files firmware / signature could be adjusted. A simple payload in the url would also prob. do the trick..

The filenames used in 1.3.x were static values and so if signing and CDN mirroring got out of sync you had the issues reported.  To solve this, the software was reconfigured to grab the small signature first and if necessary grabbing a URL with the build suffix applied to it.  This prevented the impossibility of CDN getting out of sync.  The actual code changes are here:
https://github.com/fwupd/fwupd/pull/1950
https://github.com/fwupd/lvfs-website-archived/pull/569

>Its crashing systemd units in LTS versions, and its a shame that an LTS system is not getting a proper update for this.

I don't think we have proof that is the same root cause as `fwupd-refresh.service` failing, but thinking more about it that does seem like a very convincing hypothesis.

>A network error perhaps? Can we make fwupd-refresh.service return the verbose output somehow?

We can't easily add `--verbose` to everyone's `fwupd-refresh.service` at enough of a scale that someone who gets affected will show us the output to confirm it.  But I wonder if we can contrive something like a VM or container refreshing every 5 minutes with `--force` for 24 hours to see if that can hit it and confirm?

saschaarthur, 2021-01-12:
> FYI it's in the order of minutes to retry, not hours.
> The reason for locking discussion on those two issues was that they were very noisy and the root cause was getting lost. The fix for them is [...]

Having 4 servers which are affected since more then 7 hours. CDNs are made for storing static files for days, weeks or even years.  
Since you use (or at least you did at some point) AWS, the default caching time is at least 24 hours (https://aws.amazon.com/premiumsupport/knowledge-center/cloudfront-serving-outdated-content-s3/). So its definitely not minutes.

> 
> > To be honest i really dont understand as discussed in #391 why exactly the daemon should have a huge update (1.4 vs 1.3) with a bunchload of new libs when the CDN settings for this files firmware / signature could be adjusted. A simple payload in the url would also prob. do the trick..
> 
> The filenames used in 1.3.x were static values and so if signing and CDN mirroring got out of sync you had the issues reported. To solve this, the software was reconfigured to grab the small signature first and if necessary grabbing a URL with the build suffix applied to it. This prevented the impossibility of CDN getting out of sync. The actual code changes are here:
> #1950
> [fwupd/lvfs-website-archived#569](https://github.com/fwupd/lvfs-website-archived/pull/569)

Thats fine and should be like that. Files which are somewhat dynamic and released to CDNs should always contain a version suffix either in the filename itself or url param when you request those, everything else will lead to huge CDN issues like we have here. It could even happen that you receive the 'old files' in the next request again..

The only way to solve those different versions, is to force refresh the CDN-cache (invalidate the cache keys) (https://docs.aws.amazon.com/AmazonCloudFront/latest/DeveloperGuide/Invalidation.html). 

This only solves the cause, not the design issue. 

From reading the code i dont see why it should fix the problem with

https://cdn.fwupd.org/downloads/firmware.xml.gz
and its signature
https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Its only fixing the fw downloads (and its metadata) itself, but not the xml (index) download. Maybe i miss something here?

> > Its crashing systemd units in LTS versions, and its a shame that an LTS system is not getting a proper update for this.
> 
> I don't think we have proof that is the same root cause as `fwupd-refresh.service` failing, but thinking more about it that does seem like a very convincing hypothesis.

The proof that this is the cause of the failing service, i posted as a picture above. Thats btw also the reason why im here. Our monitoring complained about this systemd service.

hughsie, 2021-01-13:
> Files which are somewhat dynamic and released to CDNs should always contain a version suffix

So we did that, almost one year ago; we dedicated weeks of work to fix it properly. I think the frustration you have should be directed at your LTS distro, not upstream. Being honest, your replies are also coming across very condescending when in reality you seem to know very little about the scalability or resource constraints we're working with. Mario is being much nicer than I would have been. If you can reproduce this bug with fwupd versions 1.4.* or 1.5.* please let us know. Please also continue discussion about obsolete versions of fwupd on Launchpad, not here.

superm1, 2021-01-14:
(I've added some color into the Ubuntu bug, and since we understand the root cause and that it's fixed in 1.4.x+ I'm going to close this issue)

closed 2021-01-14 by superm1

equaeghe, 2021-03-15:
>  If you can reproduce this bug with fwupd versions 1.4.* or 1.5.* please let us know.

I have fwupd 1.4.6 installed on Gentoo and seem to be hitting this (or a similar) issue:

```
Mar 15 23:09:23 … systemd[1]: Starting Refresh fwupd metadata and update motd...
░░ Subject: A start job for unit fwupd-refresh.service has begun execution
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ A start job for unit fwupd-refresh.service has begun execution.
░░ 
░░ The job identifier is 8381.
Mar 15 23:09:23 … systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ An ExecStart= process belonging to unit fwupd-refresh.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Mar 15 23:09:23 … systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ The unit fwupd-refresh.service has entered the 'failed' state with result 'exit-code'.
Mar 15 23:09:23 … systemd[1]: Failed to start Refresh fwupd metadata and update motd.
░░ Subject: A start job for unit fwupd-refresh.service has failed
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ A start job for unit fwupd-refresh.service has finished with a failure.
░░ 
░░ The job identifier is 8381 and the job result is failed.
```

superm1, 2021-03-17:
> I have fwupd 1.4.6 installed on Gentoo and seem to be hitting this (or a similar) issue:
>
> ```
> Mar 15 23:09:23 … systemd[1]: Starting Refresh fwupd metadata and update motd...
> ░░ Subject: A start job [...]

Can you please open a separate issue to debug it and add everything in the bug template?

brainchild0, 2022-09-30:
I observe the problem on two systems currently running Linux Mint 21, based on Ubuntu 22.04 (jammy). The systems have no particular commonality in software or hardware other than both being x86 64-bit machines running the same distribution. All software is up to date from distribution repositories. The fwupd version is 1.7.5

```
$ systemctl status fwupd-refresh.service
× fwupd-refresh.service - Refresh fwupd metadata and update motd
     Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static)
     Active: failed (Result: exit-code) since Fri 2022-09-30 00:07:46 EDT; 3h 5>
TriggeredBy: ● fwupd-refresh.timer
       Docs: man:fwupdmgr(1)
    Process: 2719253 ExecStart=/usr/bin/fwupdmgr refresh (code=exited, status=1>
   Main PID: 2719253 (code=exited, status=1/FAILURE)
        CPU: 45ms

Sep 30 00:07:46 epl-envy-mint systemd[1]: Starting Refresh fwupd metadata and u>
Sep 30 00:07:46 epl-envy-mint systemd[1]: fwupd-refresh.service: Main process e>
Sep 30 00:07:46 epl-envy-mint systemd[1]: fwupd-refresh.service: Failed with re>
Sep 30 00:07:46 epl-envy-mint systemd[1]: Failed to start Refresh fwupd metadat>
```

```
$ sudo fwupdmgr refresh      
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 1 local device supported
$ echo $?
0
```

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.0-051900-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

superm1, 2022-09-30:
> I observe the problem on two systems currently running Linux Mint 21, based on Ubuntu 22.04 (jammy). The systems have no particular commonality in software or hardware other than both being x86 [...]

Please see https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1909603