fwupd/fwupd #2537: fwupdmgr security level: detect zram

edwintorok, 2020-10-31:
**Describe the bug**
fwupdmgr security --force shows 'Linux swap: unencrypted' as Red, which is wrong.
Fedora 33 uses [zram as swap](https://fedoraproject.org/wiki/Changes/SwapOnZRAM), hence encrypting swap in this case would be pointless as its contents doesn't leave RAM.

**Steps to Reproduce**
On Fedora 33 run:
```
$ sudo fwupdmgr security --force
Host Security ID: HSI:2+! (v1.5.0)

HSI-1
✔ TPM v2.0:                      Found

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ fwupd plugins:                 Untainted
✘ Linux swap:                    Unencrypted

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues

$ sudo swapon -s
Filename				Type		Size	Used	Priority
/dev/zram0                             	partition	4194300	0	100
```

**Expected behavior**
If all swap devices are zram (or if there are no swap devices) then this should be treated as green,
and perhaps with a different text than unecrypted, e.g. say "Compressed RAM only"

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf install

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
System Product Name
│
├─AMD Ryzen 9 3900X 12-Core Processor            :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17
[... 130 lines cut ...]
│ │   Device Flags:       • Internal device
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:     • Internal device
│     
└─WDC WD7501AALS-00J7B0:
      Device ID:          1cb4548db4ea4d3a939a960a602d905d8f035e7a
      Summary:            ATA Drive
      Current version:    05.00K05
      Vendor:             Western Digital (ATA:0x101C)
      Serial Number:      WD-WMATV0156388
      GUIDs:              34d34cc5-5ad6-5ec9-9adb-8e8c253d0a61 ← IDE\WDC_WD7501AALS-00J7B0___________________05.00K05
                          29290fa4-061c-577a-94d0-4ff7bf340567 ← IDE\0WDC_WD7501AALS-00J7B0___________________
                          33d7d102-aa9b-5013-9d29-bb028a8f2d1e ← WDC WD7501AALS-00J7B0
      Device Flags:       • Updatable
                          • System requires external power source
                          • Needs a reboot after installatio
```

**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? yes
- Is this a regression? no

edwintorok, 2020-10-31:
https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues should also be updated to point to the zram-as-swap as an alternative to encrypting swap.
(FWIW I've also converted a Ubuntu workstation to zram-on-swap and it has visibly improved system responsiveness  when using Firefox on sites that eat up massive amounts of RAM - such as leaving a Jenkins tab open. It is better than turning off swap, which would cause pagecache to be dropped when under memory pressure, which would lead to page-cache thrashing.
Unless you want to hibernate, or have very limited amounts of RAM I don't think disk based swap is needed/useful).

hughsie, 2020-10-31:
I think https://github.com/fwupd/fwupd/pull/2529 addresses the zram case, no?

edwintorok, 2020-10-31:
Thanks, it looks like the right approach (I haven't tried compiling it yet, but looking for `/dev/zram` as prefix should catch this, not sure how it'll be reported, reporting zram as encrypted isn't quite accurate, might be nicer to be a tri-state: encrypted, not encrypted, and not-applicable/internal/memory-only for zram?)

edwintorok, 2020-11-04:
fwupd-1.5.1-1.fc33.x86_64 now shows this, with only HSI-3 and -U and -A being red:
```
Host Security ID: HSI:2 (v1.5.1)

HSI-1
✔ TPM v2.0:                      Found

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
```

closed 2020-11-04 by edwintorok

dnoliver, 2021-02-25:
I think I am running into this reported issue. 

I am using the same swap encryption configuration since F31, and before the firmware update manager reported my swap as encrypted. I mostly took the instructions from here https://wiki.archlinux.org/index.php/Dm-crypt/Swap_encryption

Versions are (this is Fedora IoT Edition):

```shell
[root@fitlet-1 ~]# cat /etc/redhat-release
Fedora release 33 (Thirty Three)

[root@fitlet-1 ~]# rpm -qa fwupd
fwupd-1.5.5-1.fc33.x86_64
```

Security report say that swap is not encrypted.

```shell
[root@fitlet-1 ~]# fwupdmgr security --force
Unknown                  [***************************************]
Host Security ID: HSI:2! (v1.5.5)

HSI-1
? Intel DCI debugger:            Disabled
? SPI BIOS region:               Locked
? SPI lock:                      Enabled
? SPI write:                     Disabled
? TPM v2.0:                      Found
? TXE manufacturing mode:        Locked
? TXE override:                  Locked
? TXE v0:3.1.70.2334:            Valid
? UEFI platform key:             Valid

HSI-2
? IOMMU:                         Enabled
? Intel DCI debugger:            Locked
? TPM PCR0 reconstruction:       Valid

HSI-3
? Suspend-to-ram:                Disabled
? Intel CET Enabled:             Not supported
? Pre-boot DMA protection:       Disabled
? Suspend-to-idle:               Disabled

HSI-4
? Intel SMAP:                    Enabled
? Encrypted RAM:                 Not supported

Runtime Suffix -!
? Linux kernel:                  Untainted
? Linux kernel lockdown:         Enabled
? fwupd plugins:                 Untainted
? Linux swap:                    Unencrypted

This system has HSI runtime issues.
 ? https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```

My swap state is the following:

```shell
[root@fitlet-1 ~]# swapon -s
Filename                                Type            Size    Used    Priority
/dev/zram0                              partition       4194300 0       100
/dev/dm-1                               partition       12464124        0       -2

[root@fitlet-1 ~]# lsblk
NAME                     MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
sda                        8:0    0 238.5G  0 disk
|-sda1                     8:1    0   200M  0 part  /boot/efi
|-sda2                     8:2    0   512M  0 part  /boot
`-sda3                     8:3    0 237.8G  0 part
  |-system-root          253:0    0  47.6G  0 lvm   /sysroot
  |-system-swap          253:1    0  11.9G  0 lvm   [SWAP]
  |-system-home          253:2    0  11.9G  0 lvm
  | `-luks-system-home   253:5    0  11.9G  0 crypt /var/home
  |-system-opt           253:3    0  47.6G  0 lvm
  | `-luks-system-opt    253:7    0  47.5G  0 crypt /var/opt
  `-system-docker        253:4    0 118.9G  0 lvm
    `-luks-system-docker 253:6    0 118.9G  0 crypt /var/lib/docker
zram0                    252:0    0     4G  0 disk  [SWAP]
```

So, with my configuration, I end up with 2 swap devices, which may be a problem.

But I guess the zram should pass this test based on the report from @edwintorok, an my system-swap should pass this test because it is encrypted. 

Should I file a bug for this?
Is my swap configuration wrong, and I should rather deactivate zram-swap and leave system-swap, or the other way around?

hughsie, 2021-02-25:
> Is my swap configuration wrong

You only get the checkmark for encrypted swap if they're **all** encrypted.

dnoliver, 2021-03-01:
I tested a little bit more, by disabling and enabling swap-on-zram and swap-on-disk, and running the security test after it.

**swap-on-zram** is identified as "encrypted" (not actually encrypted, but still means we do not dump secrets to disk)
**swap-on-disk** is identified as "not encrypted". This is weird, because I use the same swap encryption configuration since always, I did not found any new instructions saying that my config is no longer valid, `swapon` say I have a random `/dev/dm-X` device, but `lsblk` says my system-swap is not "crypt". Is it possible that the new detection logic based on `udisk2` thinks my swap is plain when it is not? also, it is possible that the old detection logic was identifying my swap as encrypted, but it was actually plain? 

Anyways, I think **swap-in-zram** is the way to go for me, and the default in Fedora IoT, so there is less configuration to do.