fwupd/fwupd #2540: How do I update my Lenovo ThinkPad USB-C Dock Gen2 USB Audio dock?

kaihendry, 2020-11-02:
I want to update **ThinkPad USB-C Dock Gen2 USB Audio** and also query the existing version in order to respond to https://twitter.com/LenovoSupport/status/1323095074545242113

I'm not sure if https://download.lenovo.com/consumer/options/thinkpad_usb-c_dock_gen2_fw_v10606161.txt addresses my [restarting dock issues](https://dabase.com/blog/2020/Thinkpad_Gen2_Dock/), but I'm hoping the [July 2020 update](https://support.lenovo.com/sg/en/downloads/ds539092-firmware-for-windows-10-7-32-bit-64-bit-thinkpad-usb-c-dock-gen-2?Source=social_eservices&Medium=TWITTER&Term=eServices_&Content=20201102_20201102025022&linkId=103406460) will!

hughsie, 2020-11-02:
The cxaudio device is supported in 1.5.0 and newer, although we're waiting for final validation of the firmware from the ODM. If you manage to obtain the SREC format it can be applied using `fwupdtool install-blob` -- otherwise you'll have to wait for the ODM to finish validation I'm afraid.

kaihendry, 2020-11-02:
**cxaudio**? I'm not sure how that's related to my dock. :sweat: 

Is this the reason for the restart loop I'm suffering with?

hughsie, 2020-11-02:
Ohh, I thought that ThinkPad USB-C Dock Gen2 USB **Audio** meant that specifically you wanted to update the audio chip (which is made by Connexant, now Synaptics). I don't know anything about what firmware updates are supposed to fix; those ACPI kernel warnings doesn't look much like dock stuff to me. I'd err on trying an up to date mainstream kernel like Fedora/Ubuntu and see if the restart issue happens there.

As for a version; a dock is 6-12 different USB devices in a trenchcoat; there isn't one version number you could construct for that.

kaihendry, 2020-11-02:
> As for a version; a dock is 6-12 different USB devices in a trenchcoat; there isn't one version number you could construct for that.

haha, ok. Well, I need a reliable dock for working, so this issue is sucking at my will hard. If I could be informed about a more reliable one, I would purchase it.

hughsie, 2020-11-02:
> Well, I need a reliable dock for working

Give me two weeks; the firmware updates are nearly ready now.

superm1, 2020-11-02:
> As for a version; a dock is 6-12 different USB devices in a trenchcoat; there isn't one version number you could construct for that.

FYI - that's a statement for Lenovo's and all vendors in the Lenovo docks.  For Dell's WD19* docks there is a "package" version.

closed 2020-12-10 by stale[bot]

ivankovnatsky, 2021-04-24:
probably not stale.

ivankovnatsky, 2021-04-24:
/open

seltzered, 2021-07-31:
FYI, looks like there's an update by lenovo (v1.08, April 2021) for the dock https://support.lenovo.com/us/en/downloads/ds539092-firmware-for-windows-10-7-32-bit-64-bit-thinkpad-usb-c-dock-gen-2  that wasn't added by the vendor into LVFS (tried searching: https://fwupd.org/lvfs/search?value=Lenovo+ThinkPad+USB-C+Dock+Gen2 )

superm1, 2023-04-20:
What for? What's the question.

bhack, 2023-04-20:
Is there any plan to support the firmware update of this device?

superm1, 2023-04-20:
That's not a question for fwupd developers, you should ask the manufacturer.