fwupd/fwupd #2542: Daemon has loaded 3rd party code

exploide, 2020-11-03:
**Describe the bug**

Just updated to `fwupd-1.5.1-1.fc33.x86_64` from Fedora's `updates-testing` repository. Now I receive the following message when running any fwupd command:

```
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
```

**Steps to Reproduce**

- `sudo dnf upgrade --enablerepo=updates-testing fwupd`
- `fwupdmgr update`

**Expected behavior**

Since I did not add any 3rd party code, I did not expect to see this message.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
20N2CTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          934297aea936899e588ad4bf075cc02ee5b1272b
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              dd092983-daaf-587a-9ed7-dd8d178adbe7
│                         abc4cc28-099d-5220-b45d-129c21cd9da7
│                         52265728-359a-574c-9a6a-a23d3b1f952d
│                         f117e89e-a75f-5f33-9e8e-c4aded9cbadf
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:    0.1.19
│     Minimum Version:    0.1.19
│     Vendor:             DMI:LENOVO
│     GUIDs:              38ea6335-29ca-417b-8cd4-6b4e5e866f92
│                         78c6acda-f389-5111-839d-7f258cef874b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          b1c32d855d85a2124e865f0a03f8c2384dcca167
│     Current version:    192.47.1524
[... 112 lines cut ...]
│     Current version:    26403
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     GUIDs:              26aac102-e03d-4eb0-a663-9372614a765a
│                         2fba53b5-bcba-5f5c-bb3c-e9612696e0d9
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UHD Graphics 620 (Whiskey Lake):
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              3c99f0ab-4e4e-5df2-80f1-843e5b8399ff
                          5461564d-d717-53e3-95c6-706973f9245d
                          1bb467a3-4af9-53a9-ab48-f588ee6e8066
                          52aa8ac7-c837-561f-af32-b7979cd2f6ed
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
```

**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? Yes

superm1, 2020-11-03:
Can you run `fwupdtool --verbose get-devices` and we can see what it shows as the 3rd party code?

exploide, 2020-11-03:
> Can you run fwupdtool --verbose get-devices and we can see what it shows as the 3rd party code?

Sure, here it is:

<details>
<summary>fwupdtool --verbose get-devices</summary>

```
# fwupdtool --verbose get-devices
17:21:16:0648 FuDebug              Verbose debugging enabled (on console 1)
17:21:16:0651 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
17:21:16:0651 FuConfig             loading config values from /etc/fwupd/daemon.conf
17:21:16:0652 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
17:21:16:0653 FuRemoteList         loading /usr/share/fwupd/metainfo
17:21:16:0653 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
17:21:16:0653 XbSilo               file: b340f2fe-bbe0-7629-b66d-a4540a2bf65d, current:b340f2fe-bbe0-7629-b66d-a4540a2bf65d, cached: (null)
17:21:16:0653 XbSilo               loading silo with file contents
17:21:16:0653 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
17:21:16:0653 Fwupd                converting Keyring value to Jcat
17:21:16:0653 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
17:21:16:0653 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
17:21:16:0653 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
17:21:16:0653 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
17:21:16:0653 FuRemoteList         ordering lvfs-testing=lvfs+1
17:21:16:0653 FuRemoteList         ignoring unfound remote fwupd
17:21:16:0653 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
17:21:16:0653 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
17:21:16:0654 Jcat                 loaded 1 certificates
17:21:16:0654 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
17:21:16:0654 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
17:21:16:0654 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
17:21:16:0654 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
17:21:16:0654 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
17:21:16:0654 Jcat                 loaded 1 certificates
17:21:16:0654 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
17:21:16:0654 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
17:21:16:0654 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
17:21:16:0654 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
17:21:16:0660 FuEngine             client certificate exists and working
17:21:16:0661 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
17:21:16:0661 FuHistory            got schema version of 6
17:21:16:0661 FuHwids              smbios property Manufacturer=LENOVO
17:21:16:0661 FuHwids              smbios property EnclosureKind=a
17:21:16:0661 FuHwids              smbios property Family=ThinkPad T490
17:21:16:0661 FuHwids              smbios property ProductName=20N2CTO1WW
17:21:16:0661 FuHwids              smbios property ProductSku=LENOVO_MT_20N2_BU_Think_FM_ThinkPad T490
17:21:16:0661 FuHwids              smbios property BiosVendor=LENOVO
17:21:16:0661 FuHwids              smbios property BiosVersion=N2IET91W (1.69 )
17:21:16:0661 FuHwids              smbios property BiosMajorRelease=01
17:21:16:0661 FuHwids              smbios property BiosMinorRelease=45
17:21:16:0661 FuHwids              smbios property BaseboardManufacturer=LENOVO
17:21:16:0661 FuHwids              smbios property BaseboardProduct=20N2CTO1WW
17:21:16:0664 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
17:21:16:0664 XbSilo               file: 758538d3-bdd0-00df-d738-96efcfbc8908, current:758538d3-bdd0-00df-d738-96efcfbc8908, cached: (null)
17:21:16:0664 XbSilo               loading silo with file contents
17:21:16:0665 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
17:21:16:0665 XbSilo               file: 4d64cb7b-e2f4-f1c7-f83a-814b151d9b13, current:4d64cb7b-e2f4-f1c7-f83a-814b151d9b13, cached: (null)
17:21:16:0665 XbSilo               loading silo with file contents
17:21:16:0665 FuEngine             486 components now in silo
17:21:16:0669 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_logitech_hidpp.so)
17:21:16:0669 FuPlugin             added udev subsystem watch of hidraw
17:21:16:0670 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_uefi.so)
17:21:16:0671 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_nitrokey.so)
17:21:16:0671 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_redfish.so)
17:21:16:0671 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_wacom_raw.so)
17:21:16:0671 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_elantp.so)
17:21:16:0671 FuPlugin             added udev subsystem watch of i2c-dev
17:21:16:0672 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_coreboot.so)
17:21:16:0672 FuPlugin             init(/usr/lib64/fwupd-plugins-3/libfu_plugin_tpm.so)
[... cut ...]
17:21:17:0583 FuPlugin             destroy(uefi)
17:21:17:0584 FuPlugin             destroy(modem_manager)
17:21:17:0586 FuPlugin             destroy(linux_lockdown)
17:21:17:0586 FuPlugin             destroy(tpm_eventlog)
17:21:17:0587 FuPlugin             destroy(linux_swap)
17:21:17:0587 FuPlugin             destroy(dell)
17:21:17:0588 FuPlugin             destroy(flashrom)
17:21:17:0588 FuPlugin             destroy(linux_tainted)
17:21:17:0589 FuPlugin             destroy(redfish)
```

</details>

superm1, 2020-11-03:
Looks like it's these two plugins:

```
17:21:16:0679 FuEngine             modem_manager has incorrect built version 79c26d31c1eb533fd0b764e3451b8d8cb4954f6839454cf956a7983104786fee
17:21:16:0681 FuEngine             flashrom has incorrect built version 79c26d31c1eb533fd0b764e3451b8d8cb4954f6839454cf956a7983104786fee
```

They're supposed to come as recommends: https://github.com/fwupd/fwupd/blob/master/contrib/fwupd.spec.in#L147

@hughsie packaging bug?

exploide, 2020-11-03:
Oh, these are separate packages now. I needed to do `sudo dnf upgrade --enablerepo=updates-testing fwupd\*` with the wildcard in the end to also get these from `updates-testing`. Then the issue is gone.

So probably not a bug, just me installing only half of the updates. On the other hand, shouldn't the rpm have the correct dependencies? Not sure. Feel free to close this if nothing should be changed for packaging.

hughsie, 2020-11-03:
> shouldn't the rpm have the correct dependencies

Sure, they're `Recommends`. It's not a hard error if you chose to not update (or install) the new updates, unlike a `Depends` would be.

exploide, 2020-11-03:
I see. Thanks for clarifying.

closed 2020-11-03 by exploide