fwupd/fwupd #2543: Probably should de-emphasize snap

hughsie, 2020-11-03:
At the moment looking at the README the only installation method we suggest is Snap. This makes me feel somewhat uneasy, and has made people make more than a few people think that fwupd depends on Snap :(

[image]
...from https://www.theregister.com/2020/11/02/linux_mint_chromium_build/

I think my general thought is that I don't think end users should ever compile fwupd from scratch; it's a complicated project with dozens of dependencies (and as many configuration options) and there's just too many things that can go wrong. Users should just have fwupd installed by their distro, hopefully by a packager who does understand all the config options. I don't think offering it as a snap absolves the downstream of doing the right thing and I **certainly** don't want to see a snap-only fwupd install method like Canonical is currently doing with Chromium.

My personal opinion is you're not supposed to replace low level parts of your OS yourself. If you install fwupd 1.5.1 on your Ubuntu LTS it'll probably install fine, but we've done exactly zero QA on that as a configuration and the software store frontend might work fine, or might explode in a ball of flames. The snap or flatpak version might be useful to update a specific device on the command line that needs a bleeding edge fwupd version (because the specific device support is so new) but I really don't think using the snap or flatpak should ever be seen as a replacement to the system version.

superm1, 2020-11-03:
>At the moment looking at the README the only installation method we suggest is Snap. This makes me feel somewhat uneasy, and has made people make more than a few people think that fwupd depends on Snap :(

Huh, I guess I haven't actually _read_ it in a while.

> certainly don't want to see a snap-only fwupd install method like Canonical is currently doing with Chromium.

Me either.

>I think my general thought is that I don't think end users should ever compile fwupd from scratch; it's a complicated project with dozens of dependencies (and as many configuration options) and there's just too many things that can go wrong. 

Yeah in general most people shouldn't be compiling.

>The snap or flatpak version might be useful to update a specific device on the command line that needs a bleeding edge fwupd version (because the specific device support is so new)

Now I've found snap is incredibly useful for ODMs to use, particularly because OSVs don't roll out newer versions in LTS release.

>but I really don't think using the snap or flatpak should ever be seen as a replacement to the system version.

The `confined` snap I completely agree, but even the `classic` snap?  The classic snap is essentially the same thing as compiling all the deps by hand, throwing it in a squashfs, mounting the squashfs and then running some postinst hooks to connect it to PK.  There isn't a lot of opportunity for it to go wrong.

I think we can still message it that the best version to use is the version provided by your distro, unless you "know" you need a newer version for update XYZ.

hughsie, 2020-11-03:
>  The classic snap is essentially the same thing as compiling all the deps by hand

Agreed, but it also doesn't make it magically QAd. e.g. we don't check the runtime systemd version, only the build time version. I guess my issue here is that the flatpak doesn't pretend to be "_as good as the distro version_" but the snap one does; and when it breaks it's fwupd that looks bad, not the distro.

superm1, 2020-11-03:
The systemd confinement stuff is different upstream than the generated unit, this is true.  So our stuff that adds systemd rules won't work right.

>and when it breaks it's fwupd that looks bad, not the distro.

Hmm.  Maybe we need to add code explicitly warning or telling people this somewhere in the snap build.

hughsie, 2020-11-03:
> Now I've found snap is incredibly useful for ODMs to use

If you can say, does Dell specify an OS and version for the ODMs to test against? I know for the ODMs of Lenovo I've basically been refusing to offer plugin development support unless they're running the latest Fedora or Ubuntu. I do COPR builds with the git master version of fwupd which I know a few ODMs use.

hughsie, 2020-11-03:
> we need to add code explicitly warning or telling people this somewhere

I don't think this is a FwupdPluginFlag, but I think it's the same kind of thing -- maybe we "taint" the daemon like we do for 3rd party plugins?

superm1, 2020-11-03:
> If you can say, does Dell specify an OS and version for the ODMs to test against?

"The same OS that they'll be shipping systems with".

Which is why it's quite convenient to use snap for them.  It removes any packager backport mistake element from the equation and the snap works the same on any underlying OS.

> I don't think this is a FwupdPluginFlag, but I think it's the same kind of thing -- maybe we "taint" the daemon like we do for 3rd party plugins?

I think it needs to be a different kind of taint if so.

closed 2020-11-03 by hughsie