fwupd/fwupd #2545: Make zsh autocompletion a thing

lhirlimann, 2020-11-04:
make fwupdmgr copmletion for options available.
It works in bash

eg I type sudo fwupdmgr and then tab and nothing is show under zsh , while options are provided for bash.

hughsie, 2020-11-04:
I don't think any of us use zsh I'm afraid. If you want to contribute the missing support I'd be happy to review a PR.

mavjs, 2020-11-23:
There seems to be a @zsh-users project that according to their README.md

> This projects aims at gathering/developing new completion scripts that are not available in Zsh yet. The scripts may be contributed to the Zsh project when stable enough.

There already exists a zsh-completions for `fwupdmgr` - https://github.com/zsh-users/zsh-completions/blob/master/src/_fwupdmgr

Perhaps @nicoulaj - the author of `_fwupdmgr` could send a pull request?

nicoulaj, 2020-12-28:
I could do this but if there is no one to maintain it in this project, I think it will be better maintained at zsh-users/zsh-completions.

hughsie, 2021-01-10:
> I think it will be better maintained at zsh-users/zsh-completions.

Strongly disagree; we add commands pretty frequently and we manage to update the "fish" completion list even though none of us use it!

superm1, 2021-01-11:
Completely agree with @hughsie.  Submit the zsh and as long as it's obvious how to amend it we can rev it with bash and fish completion.  If it's a complex new command we can loop in the submitter to help.

hughsie, 2021-01-19:
Closing for now, please open a PR if you'd like.

closed 2021-01-19 by hughsie