fwupd/fwupd #2548: Discussion around usage of PCR0 in fwupd

superm1, 2020-11-05:
Current implementation
-----
Currently fwupd fetches a quote of PCR0 from the TPM and assigns it as a checksum to the `System Firmware` device, and LVFS allows setting a valid PCR0 value on a UEFI firmware capsule.  In doing this it lets you check that you are running one of the 'expected' PCR0 values.
Conceptually this makes sense that you want to validate that you are using an expected PCR0 value, but if you dig into the details from the [TCG Platform Firmware Profile](https://trustedcomputinggroup.org/wp-content/uploads/TCG_PCClient_PFP_r1p05_05_3feb20.pdf) of what PCR0 actually represents you find that it's not a good expectation.

> PCR[0] typically represents a consistent view of state of the Host Platform code between boot cycles. This allows Attestation and
Sealed Storage policies to be defined as those using the less changeable platform manufacturer-provided
components of the transitive trust chain. This PCR contains the measurement of code of components provided by
the Host Platform manufacturer. Therefore, the verifier or the entity performing the seal operation can choose to seal (or Quote)
to only those components provided and updated by the Host Platform’s manufacturer. This is also the reason
embedded Driver binaries are measured into PCR[0] as well, thus providing this same consistent view of the
platform regardless of user-selected options.

This view means that code measured into PCR0 are those on a **manufacturer trust boundary** - as in they're provided and updated only by the host manufacturer.  There is no indication or guarantee that they are actually *delivered in* the System Firmware binary that comes in a UEFI capsule update today.  Updated binaries can be distributed however a manufacturer sees fit.  For example if a manufacturer wanted to offer a separate capsule upgrade for every single UEFI firmware volume used in the boot chain, that's entirely valid.  Each of them would still be measured into PCR0.

Problems
----------
So the problem with assigning PCR0 to `System Firmware` is that it makes a gross assumption that all modifications are delivered by the UEFI capsule update associated with monolithic `System Firmware`, when in reality individual pieces of code can be updated from other avenues within the host manufacturer's control.  The host manufacturer may also at their own discretion choose to only update certain components rather than all those delivered in a UEFI capsule update.  As long as the code is used in the standard boot path it should be measured.

This essentially means that selecting a single pre-predictable PCR0 value *before* an update is not a widely realistic expectation, **particularly** because a description matching which components are about to be updated in a given update doesn't exist in the TCG event log.

Furthermore it puts a lot of focus on the final value of PCR0, when in reality:
1. Individual events are interesting and can/should be measured as well
2. Other PCRs are also quite interesting for the trust of the preboot system.

### Individual events
As described in the *Firmware Integrity Measurement specification* [FIM](https://trustedcomputinggroup.org/wp-content/uploads/TCG_PC_Client_FIM_v1_r40_02dec2020.pdf) the intent of the strategy (pages 6-7) around a firmware integrity verifier is for the TPM quote to be used to verify log integrity, but the verifier should perform evaluations based upon the individual events/measurements in the log.  The publication of the OEM's expected values should be coming as part of a RIM, described later.

### Other PCRs
Firmware configuration which is measured into `PCR1` can strongly affect the security of the platform, including CPU microcode, boot entry changes and other BIOS settings changes.  These should be compared between boots for a comprehensive story.

Additionally external EFI drivers or applications loaded during boot time can potentially defeat security, and these are measured into `PCR2` or `PCR4` as they're not on the manufacturer's trust boundary.  Examples include drivers from an add-in card or external PCIe device.

Short term solution
----
In the short term I think that fwupd and LVFS need to change how `PCR0` is used.  The end result of a "predictable" checksum for the machine matching a list of expected checksums doesn't scale as it can be a matrix based upon how many things the OEM changes with any update delivered in any method.
* Specifically in fwupd and LVFS I suggest we should tear out the `PCR0` from the UEFI *System Firmware* device.
* Instead we should make the TPM plugin reconstruct **entire** TCG event log from PCR's 0-7, and measure that into the HSI values.
  - This will rightly put the focus on errors in the OEM's TCG event log implementation if they exist.
  - The event log values should be stored into the fwupd history database, and perhaps we can instead create a mechanism for when "expected" changes are to occur.  For example when updating anything with the uefi plugin, set the next boot as expecting to find new PCR0 values.  Offer a hook for `efibootmgr` (and other similar tools) to call to indicate PCR1 will change on next boot.  When bootloader is updated offer a hook to notify PCR7 will change.

Long term solution
-------------
The TCG has been working on the *Reference Integrity Manifest Specification* [RIM](https://trustedcomputinggroup.org/wp-content/uploads/TCG_PC_Client_RIM_r1p04_pub.pdf) and *Firmware Integrity Measurement specification* [FIM](https://trustedcomputinggroup.org/wp-content/uploads/TCG_PC_Client_FIM_v1_r40_02dec2020.pdf) which describe the relationship of OEM's expected values to how to properly measure them.  The model used to build RIM is specified in the [RIM information model](https://trustedcomputinggroup.org/wp-content/uploads/TCG_RIM_Model_v1-r13_2feb20.pdf).  Some of these specs are in active review *right now*.  Some highlights from the specs I want to note:

> Reference Integrity Measurements, as specified in the TCG Reference Integrity Manifest Specification (RIM) [4] are the set of reference measurements that reflect a baseline, or golden, set of measurements against which a comparison can be made.

The RIM bundle is produced and signed by the manufacturer to reflect the set of measures as expected for the platform.

>A platform OEM can provide a set of RIMs and a Platform Certificate to a Verifier for an endpoint delivered to the Verifier that allows the Verifier to establish a baseline for that computer.

>A RIM Bundle is used by a Verifier as reference for the appraisal process. To perform the appraisal process, the Verifier also needs an Event Log and a TPM Quote from an attestor (as described by the TAP). The values from the attestor are appraised against the PCRIM during a verification process.

>For PC Clients the RIM Bundle is inherently bound to the Firmware lifecycle. The RIM Bundle should be updated during the process of updating the Firmware

The express purpose of the RIM bundle is to perform this appraisal with more granularity than fwupd currently does with a simplfied version of by looking at PCR0.  The key difference is that the RIM contains a concept of the device lifecycle, including other PCRs, and specifically modifications from firmware updates, and any added in card or configuration options.  The bundle of base and supplemental RIMS can be used to appriase the full device.

### Notable comments
It's notable that many of the things that fwupd measures at runtime are asserted in platform certificate (which may be delivered out of band).  Many of those assertions are concepts that were considered for HSI.  I think that collapsing some aspects of HSI into a FIM verifier implementation would make sense.

For example:
> 5.	firmwareCapabilities [5] - This assertion provides an indication of the security capabilities provided by the firmware. If an individual capability is not provided it SHALL be omitted.  If none of the capabilities are provided, this assertion SHALL be omitted.

TCG Participation
----
As the FIM and RIM specs are undergoing public review right now, I think it makes sense to think about how fwupd could behave as a FIM verifier implementation.  If there are gaps in the RIM, RIM information model, or FIM specs now is the time to discuss them with TCG and propose modifications.  Once they are released it will of course take some time for BIOS implementations to catch up to the specs to validate, but I think that this matches the direction for the fwupd's HSI features.

hughsie, 2020-11-05:
> I suggest we should tear out the PCR0 from the UEFI System Firmware device

That's a brilliant write-up, many thanks for that. My main concern would be that removing the PCR0 means there's no way to verify _this is the system BIOS my vendor put there_ -- I know PCR0 isn't that, but it's pretty close from 40,000ft in _most_ cases. 

The best way to get the checksum of course would be to read out the contents of the SPI in the kernel and only include the FVH stuff, i.e. not EFI vars, GbE, EC or ME. I think that the platform-integrity kernel driver might let us do that in the future, at least for some hardware.

Perhaps also we can use the the RIM to generate a simple checksum based on some heuristic, but it's not quite what I expected of a "does the checksum match" like we have for other firmware.

superm1, 2020-11-05:
> That's a brilliant write-up, many thanks for that. My main concern would be that removing the PCR0 means there's no way to verify this is the system BIOS my vendor put there -- I know PCR0 isn't that, but it's pretty close from 40,000ft in most cases.

I guess the other part I wanted to make clear is that the "Attestation checksums" in HSI doesn't make as much sense.  Please take some time to digest the 3 specs I referenced.  The window for feedback is open until December, so if you think anything needs to be changed or added now is the right time.

My thinking was that really what is wanted in place of HSI attestation checksums is a RIM bundle.  And that _might_ not make sense to come **from** LVFS.  Maybe it's something put on the ESP by the OEM or you get from an OEM portal.

hughsie, 2020-11-05:
> the "Attestation checksums" in HSI doesn't make as much sense

I've also heard other feedback saying `-U` doesn't make sense; on the logic you might not be using the LVFS, or for that matter **care** about the LVFS if it's available internally. `-U` also talks about the system firmware but doesn't cover the EC, or equally important ME -- so maybe we just remove both suffixes?

> or you get from an OEM portal

More thinking required I think; the LVFS is certainly the trusted entity for most people here.

superm1, 2020-11-05:
> I've also heard other feedback saying -U doesn't make sense; on the logic you might not be using the LVFS, or for that matter care about the LVFS if it's available internally. -U also talks about [...]

Fine by me.

> More thinking required I think; the LVFS is certainly the trusted entity for most people here.

Maybe the platform certificate goes to LVFS?  Or LVFS pulls it from OEM's site?

hughsie, 2020-11-05:
> Fine by me.

https://github.com/fwupd/fwupd/pull/2552

> Maybe the platform certificate goes to LVFS

Yes, something like that. I'm not sure we want client systems connecting to OEM websites directly.

superm1, 2020-11-05:
> Yes, something like that. I'm not sure we want client systems connecting to OEM websites directly.

These could be opportunities to in the spec decide where they should be stored on the system (like a path on ESP).

superm1, 2020-11-05:
Sure.  But all software should put it in an agreeable location.  For example if the system ships it's there, fwupd places it there, Windows something puts it there, if you do a BIOS update it drops it there.

hughsie, 2020-11-05:
What does putting it in the ESP give us, rather than just /var/lib/<something>?

superm1, 2020-11-05:
OS independent.  If it's there when an OS installer sees it, OS can preserve it.

closed 2020-12-08 by superm1