fwupd/fwupd #2549: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)

XVilka, 2020-11-05:
**Describe the bug**

Any fwupd-related tool/action returns the same error `ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)` in all cases. 

```
$ fwupdtool update
Loading…                 [-                                      ]ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 
Loading…                 [***************************************]
Devices with no available firmware updates: 
 • FiiO Q5
 • SAMSUNG MZVLB1T0HALR-000H1
 • ST1000DM003-1ER162
 • System Firmware
 • UEFI dbx
```

**Steps to Reproduce**

1. Fresh Fedora 33 install
2. Run any `fwupd`-related tool, e.g. `fwupdtool update`

**Expected behavior**

Update/action without the error

**fwupd version information**
Please provide the version of the daemon and client.
```shell
# fwupdmgr --version
client version: 1.5.1
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.1

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
HP ProDesk 680 G4 MT
│
├─FiiO Q5:
│     Device ID:          0fef0a0c55f6442bffaebd774ae771341c89571b
│     Current version:    0.52
│     Vendor:             GuangZhou FiiO Electronics Co.,Ltd (USB:0x2972)
│     Serial Number:      FA507055
│     GUIDs:              fd26e369-c5dc-5891-87d4-683ca2640d48 ← USB\VID_2972&PID_0031&REV_0052
│                         3536ede9-2e66-5403-b6f5-f5a7d7bce268 ← USB\VID_2972&PID_0031
│     Device Flags:       • Updatable
│   
├─Intel(R) Core™ i7-9700 CPU @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000d6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         50a811ae-a8fd-5cd0-90f4-33583974b789 ← CPUID\PRO_0&FAM_06&MOD_9E&STP_D
│     Device Flags:       • Internal device
│   
├─SAMSUNG MZVLB1T0HALR-000H1:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    EXA73H1Q
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:      S3WSNX0N125291
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801&REV_00
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                         47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         2db695a0-5e3c-53bd-9906-78910c9bf74d ← SAMSUNG MZVLB1T0HALR-000H1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─ST1000DM003-1ER162:
│     Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:            ATA Drive
│     Current version:    CC61
│     Vendor:             Seagate (ATA:0x1BB1)
│     Serial Number:      Z4Y7SXN9
│     GUIDs:              93ead325-0b8f-5eef-9625-da930a6f0409 ← IDE\ST1000DM003-1ER162______________________CC61
│                         ad26324f-4ce3-52d2-a80b-dff36f5f737c ← IDE\0ST1000DM003-1ER162______________________
│                         421c1840-4224-53ae-bd2e-74d5f15f2cfb ← ST1000DM003-1ER162
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          f3faa8637f89a4a54ae9c919d077064ad51ba963
│ │   Current version:    135168
│ │   Minimum Version:    1
│ │   Vendor:             HP (DMI:HP)
│ │   GUIDs:              2c316b17-c139-4ffa-9582-5ca125a1fc84
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       69160351-a450-582e-9817-cc96ae775a91 ← UEFI\RES_{2C316B17-C139-4FFA-9582-5CA125A1FC84}
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─Intel AMT [unprovisioned]:
│ │     Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
│ │     Summary:          Hardware and firmware technology for remote out-of-band management
│ │     Current version:  12.0.49.1534
│ │     Bootloader Version:12.0.49.1534
│ │     Vendor:           Intel Corporation
│ │     GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
│ │     Device Flags:     • Internal device
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  78
│       Minimum Version:  78
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            95641a6b-bc03-5b8b-bdfc-c92dab523a06 ← UEFI\CRT_C8B4C6D9755968B6DEB48D4768DD5E9020D3FF4ADC9A62521F54252DD4241678
│                         7689caf4-c147-5c67-bff9-5dbe59a441bd ← UEFI\CRT_C8B4C6D9755968B6DEB48D4768DD5E9020D3FF4ADC9A62521F54252DD4241678&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
└─UHD Graphics 630 (Desktop 9 Series):
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    02
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              03f30c74-e6c2-5853-bee2-1524d39abee0 ← PCI\VEN_8086&DEV_3E98&SUBSYS_103C8580&REV_02
                          ac56d9a4-54b9-5f86-a885-70fca6a60091 ← PCI\VEN_8086&DEV_3E98&SUBSYS_103C8580
                          e9ceb7d8-e213-5ec4-8805-6330d894b317 ← PCI\VEN_8086&DEV_3E98&REV_02
                          8655a49b-b4b2-54ff-a6c6-06009dd64f13 ← PCI\VEN_8086&DEV_3E98
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available

```

**Additional questions**
- Operating system and version: Fedora 33 x86_64
- Have you tried rebooting? Yes
- Is this a regression? Never tested before on the same hardware

superm1, 2020-11-05:
This seems to be expected behavior from fwupd perspective if the TPM isn't working.  Is it turned off in firmware setup?

XVilka, 2020-12-07:
Yes, it was turned off. Sorry for the trouble - didn't realize it's a requirement for fwupd to work.

closed 2020-12-07 by XVilka

catskul, 2021-04-21:
While this might be expected behavior, the error behavior should be improved. IMO this should be reopened.

LinuxOnTheDesktop, 2021-05-12:
'This seems to be expected behavior from fwupd perspective if the TPM isn't working.' One would not know that from the error message (an error message that I received just now).

fswasey, 2021-07-22:
This is the last message in my system log before the system spontaneously rebooted itself.  How is that "expected behavior"?

hughsie, 2021-07-22:
@fswasey I don't think fwupd rebooted your machine, sorry.