fwupd/fwupd #2553: UEFI ESP partition is in /boot/efi but not recognized

bin101, 2020-11-05:
**Describe the bug**
Every fwupd command responses with "UEFI ESP partition not detected or configured"

**Steps to Reproduce**
$ fwupdmgr refresh
WARNING: UEFI ESP partition not detected or configured

**Expected behavior**
The UEFI ESP partition should be detected

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

**Additional questions**
- Operating system and version: Manjaro 20.2
- Have you tried rebooting? Yes
- Is this a regression? don't know

```shell
$ tree /boot/
/boot/
├── amd-ucode.img
├── efi
│   └── EFI
│       ├── boot
│       │   └── bootx64.efi
│       └── manjaro
│           ├── fw
│           │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│           │   └── fwupd-603baf73-b997-45b5-86b4-2f981a008e18.cap
│           ├── fwupdx64.efi
│           └── grubx64.efi
├── grub
.
.
.
```

```shell
$ /usr/bin/fwupdtool esp-list --verbose                               
14:42:08:0585 FuDebug              Verbose debugging enabled (on console 1)
14:42:08:0592 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
14:42:08:0596 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4
14:42:08:0603 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
```

**If I set OverrideESPMountPoint to /boot/efi the warning disappears.**

hughsie, 2020-11-05:
I think if you install 1.5.1 you'll get a link to the wiki page https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found

bin101, 2020-11-05:
I already modified OverrideESPMountPoint to /boot/efi as mentioned in the last sentence.

But the wiki says /boot/efi should be recognized as common path but it shouldn't.

hughsie, 2020-11-05:
Are you using UDisks?

bin101, 2020-11-05:
Yes,
first I used OverrideESPMountPoint=/boot/efi as workaround. Then I recognized that my /boot/efi hasn't the desired partition type GUID as "$ /usr/bin/fwupdtool esp-list --verbose" already told me. I used gparted to change it and after a reboot I was able to remove the OverrideESPMountPoint workaround.

But this is not what I want to point out, what I want to ask: Shouldn't common path recognition should be a fallback if the efi partition wasn't recognized by the partition GUID?

hughsie, 2020-11-05:
> Shouldn't common path recognition should be a fallback if the efi partition wasn't recognized by the partition GUID

I think using heuristics like this is somewhat dangerous; I think the GUID has to be correct if we want the firmware updater to also find it when we reboot with the fwupdx64.efi helper.

bin101, 2020-11-05:
Ok, then I will close this :+1:

closed 2020-11-05 by bin101

Vladimir-csp, 2020-11-14:
Please reopen.
fwupd 1.5.1, ESP has correct GUID, mounted to `/boot/efi`. `OverrideESPMountPoint=/boot/efi` is set (no udisks).
Still throws this warning.

superm1, 2020-11-14:
Can you please check again with master? We did end up adding a heuristic.

Vladimir-csp, 2020-11-15:
Thanks. I'm having trouble building it on Debian testing, so I'll probably wait for next release.

superm1, 2020-11-16:
@Vladimir-csp if you want to share a log of your build error, happy to try to help.  I'd suspect you're probably missing tpm2-tss 3.0.1-4 which hasn't migrated to testing yet though.

Vladimir-csp, 2020-11-16:
Dug into it and, ironically, a test fails during build:
```
Bail out! FuPluginDell:ERROR:../plugins/dell/fu-self-test.c:488:fu_test_self_init: assertion failed (error == NULL): failed to startup using uefi: invalid OverrideESPMountPoint=/boot/efi specified in config: failed to call org.freedesktop.UDisks2.GetBlockDevices(): The name org.freedesktop.UDisks2 was not provided by any .service files (g-dbus-error-quark, 2)
```
Built without tests, it works.
Also turns out that I forgot to restart fwupd after editing uefi.conf, so my first message here is false, 1.5.1 works when ESP mountpoint is manually specified.
On the other hand, master does not work without it, so what kind of heuristic does it contain?

superm1, 2020-11-16:
> Also turns out that I forgot to restart fwupd after editing uefi.conf, so my first message here is false, 1.5.1 works when ESP mountpoint is manually specified.

OK that's reassuring.

>On the other hand, master does not work without it, so what kind of heuristic does it contain?

Well you need udisks2 installed to start.  If you don't have it, won't be able to use the heuristic.  So I guess from your above error you don't have it installed.  If you do have it installed it should try to use ESP GUID, followed by looking for internal FAT32 partitions for finding an ESP.

Vladimir-csp, 2020-11-16:
Thank you. I have no further questions, sorry for a false alarm.