fwupd/fwupd #2554: System firmware update for Dell XPS 7590 - failed to update to 0: invalid firmware format

fatalhare, 2020-11-05:
I apologize if this has been reported previously - I searched through closed issues briefly and didn't see anyone else with an XPS 7590 having this error.

**Describe the bug**
Try to update system firmware from 1.0.0 to 1.17.1. Firmware downloads okay and proceeds to reboot but the flash immediately fails. Log says 

`Update Error:       failed to update to 0: invalid firmware format`

**Steps to Reproduce**
Run `fwupdmgr get-updates` and proceed through prompts to reboot, or try to upgrade through gnome-software interface, which also fails.

**Expected behavior**
Expect firmware to update successfully

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): 
`Pacman`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
XPS 15 7590
│
├─Thunderbolt host controller:
│     Device ID:          bf092f6517bb4da97273e9fe626d8bea5947023e
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    41.00
│     Vendor:             Dell (TBT:0x00D4)
│     GUIDs:              cabd17f2-67af-5a15-812c-e36de87d6133
│                         87ef6395-d440-5709-b471-1e152117c8a0
│                         dc6ebd31-1fe5-5230-8090-dd0b06da4f87
│                         dd244205-fa73-594e-b5ce-3aab372825e0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─USB3.1 Hub:
│     Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
│     Summary:            USB 3.x Hub
│     Current version:    30.3
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              676232b1-3368-595f-8672-a882831d57c9
│                         08887a10-1663-55e7-ab55-a34cc4f0d011
│                         e2db8274-3ab0-5120-9d9a-2f9693cd7db8
│                         a639a3d8-c852-5dd3-a325-d7b04e163b79
│                         9bed0c73-4ed7-5734-9919-1a217b2a63c0
│                         7e650fa0-f815-531c-be10-ad34374b4748
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─Intel(R) Core™ i7-9750H CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000d6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
[... 466 lines cut ...]
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

5 directories, 314 files
```

**Additional questions**
- Operating system and version: 
Arch linux, 5.9.4-arch1-1

- Have you tried rebooting?
Yes

- Is this a regression?
Has never worked for me before

- Are you using an NVMe disk?
Yes

- Is secure boot enabled?
Disabled

Thanks for your help!

superm1, 2020-11-05:
`1.17.1`.  That's trying to upgrade to the wrong system. 
`e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0` is the FW for `XPS 15 9570/Precision 5530`.

Your system should be using `417d4c2a-87d1-4d7c-bcea-322041f2d5a3`.

I think this is a case from a while back that the wrong GUID got used and it was updated later.  Can you please manually download the EXE from support.dell.com and upgrade using the F12 menu at POST from a USB key or copying it to your EFI system partition?  After that hopefully it's corrected.

superm1, 2020-11-05:
> invalid firmware format

It's fortunate that the firmware will reject it if it's wrong system though so you don't have a brick :)

hughsie, 2020-11-05:
>  e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0 is the FW for XPS 15 9570/Precision 5530.

Eeel, do we need to pull a firmware back from stable on the LVFS perhaps?

superm1, 2020-11-05:
>Eeel, do we need to pull a firmware back from stable on the LVFS perhaps?

If it was bricking things I'd say yes.  But I think the answer here is actually a CHID.  I want to make sure that after he's on a current firmware the GUID changes though otherwise it's a larger problem..

fatalhare, 2020-11-05:
> > invalid firmware format
> 
> It's fortunate that the firmware will reject it if it's wrong system though so you don't have a brick :)

Very fortunate indeed!

After downloading the latest update from dell (and verifying my recent backups are intact) I did the update by flashing from a USB key. 

```shell
$ fwupdmgr get-devices --show-all-devices
XPS 15 7590
│
├─Thunderbolt host controller:
│     Device ID:          bf092f6517bb4da97273e9fe626d8bea5947023e
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    41.00
│     Vendor:             Dell (TBT:0x00D4)
│     GUIDs:              cabd17f2-67af-5a15-812c-e36de87d6133
│                         87ef6395-d440-5709-b471-1e152117c8a0
│                         dc6ebd31-1fe5-5230-8090-dd0b06da4f87
│                         dd244205-fa73-594e-b5ce-3aab372825e0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Intel(R) Core™ i7-9750H CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000d6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543
│     Device Flags:       • Internal device
│   
├─KXG60ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    10604107
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     GUIDs:              8a412b95-5c07-5ce4-b63a-f9cbf254dd5d
│                         344c2804-4948-e811-842f-0ed5f89f718b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          90dce4a3c3eb770c96eaa7fa7f45e984fc66206d
│ │   Current version:    1.8.1
│ │   Minimum Version:    1.8.1
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              417d4c2a-87d1-4d7c-bcea-322041f2d5a3
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       82ffca40-d912-551d-89c7-a02973327fd9
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─Intel AMT [unprovisioned]:
│ │     Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
│ │     Summary:          Hardware and firmware technology for remote out-of-band management
│ │     Current version:  12.0.68.1606
[... 49 lines cut ...]
│                         741692a4-ae19-5ecc-9e48-e644604c2345
│                         a90f8ef5-de8d-5810-a6c2-d0b88b1e0be8
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─UHD Graphics 630 (Mobile):
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              67a16a3d-a85e-5df7-a160-6ff781c51e31
                          106f82d7-5ece-5f9a-ad60-eb17a266152a
                          12c60b98-52cd-500a-b99e-087aa6b47f0b
                          db677673-1102-5481-a11e-6f397e123ced
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
    
________________________________________________

Devices that were not updated correctly:

 • System Firmware (1.0.0 → 1.17.1)
```

So it looks like update went well and the GUID (or one of them, at least) matches up. I still have the error message at the bottom for the old update attempt, but I'm guessing that will disappear after a successful update sometime in the future.

Thanks very much for your help. If you need anything else from me, let me know.

closed 2020-11-06 by fatalhare

Albonycal, 2024-05-01:
Hey, I also did the same thing. Used the wrong GUID initially and it was rejected. I have the same message when I do `fwupdtool get-devices `  Is that something to worry about? The system works fine. 
Is there any way to get rid of that message? 

Thanks

superm1, 2024-05-06:
@CragW ^

CragW, 2024-05-06:
> Hey, I also did the same thing. Used the wrong GUID initially and it was rejected. I have the same message when I do `fwupdtool get-devices ` Is that something to worry about? The system works fine. Is there any way to get rid of that message?
> 
> Thanks

@Albonycal can you pl. share back the output from `$ sudo fwupdtool get-devices -vv`

Albonycal, 2024-05-06:
[fwupd.log](https://github.com/fwupd/fwupd/files/15216131/fwupd.log)
Sure

CragW, 2024-05-06:
> [fwupd.log](https://github.com/fwupd/fwupd/files/15216131/fwupd.log) Sure

This is a Lenovo machine, you probably want to create a new issue for the team to fix it.

Albonycal, 2024-05-06:
What do you mean exactly? My device isn't in LVFS, I upgraded it manually by extracted the .fd file from the .exe supplied by Lenovo

CragW, 2024-05-06:
> What do you mean exactly? My device isn't in LVFS, I upgraded it manually by extracted the .fd file from the .exe supplied by Lenovo. 

The error message means your system firmware declined the update attempt, most likely the `file` is NOT compatible to your machine or essentially is in an unsupported format.

> Edit: Okay I get it, you are a dell employee that's why you were pinged into this issue 👍 Sorry for not being specific in my initial comment

No problem. :)

Albonycal, 2024-05-06:
> The error message means your system firmware declined the update attempt, most likely the file is NOT compatible to your machine or essentially is in an unsupported format.

Yeah I know, the thing that caused confusion is this :
```
UEFI Device Firmware:
      Device ID:          2436ce30f3e31557eecfe30d7e0396e61bc00aa2
      Summary:            UEFI System Resource Table device (updated via NVRAM)
      Current version:    3091039
      Vendor:             DMI:LENOVO
      Update State:       Failed
      Update Error:       failed to update to 0: error-invalid-format
      ```
      ```
      System Firmware:
│ │   Device ID:          1fe435153b220289b1972071582b69a8e401a6cb
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1497841697
│ │   Minimum Version:    1380122624
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
```

I didn't know which one was the actual device because the names were confusing, so I just "updated" both using the same file.