fwupd/fwupd #2556: Segmentation fault on Ubuntu 20.04 (client version 1.3.11)

gareth-cross, 2020-11-06:
**Describe the bug**
Launching fwupd (either on its own or via systemctl) causes segmentation fault. Let me know if this is the wrong place to post this - I wasn't sure whether to post here or report a bug on the ubuntu package. 

To begin debugging, I ran `/usr/libexec/fwupd/fwupd -v` in GDB. The output and backtrace follows:

```
Starting program: /usr/libexec/fwupd/fwupd -v

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff46e7700 (LWP 7368)]
[New Thread 0x7ffff3ee6700 (LWP 7369)]
04:05:01:0242 FuDebug              Verbose debugging enabled (on console 1)
04:05:01:0242 FuConfig             loading config values from /etc/fwupd/daemon.conf
04:05:01:0242 FuRemoteList         loading /usr/share/fwupd/metainfo
04:05:01:0243 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
04:05:01:0243 XbSilo               file: 93057c4e-8810-5d9f-6a02-1163bd5b4c6d, current:93057c4e-8810-5d9f-6a02-1163bd5b4c6d, cached: (null)
04:05:01:0243 XbSilo               loading silo with file contents
04:05:01:0243 Fwupd                looking for os-release at /etc/os-release
04:05:01:0243 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
04:05:01:0243 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
04:05:01:0243 Fwupd                converting Keyring value to Jcat
04:05:01:0244 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
04:05:01:0244 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
04:05:01:0244 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
04:05:01:0244 Fwupd                converting Keyring value to Jcat
04:05:01:0244 FuRemoteList         ignoring unfound remote fwupd
04:05:01:0244 FuRemoteList         ordering lvfs-testing=lvfs+1
04:05:01:0244 FuRemoteList         ignoring unfound remote fwupd
04:05:01:0244 FuRemoteList         ignoring unfound remote fwupd
04:05:01:0244 FuRemoteList         ignoring unfound remote fwupd
04:05:01:0244 FuEngine             failed to sign using keyring: Failed to open file “/var/lib/fwupd/pki/secret.key”: Permission denied
04:05:01:0244 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
04:05:01:0245 FuHistory            got schema version of 5
04:05:01:0245 FuIdle               setting timeout to 7200s
04:05:01:0245 FuEngine             Failed to load SMBIOS: Failed to open file “/sys/firmware/dmi/tables/smbios_entry_point”: Permission denied
04:05:01:0245 FuHwids              ignoring Manufacturer: no structure with type 01
04:05:01:0245 FuHwids              ignoring EnclosureKind: no structure with type 03
04:05:01:0245 FuHwids              ignoring Family: no structure with type 01
04:05:01:0245 FuHwids              ignoring ProductName: no structure with type 01
04:05:01:0245 FuHwids              ignoring ProductSku: no structure with type 01
04:05:01:0245 FuHwids              ignoring BiosVendor: no structure with type 00
04:05:01:0245 FuHwids              ignoring BiosVersion: no structure with type 00
04:05:01:0245 FuHwids              ignoring BiosMajorRelease: no structure with type 00
04:05:01:0245 FuHwids              ignoring BiosMinorRelease: no structure with type 00
04:05:01:0245 FuHwids              ignoring BaseboardManufacturer: no structure with type 02
04:05:01:0245 FuHwids              ignoring BaseboardProduct: no structure with type 02
04:05:01:0245 FuHwids              HardwareID-0 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-1 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-2 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-3 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-4 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-5 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-6 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-7 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-8 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-9 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-10 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-11 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-12 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-13 is not available, not available as 'Manufacturer' unknown
04:05:01:0245 FuHwids              HardwareID-14 is not available, not available as 'Manufacturer' unknown
04:05:01:0250 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
04:05:01:0250 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
04:05:01:0250 XbSilo               file: 1b0400a0-acd2-fb11-baad-9bde91a1f4e0, current:1b0400a0-acd2-fb11-baad-9bde91a1f4e0, cached: (null)
04:05:01:0250 XbSilo               loading silo with file contents
04:05:01:0251 FuEngine             remote lvfs-testing not enabled, so skipping
04:05:01:0251 FuEngine             remote vendor not enabled, so skipping
04:05:01:0251 FuEngine             remote vendor-directory not enabled, so skipping
04:05:01:0251 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
[... 121 lines cut ...]
```
fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Not 100% sure, I believe it was installed by doing `sudo apt install fwupd-gui` - **or** it came with stock Ubuntu.

**fwupd device information**
Running ` fwupdmgr get-devices --show-all-devices` does not provide output, because the process encounters the fault (I assume). I get the following:
```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

**Additional questions**
- Operating system and version: Ubuntu 20.04.1, Kernel 5.4.0-52-generic
- Have you tried rebooting? Yes.
- Is this a regression? Not sure if it ever worked, this is a brand new install. I have not seen it run successfully yet.

It looks superficially similar to this: https://bugzilla.redhat.com/show_bug.cgi?id=1889122

Let me know if there are other debugging steps I can take or useful information I can provide. Thank you for your assistance.

hughsie, 2020-11-06:
> #0  0x00005555555822c8 in fu_plugin_list_depsolve ()

Can you try with debuginfo installed please?

gareth-cross, 2020-11-07:
updated backtrace:
```
Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
0x00005555555822c8 in fu_plugin_list_depsolve (self=0x7fffe40128d0, error=error@entry=0x7fffffffdf38) at ../src/fu-plugin-list.c:147
147				deps = fu_plugin_get_rules (plugin, FU_PLUGIN_RULE_RUN_AFTER);
(gdb) bt
#0  0x00005555555822c8 in fu_plugin_list_depsolve (self=0x7fffe40128d0, error=error@entry=0x7fffffffdf38) at ../src/fu-plugin-list.c:147
#1  0x0000555555578a2d in fu_engine_load_plugins (self=self@entry=0x555555603020, error=error@entry=0x7fffffffdf38) at ../src/fu-engine.c:4848
#2  0x00005555555792ab in fu_engine_load (self=0x555555603020, flags=FU_ENGINE_LOAD_FLAG_NONE, error=0x7fffffffdf38) at ../src/fu-engine.c:5242
#3  0x0000555555566c8f in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:1598
```

In the context of the code:
```
for (guint i = 0; i < self->plugins->len; i++) {
			FuPlugin *plugin = g_ptr_array_index (self->plugins, i);
			deps = fu_plugin_get_rules (plugin, FU_PLUGIN_RULE_RUN_AFTER);  // <- where the segfault happens
```
According to gdb, `i=0`, `plugin` has name `wacom_usb`, `self->plugins` has length 41 - so it appears to be on the first iteration of the loop. If I place a breakpoint in fu_plugin_get_rules I don't see any locals (optimized out maybe?).

Anything else I could run to get more useful info?

hughsie, 2020-11-07:
Hmm, no nothing jumps out. 1.3.x isn't really a supported release anymore, so if you could try 1.4.x or 1.5.x and see if you get the same problem that would be useful.

superm1, 2020-11-07:
This feels like stack corruption to segfault on that.  By chance can you also reproduce it with `fwupdtool get-devices`?

gareth-cross, 2020-11-08:
@superm1 Yes that command also segfaults in the same place. I will try 1.5, thanks @hughsie

superm1, 2020-11-08:
If you can readily reproduce it with fwupdtool, I wonder if it can be narrowed to a selection of plugins I'm the whitelist too.

yaperez-anssi, 2020-11-16:
I'm apparently able to reproduce on a ThinkPad X280 running Debian sid (fwupd 1.4.6):

```
fwupdtool -v get-devices 
14:57:01:0381 FuDebug              Verbose debugging enabled (on console 1)
14:57:01:0391 FuConfig             loading config values from /etc/fwupd/daemon.conf
14:57:01:0392 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
14:57:01:0392 FuRemoteList         loading /usr/share/fwupd/metainfo
14:57:01:0393 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
14:57:01:0393 XbSilo               file: 8ac2593a-1867-81d5-3051-6551716c8c54, current:8ac2593a-1867-81d5-3051-6551716c8c54, cached: (null)
14:57:01:0393 XbSilo               loading silo with file contents
14:57:01:0393 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.dpkg-old
14:57:01:0393 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
14:57:01:0393 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.dpkg-old
14:57:01:0393 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
14:57:01:0393 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
14:57:01:0393 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
14:57:01:0393 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/fwupd.conf
14:57:01:0394 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
14:57:01:0394 FuRemoteList         ordering lvfs-testing=lvfs+1
14:57:01:0394 FuRemoteList         ordering lvfs=fwupd+1
14:57:01:0394 FuRemoteList         ordering lvfs-testing=lvfs+1
14:57:01:0394 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
14:57:01:0394 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
14:57:01:0394 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
14:57:01:0394 Jcat                 loaded 1 certificates
14:57:01:0394 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:57:01:0394 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
14:57:01:0394 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
14:57:01:0394 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
14:57:01:0394 Jcat                 loaded 1 certificates
14:57:01:0394 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:57:01:0394 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
14:57:01:0394 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
14:57:01:0394 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
14:57:01:0402 FuEngine             client certificate exists and working
14:57:01:0402 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
14:57:01:0402 FuHistory            got schema version of 6
14:57:01:0403 FuHwids              smbios property Manufacturer=LENOVO
14:57:01:0403 FuHwids              smbios property EnclosureKind=a
14:57:01:0403 FuHwids              smbios property Family=ThinkPad X280
14:57:01:0403 FuHwids              smbios property ProductName=20KESD8C00
14:57:01:0403 FuHwids              smbios property ProductSku=LENOVO_MT_20KE_BU_Think_FM_ThinkPad X280
14:57:01:0403 FuHwids              smbios property BiosVendor=LENOVO
14:57:01:0403 FuHwids              smbios property BiosVersion=N20ET55W (1.40 )
14:57:01:0403 FuHwids              smbios property BiosMajorRelease=01
14:57:01:0403 FuHwids              smbios property BiosMinorRelease=28
14:57:01:0403 FuHwids              smbios property BaseboardManufacturer=LENOVO
14:57:01:0403 FuHwids              smbios property BaseboardProduct=20KESD8C00
14:57:01:0406 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
14:57:01:0406 XbSilo               file: aed3357f-6b5f-e962-3d3a-575f7484f59a, current:aed3357f-6b5f-e962-3d3a-575f7484f59a, cached: (null)
14:57:01:0406 XbSilo               loading silo with file contents
14:57:01:0406 FuEngine             remote lvfs-testing not enabled, so skipping
14:57:01:0406 FuEngine             no /usr/share/fwupd/remotes.d/fwupd/metadata.xml, so skipping
14:57:01:0406 FuEngine             remote vendor not enabled, so skipping
14:57:01:0406 FuEngine             remote vendor-directory not enabled, so skipping
14:57:01:0406 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
14:57:01:0407 XbSilo               0.02ms      parse blob
14:57:01:0407 XbSilo               0.05ms      loaded file
14:57:01:0407 XbSilo               file: 07427fe8-4939-8fc0-f6a8-165fe01c8f28, current:07427fe8-4939-8fc0-f6a8-165fe01c8f28, cached: (null)
14:57:01:0407 XbSilo               loading silo with file contents
14:57:01:0407 XbSilo               0.01ms      parse blob
14:57:01:0407 XbSilo               0.30ms      query on / with `components/component` limit=0 -> 468 results
14:57:01:0407 FuEngine             468 components now in silo
14:57:01:0407 XbSilo               0.41ms      query on / with `components/component/provides/firmware` limit=0 -> 554 results
14:57:01:0408 XbSilo               0.17ms      query on / with `components/component/provides/firmware` limit=0 -> 554 results
[... cut ...]
15:00:29:0455 FuEngine             failed to find specified plugin vli: no plugin vli found
15:00:29:0463 FuEngine             failed to find specified plugin vli: no plugin vli found
15:00:29:0467 FuEngine             failed to find specified plugin vli: no plugin vli found
15:00:29:0476 FuEngine             device ID 093ef0be8328a2c4ed2fe55cd36aae3171b92ade was not found
15:00:29:0476 FuEngine             Emitting PropertyChanged('Status'='idle')
Loading…                 [***************************************]
No detected devices
```

superm1, 2020-11-16:
@yaperez-anssi

A few things:
1. Can you give the debian package versions for tpm2-tss and libgusb?  Those both had changes recently and I want to make sure you have them landed.
2. Can you please try with 1.5.1 from unstable?
3. If it persists, can you please get us a backtrace?

yaperez-anssi, 2020-11-16:
> @yaperez-anssi
> 
> A few things:
> 
>     1. Can you give the debian package versions for tpm2-tss and libgusb?  Those both had changes recently and I want to make sure you have them landed.
```
dpkg -l libtss2\* libgusb2
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version      Architecture Description
+++-======================-============-============-====================================================
ii  libgusb2:amd64         0.3.5-1      amd64        GLib wrapper around libusb1
ii  libtss2-esys0:amd64    3.0.1-4      amd64        TPM2 Software stack library - TSS and TCTI libraries
ii  libtss2-mu0:amd64      3.0.1-4      amd64        TPM2 Software stack library - TSS and TCTI libraries
ii  libtss2-rc0:amd64      3.0.1-4      amd64        TPM2 Software stack library - TSS and TCTI libraries
ii  libtss2-sys1:amd64     3.0.1-4      amd64        TPM2 Software stack library - TSS and TCTI libraries
ii  libtss2-tctildr0:amd64 3.0.1-4      amd64        TPM2 Software stack library - TSS and TCTI libraries
```
 
>     2. Can you please try with 1.5.1 from unstable?

Sorry, I missed that I wasn't at the latest version available, but that's because the SB-signed version is not yet available from unstable. I've upgraded (and removed the signed version) for testing and it *does* work, at least the get-devices stuff. Sorry for the inconvenience.

I'll downgrade to the signed version and wait for the signed 1.5 to appear in unstable.

closed 2020-12-24 by stale[bot]