fwupd/fwupd #2566: Installed tests failing without network

jtojnar, 2020-11-07:
**Describe the bug**

The `fwupd/fwupdmgr.test` installed test seems to fail with `Error resolving “cdn.fwupd.org”: Name or service not known` in NixOS VM without network access, even though we are not setting `CI_NETWORK` environment variable.

<details>
<summary>Full log</summary>

```
machine # [   27.541516] nscd[805]: 805 checking for monitored file `/etc/netgroup': No such file or directory
machine # [   56.652725] dbus-daemon[689]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.2' (uid=0 pid=836 comm="/run/current-system/sw/bin/fwupdmgr get-remotes " label="kernel")
machine # [   56.665168] systemd[1]: Starting Firmware update daemon...
machine # [   57.370261] fwupd[840]: 16:49:42:0684 FuEngine             invalid has incorrect built version invalid
machine # [   57.515454] fwupd[840]: 16:49:42:0830 FuDeviceList         ignoring device f09242e822d462eb625aaa31fb9eec65825f7b25 [optionrom:(null)] existing device 8bc8891dabab35ae76292428f63359adabdb99de [optionrom:(null)] already exists
machine # [   57.562469] dbus-daemon[689]: [system] Activating via systemd: service name='org.freedesktop.PolicyKit1' unit='polkit.service' requested by ':1.3' (uid=0 pid=840 comm="/nix/store/21v5snjn76yaha5mj2wnlpryrnd7k2lh-fwupd-" label="kernel")
machine # [   57.571842] systemd[1]: Starting Authorization Manager...
machine # [   57.642530] polkitd[846]: Started polkitd version 0.116
machine # [   57.745831] polkitd[846]: Loading rules from directory /etc/polkit-1/rules.d
machine # [   57.747281] polkitd[846]: Loading rules from directory /run/current-system/sw/share/polkit-1/rules.d
machine # [   57.754569] polkitd[846]: Finished loading, compiling and executing 3 rules
machine # [   57.757711] dbus-daemon[689]: [system] Successfully activated service 'org.freedesktop.PolicyKit1'
machine # [   57.761514] systemd[1]: Started Authorization Manager.
machine # [   57.763514] polkitd[846]: Acquired the name org.freedesktop.PolicyKit1 on the system bus
machine # [   57.780549] dbus-daemon[689]: [system] Successfully activated service 'org.freedesktop.fwupd'
machine # [   57.785211] systemd[1]: Started Firmware update daemon.
machine # [   60.370498] gnome-desktop-testing-runner[832]: FAIL: fwupd/fwupdmgr.test (Child process exited with code 1)
machine # [   60.541453] dbus-daemon[689]: [system] Activating via systemd: service name='org.freedesktop.UDisks2' unit='udisks2.service' requested by ':1.19' (uid=0 pid=951 comm="/nix/store/v5ryvn1m8pwq4hw5w3v90vg647xr845f-fwupd-" label="kernel")
machine # [   60.550268] systemd[1]: Starting Disk Manager...
machine # [   60.598618] udisksd[954]: udisks daemon version 2.8.4 starting
machine # [   60.603896] udisksd[954]: Failed to load config files: Failed to get contents of the config dir (/etc/libblockdev/conf.d/)Error opening directory “/etc/libblockdev/conf.d/”: No such file or directory. Using the built-in config
machine # [   60.830688] dbus-daemon[689]: [system] Successfully activated service 'org.freedesktop.UDisks2'
machine # [   60.833106] systemd[1]: Started Disk Manager.
machine # [   60.834270] udisksd[954]: Acquired the name org.freedesktop.UDisks2 on the system message bus
machine # [   61.154459] gnome-desktop-testing-runner[832]: PASS: fwupd/fwupd.test
machine # [   61.161633] gnome-desktop-testing-runner[832]: SUMMARY: total=2; passed=1; skipped=0; failed=1; user=0.5s; system=1.2s; maxrss=25244
machine: output: Running test: fwupd/fwupdmgr.test
Getting the list of remotes...
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: Firmware can not be updated in legacy BIOS mode
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
Standard PC (i440FX + PIIX, 1996)
│
├─Linux Vendor Firmware Service (testing):
│     Remote ID:          lvfs-testing
│     Type:               download
│     Keyring:            jcat
│     Enabled:            false
│     Priority:           1
│     Filename:           /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz
│     Filename Signature: /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz.jcat
│     Metadata URI:       https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
│     Metadata Signature: https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.jcat
│     Report URI:         https://fwupd.org/lvfs/firmware/report
│     Automatic Reporting:false
│   
├─Enable UEFI capsule updates on Dell systems:
│     Remote ID:          dell-esrt
│     Type:               local
│     Keyring:            none
│     Enabled:            true
│     Filename:           /nix/store/21v5snjn76yaha5mj2wnlpryrnd7k2lh-fwupd-1.5.1/share/fwupd/remotes.d/dell-esrt/metadata.xml
│   
├─fwupd test suite:
│     Remote ID:          fwupd-tests
│     Type:               local
│     Keyring:            none
│     Enabled:            true
│     Filename:           /nix/store/v5ryvn1m8pwq4hw5w3v90vg647xr845f-fwupd-1.5.1-installedTests/share/installed-tests/fwupd/fwupd-tests.xml
│   
├─Linux Vendor Firmware Service:
│     Remote ID:          lvfs
[... 286 lines cut ...]
ok 3 /dfu/firmware{raw}
ok 4 /dfu/firmware{dfu} # SKIP Missing kiibohd.dfu.bin
ok 5 /dfu/firmware{dfuse} # SKIP Missing dev_VRBRAIN.dfu
# End of dfu tests
PASS: fwupd/fwupd.test
SUMMARY: total=2; passed=1; skipped=0; failed=1; user=0.5s; system=1.2s; maxrss=25244
FAIL: fwupd/fwupdmgr.test (Child process exited with code 1)
```

</details>

The log says it is tainted but that seems to be caused by the first-party `invalid` plug-in.

**Steps to Reproduce**
Run installed tests.

**Expected behavior**
A clear and concise description of what you expected to happen.

fwupd 1.5.1 packaged by myself for NixOS (https://github.com/NixOS/nixpkgs/pull/101826).

**Additional questions**
- Operating system and version: NixOS unstable
- Have you tried rebooting? using fresh VM each time
- Is this a regression? not sure

hughsie, 2020-11-08:
I think the installed tests correctly expect network access.

jtojnar, 2020-11-08:
Hmm, would it be possible to add a way to skip those tests that do? Our testing infrastructure does not allow internet access to reduce non-determinism.

hughsie, 2020-11-08:
> Our testing infrastructure does not allow internet access to reduce non-determinism.

We could host the metadata and mock up the DNS if that helps. Not sure how to do that in CI tho.

jtojnar, 2020-11-08:
Could we use the Python’s [built-in HTTP server](https://docs.python.org/3/library/http.server.html#http.server.SimpleHTTPRequestHandler) and point the domain to 127.0.0.1 in `/etc/hosts`?

superm1, 2020-11-09:
I think the reason for this failure cropping up now is installed tests were apparently broken for a long period of time, but the error was getting ignored.  They were fixed recently: https://github.com/fwupd/fwupd/commit/406025ba567e905000a49a76784f1bc2e6b4901a#diff-3e63ae314fc0935e5dae47efa04ecf297de9428134d6e3ed64613ec0ca7cf287

> Could we use the Python’s built-in HTTP server and point the domain to 127.0.0.1 in /etc/hosts?
> Edit: Hmm, we would probably also need to include TLS support and add the self-signed certificate to the system.

I think this is going to be painful and not worth it.

I'd say it's down to 2 options:
1. Look for `CI_NETWORK`
2. Not running installed tests but only running the compilation time tests for NixOS.

My suggestion is the 1.  I just sent up a small script change for it.

PR #2569 (2020-11-09): trivial: installed-tests: skip tests requiring network by default
Most of the installed tests will skip if files needed to be fetched
from network are not available.  Do the same thing for fwupdmgr
tests.

Fixes: #2566

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [ ] Documentation
commit db15442c (2020-11-09): trivial: installed-tests: skip tests requiring network by default

Most of the installed tests will skip if files needed to be fetched
from network are not available.  Do the same thing for fwupdmgr
tests.

Fixes: #2566
closed 2020-11-09 by superm1