fwupd/fwupd #2567: No updatable devices on Dell 13 XPS 9300

blauch, 2020-11-08:
I started seeing the warning `WARNING: UEFI ESP partition not detected or configured` and also the message `No updatable devices`. Unfortunately, I cannot really say when the issue started (perhaps with the Nibia upgrade :man_shrugging:)  but I have been using fwupd without issues for some time already. 

I am running Manjaro 20.2 (Nibia) with kernel 5.9.3-1-MANJARO. `fwupd` was installed with `pacman`. I tried to uninstall, reboot and re-install without luck. I have a NVMe disk and secure boot is disabled.

```
$ fwupdmgr --version
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.0
```

fwupdmgr get-devices:  https://gist.github.com/blauch/99706f568a057d7f06417fac0b5a4183

```
$ sudo fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
Devices with no available firmware updates: 
 • Integrated Webcam HD
 • PC611 NVMe SK hynix 512GB
 • Touchpad
 • UEFI dbx
No updatable devices
```

```
Nov 08 09:09:05 christoph-xps139300 systemd[1]: Starting Firmware update daemon...
Nov 08 09:09:05 christoph-xps139300 fwupd[124859]: 08:09:05:0372 FuPluginUefi         cannot find default ESP: no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
Nov 08 09:09:05 christoph-xps139300 fwupd[124859]: 08:09:05:0425 FuPluginDfu          0bda:58fe is missing download capability
Nov 08 09:09:05 christoph-xps139300 fwupd[124859]: 08:09:05:0820 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0: attribute vendor_name returned no data
Nov 08 09:09:05 christoph-xps139300 fwupd[124859]: 08:09:05:0827 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:0d.3/domain1/1-0: attribute vendor_name returned no data
Nov 08 09:09:07 christoph-xps139300 systemd[1]: Started Firmware update daemon.
Nov 08 09:20:15 christoph-xps139300 systemd[1]: fwupd.service: Succeeded.
```

```
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-e8292593-e66e-4878-b051-f152535ab130-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

hughsie, 2020-11-08:
Please see https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found -- It's likely you need to change the partition type to ESP.

blauch, 2020-11-08:
Sorry, I am a bit thin on this topic but was under the impression that I have EFI system partition set up correctly

```
 $ pacman -Ss udisk  
extra/udisks2 2.9.1-1 [installed]
    Disk Management Service, version 2
community/udiskie 2.2.0-2 [installed]
    Removable disk automounter using udisks
```

```
$ sudo /usr/bin/fwupdtool esp-list --verbose
13:27:05:0853 FuDebug              Verbose debugging enabled (on console 1)
13:27:05:0859 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4
13:27:05:0864 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
13:27:05:0868 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4
no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
```

```
$ efibootmgr -v
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0000,0004,0001,0003,0005
Boot0000* UEFI PC611 NVMe SK hynix 512GB NN03N424212403F1B 1	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,AC-E4-2E-00-05-41-56-AC)/HD(1,GPT,9b5051f9-5e5d-004a-a1c0-4981095443d4,0x800,0xfa000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0001* Linux Firmware Updater	HD(1,GPT,9b5051f9-5e5d-004a-a1c0-4981095443d4,0x800,0xfa000)/File(\EFI\manjaro\fwupdx64.efi)
Boot0002* Manjaro	HD(1,GPT,9b5051f9-5e5d-004a-a1c0-4981095443d4,0x800,0xfa000)/File(\EFI\Manjaro\grubx64.efi)
Boot0003* USB NIC (IPV4)	PciRoot(0x0)/Pci(0xd,0x0)/USB(1,0)/USB(1,0)/MAC(a0cec8daf676,0)/IPv4(0.0.0.00.0.0.0,0,0)N.....YM....R,Y.
Boot0004* Linux Firmware Updater	HD(1,GPT,9b5051f9-5e5d-004a-a1c0-4981095443d4,0x800,0xfa000)/File(\EFI\manjaro\fwupdx64.efi)
Boot0005* USB NIC (IPV6)	PciRoot(0x0)/Pci(0xd,0x0)/USB(1,0)/USB(1,0)/MAC(a0cec8daf676,0)/IPv6([::]:<->[::]:,0,0)N.....YM....R,Y.
```

hughsie, 2020-11-08:
> was under the impression that I have EFI system partition set up correctly

What did you use to partition your system?

blauch, 2020-11-08:
I set it up during installation. It turned out that it was just a missing flag. I use parted to add the flag `set 1 esp on`
Sorry to have bugged you with this and thanks for the hints. 
```
$ sudo parted 
GNU Parted 3.3
Using /dev/nvme0n1
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print                                                            
Model: PC611 NVMe SK hynix 512GB (nvme)
Disk /dev/nvme0n1: 512GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End    Size   File system  Name  Flags
 1      1049kB  525MB  524MB  fat32              boot, esp
 2      525MB   105GB  105GB  ext4
 3      105GB   512GB  407GB
```

```
$ sudo fwupdmgr update

Devices with no available firmware updates: 
 • Integrated Webcam HD
 • PC611 NVMe SK hynix 512GB
 • TPM 2.0
 • Touchpad
 • UEFI dbx
Devices with the latest available firmware version:
 • System Firmware
```

superm1, 2020-11-08:
@hughsie I wonder if we want to try to use a fallback if nothing has the esp flag set to look for fat32. That's what firmware does after all.

blauch, 2020-11-08:
@superm1 I second your point and reopened this issue. For me, it stopped working out of the blue but I was able to solve this issue by adding the `esp` flag my boot partition. I will leave it up to you guys to decide if you want to close it again, create a follow-up or leave as it.

hughsie, 2020-11-09:
> set to look for fat32

Is there any way this could go wrong? I'd much prefer to be explicit if there's any chance of us doing the "wrong" thing and accidentally reading files from a USB pen drive or something.

superm1, 2020-11-09:
I think relying upon the `HintSystem` property while paired with fat32 makes sense to avoid that circumstance.
http://storaged.org/doc/udisks2-api/latest/gdbus-org.freedesktop.UDisks2.Block.html#gdbus-property-org-freedesktop-UDisks2-Block.HintSystem

superm1, 2020-11-09:
But BTW there is nothing to stop someone from partitioning a USB disk with that GUID also..

hughsie, 2020-11-09:
> But BTW there is nothing to stop someone from partitioning a USB disk with that GUID also..

Ohh absolutely not, but it's not the default...

blauch, 2020-11-09:
from my user POV, I can point out that the documentation I was pointed to is incomplete. I had udisk installed but still got the warning and my fwupd was left broken. I agree that you should check if udisk is installed since you mention it in your documentation and make some assumptions based on that.

superm1, 2020-11-09:
USB disks seem to also set the `Ejectable` property.  So I think between `HintSystem` on the block device, `Ejectable` on the device node and the fat32 filesystem it seems like a pretty good heuristic to use if the GUID was set wrong on the ESP.  I don't really know yet how widespread this problem is going to be, but I'm worried.

For example I think that Dells' recovery media for factory shipped machines doesn't set that flag for it's ESP, only the boot flag.  I don't know about MS `diskpart.exe`.

superm1, 2020-11-12:
fixed in https://github.com/fwupd/fwupd/commit/56d816a53b8b23ebb7ee06fc3d3f5b71cebbcc42

closed 2020-11-12 by superm1