fwupd/fwupd #2568: Where do I find the checksums?

qoheniac, 2020-11-09:
```
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
$ fwupdmgr --help | grep " block-firmware`
  block-firmware [CHECKSUM]         Blocks a specific firmware from being installed
```
`fwupdmgr get-updates` shows two updates and I want to block them, because my laptop got stuck during update and I was forced to remove the battery to get it back to work. It seems like the update failed in a very early stage and nothing was changed but `gnome-software` keeps reminding me of those updates and I definitely want to do this a second time. I thought maybe `fwupdmgr block-firmware` would help but I don't know where to find those checksums.

superm1, 2020-11-09:
> because my laptop got stuck during update and I was forced to remove the battery to get it back to work

That sounds terrible.  Can you share more details, we might want that vendor to remove it from stable channel...

>  I thought maybe fwupdmgr block-firmware would help but I don't know where to find those checksums.

If you don't know the checksum in advance, I believe you can just send no argument and it should prompt you.

qoheniac, 2020-11-09:
Thank you for your help! Indeed I was asked for what device I want to block firmware updates. Blocking the update for *Embedded Controller* makes it disappear, but `fwupdmgr` still shows the update for *System Firmware* after blocking it.

As you asked for more information, so here is the output of `fwupdmgr get-updates` with no blocked firmware:
```
Devices with no available firmware updates: 
 • THNSF5256GPUK TOSHIBA
 • UEFI dbx
Devices with the latest available firmware version:
 • Intel Management Engine
20HGS01800
│
├─Embedded Controller:
│ │   Device ID:          03a6930fc4c833443a7502dde379b98b961c8c57
│ │   Current version:    0.1.20
│ │   Minimum Version:    0.0.1
│ │   Vendor:             DMI:LENOVO
│ │   GUIDs:              e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
│ │                       118d4b5a-8ef0-5dc6-b320-060dc79f2e36
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T470s Embedded Controller Update:
│       New version:      0.1.21
│       Remote ID:        lvfs
│       Summary:          Lenovo ThinkPad T470s Embedded Controller Firmware
│       License:          Proprietary
│       Size:             791.8 kB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo Ltd.
│       Flags:            is-upgrade
│       Description:      
│       Lenovo ThinkPad T470s Embedded Controller Firmware version 1.21.Fixed an issue where monitors attached to ThinkPad Docking Stations are
│       
│       disconnected when battery charging is reached to 100%.
│     
└─System Firmware:
  │   Device ID:          84149fb78009cf27cc1dd5520911f46f8792dbfe
  │   Current version:    0.1.32
  │   Minimum Version:    0.1.7
  │   Vendor:             LENOVO (DMI:LENOVO)
  │   GUIDs:              7a176688-0960-47ba-931b-7829849e8347
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │                       399bff3a-e2ad-5947-859c-565bad878b00
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T470s System Update:
  │     New version:      0.1.38
  │     Remote ID:        lvfs
  │     Summary:          Lenovo ThinkPad T470s System Firmware
  │     License:          Proprietary
  │     Size:             9.4 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Description:      
  │     Lenovo ThinkPad T470s System Firmware Version 1.38.Update includes a security fix.Updated the Diagnostics module to version 04.12.001.Fixed an issue where Force PXE boot by Intel AMT does not work.
  │   
  ├─ThinkPad T470s System Update:
  │     New version:      0.1.37
  │     Remote ID:        lvfs
  │     Summary:          Lenovo ThinkPad T470s System Firmware
  │     License:          Proprietary
  │     Size:             9.4 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Description:      
  │     Lenovo ThinkPad T470s System Firmware Version 1.37. Update includes security fixes.
  │   
  ├─ThinkPad T470s System Update:
  │     New version:      0.1.36
  │     Remote ID:        lvfs
  │     Summary:          Lenovo ThinkPad T470s System Firmware
  │     License:          Proprietary
  │     Size:             9.4 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Description:      
  │     Lenovo ThinkPad T470s System Firmware 1.36.Fix includes a security.Supported BIOS password authentication before entering into MEBx.Updated the Diagnostics module to version 04.11.000.
  │     
  │     Security issues fixed:
  │     
  │     • CVE-2019-14607
  │     • CVE-2019-0185
  │     • CVE-2019-0184
  │     • CVE-2019-0117
  │     • CVE-2019-0124
  │     • CVE-2019-0123
  │     • CVE-2019-0152
  │     • CVE-2019-0151
  │   
  ├─ThinkPad T470s System Update:
  │     New version:      0.1.35
  │     Remote ID:        lvfs
  │     Summary:          Lenovo ThinkPad T470s System Firmware
  │     License:          Proprietary
  │     Size:             9.4 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Vendor:           Lenovo Ltd.
  │     Flags:            is-upgrade
  │     Description:      
  │     Lenovo ThinkPad T470s System Firmware Version 1.35.Update includes a security fix.
  │   
  └─ThinkPad T470s System Update:
        New version:      0.1.34
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad T470s System Firmware
        License:          Proprietary
        Size:             9.4 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Flags:            is-upgrade
        Description:      
        Lenovo ThinkPad T470s System Firmware Version 1.34.Fix security issue,fix system might hang at post with some KVM devices connected,fix in legacy mode from Hybrid dock network boot to system might hang with USB UEFI BIOS Support Disabled.
```

I saw other reports of that problem on the internet, for example [here](https://www.reddit.com/r/thinkpad/comments/bwzsoe/thinkpad_t470s_firmware_update_hanging/).

superm1, 2020-11-09:
OK thanks.  I'll kick this over to the Lenovo bucket and hopefully someone there can help with pulling the firmware back or investigating the problem.