fwupd/fwupd #2572: get-updates exits with non-zero exit code when no updates are available

dead10ck, 2020-11-11:
**Describe the bug**
It appears at some point between versions 1.4.5 and 1.5.1, the behavior of fwupdmgr changed so that when no updates are available, it exits with a non-zero exit code, which typically signals a failure of the command to execute properly.

**Steps to Reproduce**
Run `fwupdmgr get-updates` when no updates are available.

```
❯ fwupdmgr get-updates
Devices with no available firmware updates:
 • ThinkPad Thunderbolt 3 Dock
 • USB3.0 Hub
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
 • Intel Management Engine
 • LENSE20256GMSP34MEAT2TA
 • System Firmware
No updates available for remaining devices

❯ echo $?
2
```

**Expected behavior**
Non-zero exit codes are for signalling that a command failed, so simply having no available updates should still be a 0 exit code.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20KGS3Y900
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:          529debb663f8428b9e6a702140e92658efe4fe97
│     Current version:    33.00
│     Vendor:             Lenovo (TBT:0x0108)
│     GUIDs:              fab08d10-6524-5cb6-ab20-6d365c812f9e
│                         0f511908-5c9f-511e-9099-06823ec33614
│                         df675f10-53ac-59ca-bddd-0a86ee492920
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Thunderbolt host controller:
│     Device ID:          e559faa8524cb4bfeccda9adf549d3cf476f699a
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    43.00
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              e69c694c-de9f-5f9a-8d30-9b695ab4744b
│                         74e58157-171f-57bc-a660-bf467ded54cb
│                         5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                         b8360cb5-b152-506a-9990-49d7bbbc3b88
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:    184.79.3722
│     Minimum Version:    184.79.3722
│     Vendor:             DMI:LENOVO
│     GUIDs:              42a0a96e-c9f3-438f-9687-7826be33e4ce
│                         375afb87-ea51-5a6e-96d5-c35cc342dc65
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000e0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         54bdc429-0179-5019-9cb6-8a343324c97c
│     Device Flags:       • Internal device
│   
├─LENSE20256GMSP34MEAT2TA:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express Solid State Drive
│     Current version:    2.8.8341
│     Vendor:             Lenovo (NVME:0x17AA)
│     GUIDs:              5bd8122b-a00c-5275-82cc-f17916b4bde9
│                         535f112b-3a90-5dde-8567-919cb66d56b9
│                         910b893d-088f-5d8b-bb6a-fc2a8ec33bfe
│                         687aa28d-fcb5-5325-b011-d2d27254b967
│                         add1cbe4-4916-5115-8202-9a3e9e69e64e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          1c53551e7da69d896138fac1ae131c83ad46d923
│ │   Current version:    0.1.50
│ │   Minimum Version:    0.0.1
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─Intel AMT [unprovisioned]:
│       Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
│       Summary:          Hardware and firmware technology for remote out-of-band management
│       Current version:  11.8.79.3722
│       Bootloader Version:11.8.79.3722
│       Vendor:           Intel Corporation
│       GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
│       Device Flags:     • Internal device
│     
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:    7.63.13.6400
│ │   Vendor:             Infineon (TPM:IFX)
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       5eebb112-75ad-5536-b173-a11eb3399402
│ │                       ddf995da-1b32-5a8a-bc1b-8d5af4b38b51
│ │                       6d81ab63-db2e-50ac-934f-6be9accf5e02
│ │                       301555de-680d-5ddc-b995-7553fc9138f1
│ │   Device Flags:       • Internal device
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│     
├─UHD Graphics 620:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              2920038a-3772-52ef-8f3d-92c2fb1ac115
│                         0284ca6a-a235-5433-b190-1de21c753494
│                         fed2efa4-7045-55a1-b4fc-b29283d59fe5
│                         8b72a10c-1279-5f8e-a28a-34fb11a58240
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─USB3.0 Hub:
      Device ID:          e07e3b992d5e2cfbe14b101ab8c6f6932f83549f
      Summary:            USB 2.x Hub
      Current version:    13.101
      Vendor:             VIA Labs, Inc. (USB:0x2109)
      Install Duration:   10 seconds
      GUIDs:              de3f0846-af25-5b5e-8a03-b48f2c059707
                          7aaf809e-a3f1-5beb-8a28-09e44471534b
                          f1175e69-cf09-55de-b71b-da24d8f331d4
                          7eafc3ff-3eaa-56ec-9ec9-aeeca463e683
                          cd3a7a5a-f5cf-546e-8f18-0a758327aa3e
                          d812d765-cb6d-5486-8a25-c61bf4c8dfb0
      Device Flags:       • Updatable
                          • Cryptographic hash verification is available
    
```

**Additional questions**
- Operating system and version: Fedora 32
- Have you tried rebooting? N/A
- Is this a regression? no

hughsie, 2020-11-11:
We use EXIT_NOTHING_TO_DO as RC=2 to distinguish failure from, well, having nothing to do.

dead10ck, 2020-11-11:
@hughsie this is a breaking change. It should have been communicated via the release notes and a major version increment.

But besides this, almost all shells everywhere treat non-zero exit code as failure. Having nothing to do is not a failure condition. With very few exceptions (grep being a notable one), most CLI tools do not use non-zero exit codes to signal non-failure conditions. Semantic return codes have their uses, but they do not play well with scripts, because they have to be explicitly coded around to be of any real use. That should have been an opt-in feature.

hughsie, 2020-11-11:
> but they do not play well with scripts

I don't think you should be using `fwupdmgr` in shell scripts, it's designed to be interactive. Is `fwupdagent` what you want?

dead10ck, 2020-11-11:
I understand that, my script just checks for updates. It's meant for human consumption, so I use `fwupdmgr` so I can see the output in a readable form, but it's semi-automated. But regardless, you have a tool out there in the world, and this change breaks assumptions people have made about the behavior of the tool. It should have been communicated, and it should have either been opt-in or accompanied by a major version increment.

Also, if `fwupdmgr` is designed to be interactive, then why make this change to begin with? Human beings don't care what the return code is.

hughsie, 2020-11-11:
> and this change breaks assumptions people have made about the behavior of the tool

I'm sorry this broke your tool.

> it should have either been opt-in or accompanied by a major version increment

Patch welcome for the opt-in. Maybe we drop the rc=2 if we detect the console is non-interactive or something?

> Human beings don't care what the return code is.

From memory, the CI did.

superm1, 2020-11-11:
This behavior of returning zero was actually a bug from what we expected.  If you look back in the installed tests history, you can see that back in 0.9.5 we had expected to get 2 when there were no updates: https://github.com/fwupd/fwupd/commit/8a870d0399468916fffac684df60718eba5e4408

The behavior was adjusted to match these expectations in 1.5.0: https://github.com/fwupd/fwupd/commit/e8946a74d16e26a2fcd47f94f7ca5a439d129a4c

superm1, 2020-11-11:
But so in _general_ we don't have a stable expectation for `fwupdmgr`.  If you want to script checking for updates or looking at output you should generally be using `fwupdagent`.  Before that change was enacted you would need to manually look at stdout and stderr to tell whether there are no updates or no updatable devices (even though the net outcome was the same - no updates available).

dead10ck, 2020-11-11:
I'm a bit confused. You guys are telling me that `fwupdmgr` is not intended for use in scripts, and yet the commit message for that change explicitly says that the intention behind the change was "to allow other tools to depend on it when checking for updates rather than reading the output." To me, this signals awareness and intention to use `fwupdmgr` in an automated way.

In any case, I'm actually just learning of `fwupdagent`'s existence. It isn't in this repo's readme, and it's not mentioned in the man page for `fwupdmgr`. If this tool is intended for the scripting use case, the advertising could be improved.

But just looking at the man-page, it's hard to tell what the difference is, besides JSON output and the return code behavior. Is that the only difference? It also looks like it doesn't have the `refresh` command, so I'm not sure it's a replacement for an automated script that checks for updates.

hughsie, 2020-11-11:
> it's not mentioned in the man page for fwupdmgr

Yes, we should fix that.

> It also looks like it doesn't have the refresh command

I have a local branch here that implements it; I could dust it off if that would be helpful. I wasn't sure what JSON I should show for success or failure, or if we should skip remotes or just fail the entire thing on the first error.

superm1, 2020-11-11:
> You guys are telling me that fwupdmgr is not intended for use in scripts

The reason to not use it in scripts is that we don't have guarantees on it's output behavior.  That's part of why we made `fwupdagent` in the first place.  `fwupdmgr` definitely has flags for non-interactive use (and will look for a tty).  And we do test it's output in CI.  If you look back at `get-updates` output over the last few versions you'll see that there has been different groupings of how to show devices with updates, and flirting with tree output and the like.  It would be terrible if you tried to grep it and expect the same thing from version to version.  However if you're relying upon `fwupdagent` you would always have same JSON to work with.

> I have a local branch here that implements it; I could dust it off if that would be helpful. I wasn't sure what JSON I should show for success or failure, or if we should skip remotes or just fail the entire thing on the first error.

I think it's probably a good idea.  To @dead10ck's point "if we want people not to script `fwupdmgr`" then we should make sure that all the use cases for `fwupdagent` that you would "normally" script are fully fleshed out.

dead10ck, 2020-11-11:
I hope this isn't taken the wrong way, because I have a lot of respect and appreciation for the work you guys do on this project. But it seems like there could be improvement around the tooling story. The reality is people are going to use your tools in scripts. If you don't want your users to use `fwupdmgr` in scripts, then there needs to be a clear and well-communicated story around what to do for the automated use case. It needs to be front-and-center in the documentation, and it needs to implement all the major expected use cases people have for automated tooling; critical use cases can't just be sitting around in a local unpublished branch.

superm1, 2020-11-11:
No offense taken.  We're often knee deep in the details, and it's not obvious what someone who doesn't look at this stuff all the time fixates on.  Would you mind to help to submit a PR to help improve man pages, `README.md`, wiki, anything that sends confusing or inconsistent messages on this?

dead10ck, 2020-11-11:
@superm1 indeed, I understand that difficulty. I can't promise and commit to anything, as I have a full-time job and a young child to take care of, but I will keep this in mind next time I have some spare time!

closed 2020-11-11 by superm1

Dmole, 2021-08-13:
If `fwupdagent` is returning zero for "there are updates" and "there are no updates",
and `fwupdmgr` is returning zero for "there are updates" and non-zero for anything else,
then I'd expect a lot of scripts to just `fwupdmgr get-updates && email_me.sh`,
because the alternative would be `if [ "$(fwupdagent get-updates | jq ".Devices | length")" != "0" ] ; then email_me.sh; fi`

superm1, 2021-08-13:
FYI, the latest version of fwupd has deprecated `fwupdagent` and added json support directly to `fwupdmgr`.  You should be able to use that instead now.

Dmole, 2021-08-13:
lol so re-write scripts every fwupd update.
Ubuntu 20.04 just got the return 2 but not the json support on fwupdmgr so I expect this crazy state of fwupd to persist up to 9 years.

superm1, 2021-08-13:
> lol so re-write scripts every fwupd update.
> Ubuntu 20.04 just got the return 2 but not the json support on fwupdmgr so I expect this crazy state of fwupd to persist up to 9 years.

Return code of 2 means "nothing to do". It's supposed to be so you can distinguish failure versus success but nothing needs to happen.

PR #2573 (2020-11-11): Add a section to man page on scripting
Fixes: #2572

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [x] Documentation
commit cceeb75c (2020-11-11): Add a section to man page on scripting

Fixes: #2572