fwupd/fwupd #2575: Connection terminated unexpectedly

uriva, 2020-11-12:
```bash
$ sudo fwupdmgr update
• Thunderbolt Controller has no available firmware updates
• Integrated Webcam HD has no available firmware updates
• SSDPEMKF512G8 NVMe INTEL 512GB has no available firmware updates
Upgrade available for System Firmware from 1.1.3 to 1.4.0
XPS 15 9500 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading…             [***************************************] 10 minutes remaining…
Failed to download https://fwupd.org/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab: Connection terminated unexpectedly
```

**fwupd version information**
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5

uriva, 2020-11-12:
If I use my browser it actually starts downloading the file and stops due to network error shortly after.

superm1, 2020-11-12:
Is this the HTTP 502 problem with LVFS @hughsie ?

hughsie, 2020-11-12:
> Is this the HTTP 502 problem

No that was the redis endpoint being invalid; that only affected stuff that triggered cron jobs on the server. That download worked for me, both in the browser and using curl -- could it be a transient network issue perhaps?

uriva, 2020-11-12:
I tried a bunch of times, then used my windows to update it...

superm1, 2020-11-12:
CDN problem with a single end point I wonder?

At least for me:
```
$ wget https://fwupd.org/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab
--2020-11-12 14:04:11--  https://fwupd.org/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab
Resolving fwupd.org (fwupd.org)... 54.187.171.134, 54.187.115.215
Connecting to fwupd.org (fwupd.org)|54.187.171.134|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 23319165 (22M) [application/octet-stream]
Saving to: ‘91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab’

91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Preci 100%[==========================================================================================================================================================================>]  22.24M  6.65MB/s    in 3.7s

2020-11-12 14:04:15 (6.07 MB/s) - ‘91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab’ saved [23319165/23319165]
```

So if I try both of the download end points I can find manually:
```
$ dig fwupd.org

; <<>> DiG 9.11.3-1ubuntu1.7-Ubuntu <<>> fwupd.org
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 2101
;; flags: qr rd ra; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;fwupd.org.                     IN      A

;; ANSWER SECTION:
fwupd.org.              13      IN      A       54.187.171.134
fwupd.org.              13      IN      A       54.187.115.215

;; Query time: 1 msec
;; SERVER: 192.168.2.1#53(192.168.2.1)
;; WHEN: Thu Nov 12 14:04:58 STD 2020
;; MSG SIZE  rcvd: 70

$ wget https://54.187.115.215/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab --no-check-certificate
--2020-11-12 14:05:34--  https://54.187.115.215/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab
Connecting to 54.187.115.215:443... connected.
    WARNING: certificate common name ‘fwupd.org’ doesn't match requested host name ‘54.187.115.215’.
HTTP request sent, awaiting response... 200 OK
Length: 23319165 (22M) [application/octet-stream]
Saving to: ‘91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab.1’

91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Preci 100%[==========================================================================================================================================================================>]  22.24M  9.41MB/s    in 2.4s

2020-11-12 14:05:37 (9.41 MB/s) - ‘91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab.1’ saved [23319165/23319165]

$ wget https://54.187.171.134/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab --no-check-certificate
--2020-11-12 14:05:50--  https://54.187.171.134/downloads/91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab
Connecting to 54.187.171.134:443... connected.
    WARNING: certificate common name ‘fwupd.org’ doesn't match requested host name ‘54.187.171.134’.
HTTP request sent, awaiting response... 200 OK
Length: 23319165 (22M) [application/octet-stream]
Saving to: ‘91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab.2’

91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Preci 100%[==========================================================================================================================================================================>]  22.24M  3.54MB/s    in 6.8s

2020-11-12 14:05:57 (3.25 MB/s) - ‘91d55136b59b13d233ddb1d757d698768872aaebe0d880dce7b8b52d1db963f4-Precision_5550_1.4.0.cab.2’ saved [23319165/23319165]

```

closed 2020-12-20 by stale[bot]