fwupd/fwupd #2576: UEFI warning always printed even if uefi plugin is not built on ppc64

gyakovlev, 2020-11-12:
**Describe the bug**
the following warning is always printed on `ppc64` platform.
```
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
```
of course `fwupd` is built without uefi support.
I tried also adding `DisabledPlugins=test;invalid;uefi` to config file, just in case, warning is still printed on every `fwupdmgr` invocation.

**Steps to Reproduce**
Steps to reproduce the behavior.
run `fwupdmgr`, warning printed every time despite efi support is not even built in.

**Expected behavior**
No warning printed.

**fwupd version information**
```shell
$ fwupdmgr --version
client version: 1.5.1
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
installed via emerge (gentoo package manager)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version:
gentoo rolling ppc64le
- Have you tried rebooting?
irrelevant, but yes
- Is this a regression?
I haven't checked previous versions.

gyakovlev, 2020-11-13:
with @meklort 's help we figured out it's bios plugin

`DisabledPlugins=test;invalid;bios` does help.
I'll disable bios plugin build on ppc64 hosts in gentoo packages then.

meklort, 2020-11-13:
In the source code, it's caused by https://github.com/fwupd/fwupd/blob/master/plugins/bios/fu-plugin-bios.c#L56

gyakovlev, 2020-11-13:
actually, looks like there's no meson knob to disable bios plugin build. So I can't just opt-out on ppc without editing config =)

commit 4239ca31 (2020-11-12): Fall back to FAT32 internal partitions for detecting ESP

Fixes #2576
commit 676d30d5 (2020-11-12): Fall back to FAT32 internal partitions for detecting ESP

Fixes #2576
PR #2574 (2020-11-12): Fall back to FAT32 internal partitions for detecting ESP
Fixes #2576

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 56d816a5 (2020-11-12): Fall back to FAT32 internal partitions for detecting ESP

Fixes #2576
closed 2020-11-13 by hughsie