fwupd/fwupd #2578: regression with fwupdmgr switch-branch

meklort, 2020-11-13:
**Describe the bug**
Commit bf43978dfa1eeedf058dd3adc4d029b94d3bec59 introduces a regression when trying to flash the bcm5719 oss-firmware branch to the Talos II.

With the previous commit, `fwupdmgr switch-branch` properly detects that lvfs-testing has valid firmware that can be enabled.
When  bf43978dfa1eeedf058dd3adc4d029b94d3bec59 is used, "No supported devices" is reported instead.

**Steps to Reproduce**
- Flash the proprietary bcm5719 firmware
- Run Enable lvfs-testing
- Run fwupdmgr switch-branch

**Expected behavior**
Branch switching should be allowed, valid firmware should be found.

**fwupd version information**
built from latest soruce code. 

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
meklort@talos2 ~/g/f/build ((bf43978d…)) [1]> fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: Plugin dependencies missing
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:failed-open for more information.
T2P9D01 REV 1.00
│
├─NetXtreme BCM5719 Gigabit Ethernet PCIe:
│     Device ID:          7de5ffdca08fa52d95fd4bb42aa5d07a4b35d2dd
│     Current version:    1.43
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              ff247529-8b0a-5db7-98ef-8746aa018be6
│                         5d8b12bf-1973-58fc-8b31-3e50fe31954d
│                         3800738e-1413-5a18-846a-8782f5cc8c44
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device will backup firmware before installing
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          9e517c3cc54687578aa78274c9644219970a5b5b
│     Current version:    1.31
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              2d5d8db9-60a0-5a17-afa6-7fbfb9fe7cab
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         3800738e-1413-5a18-846a-8782f5cc8c44
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device will backup firmware before installing
│   
├─82574L Gigabit Network Connection (Gigabit CT Desktop Adapter):
│     Device ID:          4b38a5e5245b7a0ad5adfab23f47009b7be9db3a
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              9ddc6d96-8f9e-51e5-b204-1719d4c8e449
│                         7f5a805d-2c1a-5efa-8cd0-5dd66ebe05e2
│                         6af770c4-de01-5039-8737-1055825b763e
│                         64593cc0-2efd-529d-9bf4-7084542c809d
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NVMe SSD Controller SM951/PM951:
│     Device ID:          694634fa85740a2d600f2ce5e532b3d7ea467621
│     Current version:    01
│     Vendor:             Samsung Electronics Co Ltd (PCI:0x144D)
│     GUIDs:              81754dda-78ad-52c6-ab2a-a0a8547674af
│                         65c71c21-48e0-579e-9e58-bdb00d03361c
│                         f620cb9c-fedd-5a48-a91b-563c40890059
│                         f8775f26-c040-554b-b27c-bea293913427
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          4a5daefa43523fb8c39e2a5e13daa398ff748e7d
│     Current version:    01
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              2d5d8db9-60a0-5a17-afa6-7fbfb9fe7cab
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         3800738e-1413-5a18-846a-8782f5cc8c44
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          6fd1d0e4a2ed0978fcca10eb18d340423581f70a
│     Current version:    01
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              2d5d8db9-60a0-5a17-afa6-7fbfb9fe7cab
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         3800738e-1413-5a18-846a-8782f5cc8c44
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          f22ea00fe4389a36e05c1960512d4bbda758aa5c
│     Current version:    01
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              2d5d8db9-60a0-5a17-afa6-7fbfb9fe7cab
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         3800738e-1413-5a18-846a-8782f5cc8c44
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─NetXtreme BCM5719 Gigabit Ethernet PCIe (4-port 1Gb Ethernet Adapter):
│     Device ID:          8f6d04a39402a5d5e97627d001543725e0fef566
│     Current version:    01
│     Vendor:             Broadcom Inc. and subsidiaries (PCI:0x14E4)
│     GUIDs:              2d5d8db9-60a0-5a17-afa6-7fbfb9fe7cab
│                         ec5b8a9e-973b-58cc-935b-8322fabaebe9
│                         3800738e-1413-5a18-846a-8782f5cc8c44
│                         30fe13b6-aa73-5d8c-a19f-c7b600f0117a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─Samsung SSD 950 PRO 512GB:
      Device ID:          7292ecd2edec3b738a86392d3e35087ae7b107b5
      Summary:            NVM Express Solid State Drive
      Current version:    1B0QBXX7
      Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
      GUIDs:              b586caf2-ac7e-5a07-8630-dd289d8fd3f4
                          484eb66f-4175-5499-8dc9-a8cc651e8450
                          cce2386f-6d68-5f85-9f3f-ebead9288696
                          77bb4022-9503-586c-99b7-c8a64502da30
                          c90f3aaf-0ddf-5728-8efe-f3546765962b
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: Fedora 33
- Have you tried rebooting? Yes
- Is this a regression? Yes

meklort, 2020-11-13:
Note that the install command swill works fine with the .cab file, only switch-branch appears to be impacted.

hughsie, 2020-11-13:
@meklort is there any Dell KH08P firmware on the LVFS now? If you can upload some (or two different versions) even just to your embargo remote I can add a device test for this so that this kind of regression doesn't happen in the future.

closed 2020-11-13 by hughsie

meklort, 2020-11-14:
I haven't uploaded any firmware for it (and don't plan on officially supporting it) purely because I don't expect to have feature parity any time soon with it (from what I can tell, the Dell card supports NC-SI over SMBus, which is not implement or needed for the RCS hardware).

I'll upload a build just to the embargo remote for testing only.

hughsie, 2020-11-14:
> I'll upload a build just to the embargo remote for testing only.

That would be awesome, thanks -- when you do please share me the URL. I'll do the same for the nonfree code to my own account and then I can set up some nonfree -> old yours -> new yours -> nonfree testing in the device tests.

Richard.