fwupd/fwupd #2581: fwupd 1.5.1: fwupdplugin-self-test falls after fu_efivar_space_used on ppc64le and tmpfs

RiderALT, 2020-11-13:
**Describe the bug**
If buildroot located on tmpfs and arch is ppc64le, the fwupdplugin-self-test failed with assertion on 
fu_efivar_func:
```# FuCommon-DEBUG: reading /usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/data/tests/builder/firmware.tar with 10240 bytes
(/usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/libfwupdplugin/fwupdplugin-self-test:1530724): FuCommon-DEBUG: 10:43:07.418: reading /usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/data/tests/builder/firmware.tar with 10240 bytes
ok 25 /fwupd/common{firmware-builder} # SKIP User namespaces not supported in container
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/libfwupdplugin/fwupdplugin-self-test:1530724): GLib-DEBUG: 10:43:07.419: setenv()/putenv() are not thread-safe and should not be used after threads are created
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/libfwupdplugin/fwupdplugin-self-test:1530724): GLib-DEBUG: 10:43:07.419: setenv()/putenv() are not thread-safe and should not be used after threads are created
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/libfwupdplugin/fwupdplugin-self-test:1530724): GLib-DEBUG: 10:43:07.419: setenv()/putenv() are not thread-safe and should not be used after threads are created
ok 26 /fwupd/common{kernel-lockdown}
Bail out! ERROR:../libfwupdplugin/fu-self-test.c:1853:fu_efivar_func: assertion failed (total == 0x2000): (131072 == 8192)
--- stderr ---
**
ERROR:../libfwupdplugin/fu-self-test.c:1853:fu_efivar_func: assertion failed (total == 0x2000): (131072 == 8192)
-------

Full log written to /usr/src/RPM/BUILD/fwupd-1.5.1/ppc64le-alt-linux/meson-logs/testlog.txt
```
```
[builder@localhost fwupd-1.5.1]$ du -s data/tests/efi/efivars/
128K    data/tests/efi/efivars/

[builder@localhost fwupd-1.5.1]$ ls -al data/tests/efi/efivars/
total 128
drwxr-xr-x 2 builder builder 100 Nov  2 14:52 .
drwxr-xr-x 3 builder builder  60 Nov  2 14:52 ..
-rw-r--r-- 1 builder builder   0 Nov  2 14:52 BootNext-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 builder builder   1 Nov  2 14:52 SecureBoot-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 builder builder 230 Nov  2 14:52 fwupd-ddc0ee61-e7f0-4e7d-acc5-c070a398838e-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
```

I think this is due to the large page size on this architecture.
If copy the fwupd tree to a regular disk, this test stops failing.

this bug is not reproduced on x86_64 architecture.

superm1, 2020-11-13:
Exact same failure is happening in Debian on 1.5.1.
Full build log: https://buildd.debian.org/status/fetch.php?pkg=fwupd&arch=ppc64el&ver=1.5.1-1&stamp=1605286432&raw=0

```
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:37.271: creating firmware.bin with world
ok 22 /fwupd/common{cab-error-size}
# FuCommon-DEBUG: running '/<<PKGBUILDDIR>>/data/tests/spawn.sh test'
(/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): FuCommon-DEBUG: 16:53:37.272: running '/<<PKGBUILDDIR>>/data/tests/spawn.sh test'
# DEBUG: got 'this is a test'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:37.274: got 'this is a test'
# DEBUG: got 'this is another line1'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:38.278: got 'this is another line1'
# DEBUG: got 'this is another line2'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:38.278: got 'this is another line2'
# DEBUG: got 'this is another line3'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:38.278: got 'this is another line3'
# DEBUG: got 'this is another line4'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:38.278: got 'this is another line4'
# DEBUG: got 'done!'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:39.282: got 'done!'
ok 23 /fwupd/common{spawn)
# FuCommon-DEBUG: running '/<<PKGBUILDDIR>>/data/tests/spawn.sh test'
(/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): FuCommon-DEBUG: 16:53:39.283: running '/<<PKGBUILDDIR>>/data/tests/spawn.sh test'
# DEBUG: got 'this is a test'
** (/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): DEBUG: 16:53:39.285: got 'this is a test'
ok 24 /fwupd/common{spawn-timeout)
# FuCommon-DEBUG: reading /<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/data/tests/builder/firmware.tar with 10240 bytes
(/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): FuCommon-DEBUG: 16:53:39.335: reading /<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/data/tests/builder/firmware.tar with 10240 bytes
ok 25 /fwupd/common{firmware-builder} # SKIP User namespaces not supported in container
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): GLib-DEBUG: 16:53:39.335: setenv()/putenv() are not thread-safe and should not be used after threads are created
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): GLib-DEBUG: 16:53:39.335: setenv()/putenv() are not thread-safe and should not be used after threads are created
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
(/<<PKGBUILDDIR>>/obj-powerpc64le-linux-gnu/libfwupdplugin/fwupdplugin-self-test:20340): GLib-DEBUG: 16:53:39.335: setenv()/putenv() are not thread-safe and should not be used after threads are created
ok 26 /fwupd/common{kernel-lockdown}
Bail out! ERROR:../libfwupdplugin/fu-self-test.c:1853:fu_efivar_func: assertion failed (total == 0x2000): (131072 == 8192)
--- stderr ---
**
ERROR:../libfwupdplugin/fu-self-test.c:1853:fu_efivar_func: assertion failed (total == 0x2000): (131072 == 8192)
------
```

hughsie, 2020-11-14:
I think we can change that to be `>=` -- we just want to see it's adding up the on-disk sizes -- right?

superm1, 2020-11-14:
seems fine to me

closed 2020-11-15 by hughsie