fwupd/fwupd #2584: OUI quirk required: Dogfish mSATA and Mushkin ECO3 ATA devices

AJCxZ0, 2020-11-13:
Running _fwupdtool_ gives error,

```shell
$ sudo fwupdtool refresh
Loading…                 [-                                      ]ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 

OUI quirk required, please see https://github.com/fwupd/fwupd/wiki/ATA-Disk:-OUI-Quirk-Required!
---
[DeviceInstanceId=OUI\888914]
Vendor = FIXME
VendorId = ATA:UNKNOWN
---                                                                Loading…                 [***************************************]
```

The system has two SATA drives: a Mushkin ECO3 120GB SATA SSD ([MKNSSDE3120GB](https://www.poweredbymushkin.com/Home/index.php/products2/item/52-eco3/1256-eco3-120gb)) and a Dogfish mSATA 120GB SSD ([DOGFISH_SSD_120GB](https://tc-dogfish.com/collections/ssd/products/oem-ssd-state-drive-2-5inch-sata3)).

```
[DeviceInstanceId=OUI\888914]
Vendor = FIXME
VendorId = ATA:UNKNOWN
```

This is consistently reproducible running any _fwupdtool_ command.

This looks like expected behaviour since device **888914** is associated with vendor **All Components Incorporated** in [oui.txt](http://standards-oui.ieee.org/oui.txt) and that vendor does not appear in [pci.ids](https://pci-ids.ucw.cz/v2.2/pci.ids), however I would like the devices to both be recognised and firmware status reported.

```shell
$ fwupdmgr --version
client version:	1.5.1
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.1
```

Installed with _fwupd_ on Fedora 33.

```shell
$ fwupdmgr get-devices --show-all-devices
GB-BXA8-5557
│
├─AMD A8-5557M APU with Radeon(tm) HD Graphics   :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         317855fd-1273-5591-b663-c3be66e22c5f
│                         e1e4ef13-1344-5cfd-8765-214004f6d721
│                         563a0a0b-2247-5e02-9a2e-f434956376d1
│     Device Flags:       • Internal device
│   
├─DOGFISH SSD 120GB:
│     Device ID:          87778ff0311de4e3d207b5c42c22414f113fe6ac
│     Summary:            ATA Drive
│     Current version:    R0831B0
│     Update Error:       No vendor ID set
│     GUIDs:              6733e1aa-9111-521b-b950-b267407a7908
│                         a76d27ee-61b7-5729-a2ce-9488edc239db
│                         665dcbe4-9273-55a8-bc35-189f5a358187
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MKNSSDE3120GB:
│     Device ID:          a26cb0d73c131a0c8a1fcf3155aec8dfec312c62
│     Summary:            ATA Drive
│     Current version:    O1126A
│     Update Error:       No vendor ID set
│     GUIDs:              c412ef5c-5ac6-50c8-878c-1ce86250a59d
│                         b484bd6a-4236-5b2b-8746-1fa2b5fc9cac
│                         3cffb78c-f0a0-5f55-8e40-5535137af6c8
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Richland [Radeon HD 8550G]:
│     Device ID:          17076870bcf7a84a9c8e999d7e54e39b446032bb
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              9c2bd560-6056-5b69-ab56-8d64f5aaf2bc
│                         fb276dfc-4c86-5eff-a30d-63d4a9948608
│                         efc0ebb0-1375-5ecb-bd64-8bbf96a19b5e
│                         20d47ff2-7bcd-527b-a444-e9ffbb0a5ac4
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│     Device ID:          1d70a31dd3ff0941761550a38436fc990c6c3ff0
│     Current version:    6
│     Minimum Version:    6
│     Vendor:             GIGABYTE (DMI:American Megatrends Inc.)
│     GUIDs:              7039436b-6acf-433b-86a1-368ec2ef7e1f
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         733edd7c-75ee-5db0-8a0a-3138c35f755c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│   
├─VMM3200:
│     Device ID:          cd125b6d734f620f390f87d2c20b42da5b17316d
│     Summary:            Multi-Stream Transport Device
│     Current version:    0.00.00
│     Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB)
│     GUIDs:              1beb64bd-6dc8-5db2-84b9-f32964abc876
│                         b57f0a20-424c-51e0-b91b-575f3d74ed3a
│                         d6072785-6fc0-5f83-9d49-11376e7f48b1
│     Device Flags:       • Updatable
│   
└─xDuoo USB Audio 2.0:
      Device ID:          28fe7919e3b6101195936a2b7486d9f5a54afe92
      Current version:    6.152
      Vendor:             XMOS (USB:0x20B1)
      GUIDs:              aa4da743-3540-5818-a917-0ff12840986d
                          27a87fc5-f400-5dee-a47d-0b749df47742
      Device Flags:       • Updatable
```

Fedora release 33, rebooted, not observed in previous uses of fwupdtool on the same hardware.

hughsie, 2020-11-13:
Could you do a PR for the missing OUI quirks please?

AJCxZ0, 2020-11-14:
@hughsie I'd be happy to do so. Can you point me to a good reference example on which to base it?

hughsie, 2020-11-14:
Excellent! Does https://github.com/fwupd/fwupd/wiki/ATA-Disk:-OUI-Quirk-Required give enough guidance, or is it still too vague? TIA!

AJCxZ0, 2020-11-14:
Thank you for the documentation reference. That's the document at which I started, based on which I opened the issue.

> submit a pull request with the oui quirk added in the right place in plugins/ata/ata.quirk

I'm looking for a reference pull request where I can see what exactly to add where and what information I need in order to determine the _oui quirk_ addition in this case where I have two ATA devices and the `[DeviceInstanceId=OUI\888914]` not only refers to just one, but the vendor name doesn't match either (though I would guess it's the Dogfish).

hughsie, 2020-11-16:
> though I would guess it's the Dogfish

I don’t know either to be honest. Could you shutdown, remove one, boot, and see if the message persists? Then you know which one it is. Then you can follow the wiki I think.

closed 2020-12-24 by stale[bot]