fwupd/fwupd #2586: modules-load: should be installed to $prefix while $sysconfdir should be left to local admin

tinywrkb, 2020-11-15:
The build system installs the configs to load the relevant kernel modules into `$sysconfdir/modules-load.d` (or `/etc/modules-load.d`).  
Per the `modules-load.d.5` man page, this path should be left for the local administrator to configure the system, that's including overriding by creating symlinks to /dev/null with the same filenames as the configs.  
I suggest to change the install path to `$prefix/lib/modules-load.d` (or /usr/lib/modules-load.d`).

On my system `/etc/modules-load.d` is a symlink to `/var/etc/modules-load.d`, everything in `/var` is managed by tmpfiles.d and not the package manager (it's even a little more complex but I'll stop here) so I need to work around this issueand keep track of fwupd's configs which I would appreciate not having to do.  
fwupd is the only offender that puts configs in `/etc/modules-load.d`.

hughsie, 2020-11-15:
Sounds fine to me. Could you do a PR with the two line change please.

tinywrkb, 2020-11-15:
Created #2587

tinywrkb, 2020-11-16:
Closing as #2591 was merged.

closed 2020-11-16 by tinywrkb