fwupd/fwupd #2600: fwupd_client_get_devices never returns

aleixpol, 2020-11-18:
**Describe the bug**
We call fwupd_client_get_devices() from a worker thread. It never returns leaving the thread dangling.
There is no errors or warnings so I don't really know how to take an action.

```
#0 syscall() at /usr/lib/libc.so.6
#1 g_cond_wait() at /usr/lib/libglib-2.0.so.0
#2 ??() at /usr/lib/libglib-2.0.so.0
#3 g_main_loop_run() at /usr/lib/libglib-2.0.so.0
#4 fwupd_client_connect() at /home/apol/devel/playground/fwupd/libfwupd/fwupd-client-sync.c:98
#5 fwupd_client_get_devices() at /home/apol/devel/playground/fwupd/libfwupd/fwupd-client-sync.c:136
#6 FwupdBackend::checkForUpdates()::$_6::operator()() const() at /home/apol/devel/frameworks/discover/libdiscover/backends/FwupdBackend/FwupdBackend.cpp:419
#7 QtConcurrent::StoredFunctorCall0<_GPtrArray*, FwupdBackend::checkForUpdates()::$_6>::runFunctor()() at /home/apol/devel/kde5/include/QtConcurrent/qtconcurrentstoredfunctioncall.h:60
#8 QtConcurrent::RunFunctionTask<_GPtrArray*>::run() at /home/apol/devel/kde5/include/QtConcurrent/qtconcurrentrunbase.h:108
#9 QThreadPoolThread::run() at /home/apol/devel/frameworks/qt5/qtbase/src/corelib/thread/qthreadpool.cpp:100
#10 QThreadPrivate::start() at /home/apol/devel/frameworks/qt5/qtbase/src/corelib/thread/qthread_unix.cpp:329
#11 start_thread() at /usr/lib/libpthread.so.0
#12 clone() at /usr/lib/libc.so.6
```
https://bugs.kde.org/show_bug.cgi?id=419968

**Steps to Reproduce**
It's easy to reproduce using Discover, we could probably come up with a test app.

**Expected behavior**
Either getting the devices (which we used to get, so it's likely a regression) or an error message

**fwupd version information**
1.5.1
I also tried running from master, I get the same issue. Otherwise it was pacman (ArchLinux).

I'll be happy to provide any additional information required, although it seems like a problem with the library (?).

superm1, 2020-11-18:
Does the daemon crash mid thread perhaps?

It might be useful to run the daemon in verbose mode and look for logging at same time.

aleixpol, 2020-11-18:
The daemon does not crash.
I ported away from the sync calls and now it seems to work again:
https://invent.kde.org/plasma/discover/-/merge_requests/53

superm1, 2020-11-18:
On your system can you try to run `./contrib/firmware_packager/simple_client.py get-devices` and see if it gets stuck too for you?  It's gobject introspection, but those should be using the same sync methods.  So hopefully the same repro happens.  It doesn't for me on master.  If it also doesn't happen for you, a simple reproducer would be appreciated.

superm1, 2020-11-18:
And since you're multithreaded, is there multiple glib main loops running?  or multiple threads potentially using `g_dbus`?  Maybe the reproducer won't happen in a simple single threaded solution.

aleixpol, 2020-11-18:
> On your system can you try to run `./contrib/firmware_packager/simple_client.py get-devices` and see if it gets stuck too for you? It's gobject introspection, but those should be using the same sync [...]

Yes, this works fine.

> And since you're multithreaded, is there multiple glib main loops running? or multiple threads potentially using `g_dbus`? Maybe the reproducer won't happen in a simple single threaded solution.

It definitely is because of using threads. If I move it out of the separate threads it works, but then it also blocks the whole startup of the software center.

Porting to _async APIs solves the problem, although that doesn't change how something regressed and rendered our software dysfunctional. Also it doesn't work for as is as it requires bumping the required version of fwupd.

I'll try to put together a reproducer...

aleixpol, 2020-11-18:
Here's a fairly simple test case:
https://github.com/aleixpol/fwupd-block

superm1, 2020-11-19:
Thanks for the reproducer.  I'm a little naïve when it comes to compiling Qt apps.  Can you help me on Ubuntu what I'm missing?

I installed `qtbase5-dev`, `libqt5concurrent5` and `qtdeclarative5-dev`:

```
$ cmake .
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fwupd>=1.0.6'
--   Found fwupd, version 1.5.2
-- Configuring done
CMake Error at CMakeLists.txt:9 (add_executable):
  Target "fwupd-block" links to target "Qt::Core" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:9 (add_executable):
  Target "fwupd-block" links to target "Qt::Concurrent" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

superm1, 2020-11-19:
I got it to compile using this (which hopefully I didn't introduce a _different_ problem)
```
g++ -o fwupd_block fwupd-block.cpp `pkg-config fwupd --libs --cflags` `pkg-config Qt5Core --libs --cflags` `pkg-config Qt5Concurrent --libs --cflags` -fPIC
```

I can reproduce it now.

superm1, 2020-11-19:
It seems that specifically `g_bus_get (G_BUS_TYPE_SYSTEM....` never actually calls the async callback `fwupd_client_connect_get_bus_cb`.  Unless @hughsie knows this is getting into the realm it might be better to ask on Qt/Glib mailing lists.

superm1, 2020-11-19:
As for the "how" in this question:

> although that doesn't change how something regressed and rendered our software dysfunctional

The sync methods are basically wrappers of the async methods now.  You can look at 7c8a83065924a7dff86c128b0b6625c0ab205cc6 to see what I mean.

closed 2020-11-19 by hughsie

aleksander0m, 2020-12-03:
That last update makes my fwupdmgr test work (I can see progress/state updates) and it also doesn't block @aleixpol 's tester any more, I think it's looking good :D