fwupd/fwupd #2602: vli: power cycling request

memily, 2020-11-19:
**Is your feature request related to a problem? Please describe.**
Context: Our devices vary on property "power cycling is required" after updating fw. This leads to complexity such as “GPIOB reset” and “chip reset”. 

Future Issue: As we continue to add onto the list of supported VLI devices, we’re afraid the scaling of complexity will end up having diminishing returns. 

Proposal: My manager believes a “one rule” solution across all devices is best and fulfills the request of the ODMs. A rule that requires the end user to manually power cycle (physically unplug and replug). With no changes to current implementations constructed thus far.

**Describe the solution you'd like**
Code fix: A succinct print statement along the lines of “please unplug and replug your device” at the very end of the update flow for all VLI devices. We would like to run this fix by you before attempting to implement it.

hughsie, 2020-11-19:
I personally think it's much better to handle this in the code, rather than get thousands of users to do something time critical that they could get wrong. We have to consider the user experience as a first class citizen I think.

superm1, 2020-11-19:
We have something like this in logitech for unsigned device. Can it be applied for vli?

superm1, 2020-11-19:
Here's what we do specifically in Logitech:
https://github.com/fwupd/fwupd/blob/master/plugins/logitech-hidpp/fu-logitech-hidpp-peripheral.c#L655

memily, 2020-11-20:
>I personally think it's much better to handle this in the code.

We understand, but for some devices, power cycling is already the only possible hardware option to complete the update process. Because power cycling works across the board for all our devices; we want to be considerate of the overhead and maintenance that comes from adding more than the ten, unique, vli reset options to fwupd. For all of us.

>Here's what we do specifically in Logitech:

Yes! This will work for vli devices.

hughsie, 2020-11-20:
> but for some devices, power cycling is already the only possible hardware option to complete the update process

Understood, and for that we have some code in gnome-software that handles this, although it's only available on very new versions of fwupd and gnome-software that 95% of people are not going to be using yet.

> more than the ten, unique, vli reset options to fwupd

Ten? How ten? I count three... I don't think new hardware (especially composite dock devices) are going to be possible to add without writing a .quirk file with the device topology -- unless the device starts telling us the topology itself somehow. If we're writing a quirk file to tell fwupd the logical and electrical topology then it's a tiny bit of extra work to tell fwupd how the composite device should be reset.

joakim-tjernlund, 2020-11-20:
I understand that it is tempting to add a manual step for the users here but consider there are
thousands  of people updating their FW over years to come.

You should also consider unattended upgrades initiated by say IT admins, they can't do manual steps.

superm1, 2020-11-20:
> Understood, and for that we have some code in gnome-software that handles this, although it's only available on very new versions of fwupd and gnome-software that 95% of people are not going to be using yet.

To me this sounds like the best solution is to set the newer versions that can orchestrate better as minimum requirements on the firmware to update.

>You should also consider unattended upgrades initiated by say IT admins, they can't do manual steps.

Great point, completely agree.

memily, 2020-11-23:
>Ten? How ten? I count three...

When we first started coworking together, we prioritized slowly adding complexity to your workload so you do not become overwhelmed. E.g. single hub and single pd, to two hubs, parade, and share spi. That is only the tip of the iceberg. We really do offer many, many options to meet the needs of our clients... As such, our two main points of concern are:
1.	Our other reset options require more rigorous, algorithmic coding than adding more quirk files. 
2.	An ongoing issue with helping you test devices: our clients do not send us the finished product. We understand and try to work around possible discrepancies between the demo board we have and the end product you receive. Some things are out of our control. We may file the paperwork for you and for us concurrently; howsomever, we often end up waiting several months for a spare to be sent to us.

This is why we’d like to raise the flag of alarm and bring the issue of future overhead to your attention, as more devices get added gradually. 

>To me this sounds like the best solution is to set the newer versions that can orchestrate better as minimum requirements on the firmware to update.

Apologies, not everyone on our team understands what this sentence means. Can you give us an example; is this a metainfo file change? English is the second/third learned language in our country. Thank you. 

>Great point, completely agree.

We fulfill the requests of our ODMs. Some ODMs require, in their current and future spec, power cycling only. Regardless of available hardware options. We ask for your consultation.

hughsie, 2020-11-23:
> single hub and single pd, to two hubs, parade, and share spi

My mental health appreciated that :)

> Our other reset options require more rigorous, algorithmic coding than adding more quirk files

Have you got some examples? Maybe we can come up with something better than quirk files? Using quirk files to define a topology isn't elegant or beautiful, and I'd be certainly okay with a meta-format that generated those files based on something higher level. If you can share more details I'd be happy to build something more beautiful.

> our clients do not send us the finished product

Yes, I've noticed that too; the demo board is wired in a subtly different way or requires something like GPIOB reset. It's frustrating for me too.

> bring the issue of future overhead to your attention

I think we need to go back to the people adding support and actually make a plan; rather than "_add LVFS support_" added as a post-design pre-ship thing I want it to be part of the design phase. If we could for instance add USB descriptor metadata to the device itself (for instance, saying "_reset me using the device with VID:PID and GPIOB_" then we don't need quirk files and the device can self-describe.

> English is the second/third learned language in our country

My apologies; I struggle with English and I was born here. I think what Mario was saying was that we should make this "just work" and actually design a way of the device to do the right thing. If we could query the USB HUB using a VLI-specific protocol and just ask it "_are you sharing your SPI?_" and then "_what kind of device are you sharing your SPI with_" or even "_what other i2c devices are attached to you_" then we could make all this very much simpler. The idea being that we have an API that we can talk to the VLI chips with, and from that the fwupd plugin doesn't need all this manual work. If you want to add a i2c bridge on a chip for one ODM then you can just build the PCB and set an unused block of memory in the SPI that describes how it's set up. No fwupd changes required. This also means we can ship new VLI hardware that doesn't need such a new fwupd version! At the moment quick files are shipped with fwupd in the release and so users need to update to a very new fwupd just for the metadata that says "this VLI chip shares SPI with this other VLI chip".

> Some ODMs require, in their current and future spec, power cycling only. Regardless of available hardware options

I think that's a shame. If we ask tens of thousands, or even hundreds of thousands of users to do something then it's worth spending the time in the design phase to make it easy for them. If we add 10 seconds to the process for 100,000 users then we're wasting nearly 300 **hours** of user time on something we could have fixed in less than one.

The other problem I highlighted earlier: most people are using old versions of fwupd and GNOME Software. This means that if we say "_this firmware requires a client feature_" e.g. https://lvfs.readthedocs.io/en/latest/metainfo.html#client-features then that restricts the number of people who can apply the update down from most to less than 1%. Of course, as time goes on more and more users will have new fwupd and gnome-software versions and so this problem gets less, although the point is still valid.

So to sum up: I think if we **can** do this automatically and magically just work then we **should** -- and I can help you with this. If we can't then we just use the client feature, depend on a new fwupd and draw some more line art.

memily, 2020-11-27:
>Have you got some examples?... I'd be happy to build something more beautiful.

i'd like you to wrap my manager in a soft blanket, pass them a cup of hot tea, and pat them on the head reassuringly, all while saying that the horrors of xkcd1319 won't happen first! In addition, did we miss any clang style/linter news? Getting examples is a long process so we have nothing definite to give you currently. 

>It's frustrating for me too.

We sympathize and share in the problem. 

>I want it to be part of the design phase.

Do you have a specific USB descriptor in mind? Such as: add to USB device descriptor (18 bits), or randomly find an USB descriptor (string, config, interface, endpoint, etc) with a free bit?  

>At the moment quirk files are shipped with fwupd in the release and so users need to update to a very new fwupd...

Makes sense. We've gotten a lot of flack over the years for asking for English clarification! Can only do your best to waddle along. While it is not possible to change fw protocol on already, mass production devices (backwards compatibility); it might be possible on new models in the future.

>So to sum up: I think if we can do this...

We will follow your lead.

superm1, 2020-11-28:
As a suggestion even with old designs, if you can't change the descriptor you can potentially encode this information of what to do with resets in a byte somewhere in NVRAM.  I don't think it really matters which approach you take for fwupd.  Both parsing the descriptor or sending a random command to get back that byte and decode it can be done when the device is setup.

But this type of design will also simplify your tools for other OSes too.

hughsie, 2020-12-04:
> Have you got some examples

I meant more of the different ways we can reset the composite hardware.

> Do you have a specific USB descriptor in mind

Well, we can read the descriptor, or we could just read out a block of flash at a specific offset (specified in FuVliUsbhubHeader) that does something like:

    BYTE flags, where bit0 = needs-unlock-legacy813, bit1=SPI-shared-with-PD, bit2=has-i2c, bit4=is-usb3 bit4=is-usb2
    BYTE shared_with, where 0=parade, 1=PD, 2=something-else
    BYTE i2c_for, where 0 = MSP430, 1 = RTD21xx

So in three bytes of flash space you've removed the need for most of the quirk entries. I'm not suggesting 3 bytes is enough for something you can extend into the future with new devices, but `3*uint32_t`s gives you a lot of space for future devices.

joakim-tjernlund, 2021-01-01:
Any progress on this ?

closed 2021-08-03 by hughsie