fwupd/fwupd #2603: Problem with firmware update for the dock

mueihn, 2020-11-19:
**Describe the bug**
A clear and concise description of what the bug is.

The firmware updates fails with an error messages.

**Steps to Reproduce**
Steps to reproduce the behavior.

mue@schneckenhaus:~> sudo fwupdmgr update
[sudo] Passwort für mue: 
No upgrades for Latitude 7390 Thunderbolt Controller, current is 41.00: 23.00=older
Downloading 43.00 for Thunderbolt controller in Dell dock...
Entpacken …              [***************************************]
Authentifizierung …      [***************************************]
device ID 7b3cbc344094ace839263dd4711c84ea99bb6dc3 was not found
mue@schneckenhaus:~>

**Expected behavior**
A clear and concise description of what you expected to happen.

There are firmware updates available. They should have been installed.

**fwupd version information**
Please provide the version of the daemon and client.

mue@schneckenhaus:~> fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
mue@schneckenhaus:~>

It came pre-installed with my opensuse leap 15.2 distribution if I am not mistaken. Thus, it is likely part of the distribution-native package management controlled by zypper.

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.

mue@schneckenhaus:~> fwupdmgr get-devices --filter=~internal
Verarbeitung der Argumente schlug fehl: Unbekannte Option --filter=~internal
mue@schneckenhaus:~> fwupdmgr get-devices
Latitude 7390 Thunderbolt Controller
  DeviceId:             fbe0d0a92b5921e5110308659a2b2a22c625fd4b
  Guid:                 0c4c1146-581e-5a06-9efb-ecd847f88c82
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              41.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2020-11-19

Thunderbolt controller in Dell dock
  DeviceId:             41c056e49eb16aabbb076701ba290418697f0b66
  ParentDeviceId:       a455a3a21752e5d31c228f1f5b51761dad88a432
  Guid:                 c94770ca-1773-592c-b20a-e87243bc7cd0
  Summary:              Thunderbolt controller
  Plugin:               thunderbolt
  Flags:                updatable|require-ac|supported|registered
  Vendor:               Dell Inc
  VendorId:             TBT:0x00D4
  Version:              40.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  InstallDuration:      22
  Created:              2020-11-19

RTS5487 in Dell dock
[... 219 lines cut ...]
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0013 XbSilo               loading silo with file contents
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0013 XbSilo               0,02ms      parse blob
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0014 XbSilo               0,93ms      query on / with `components/component` limit=0 -> 500 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0014 FuEngine             500 components now in silo
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0016 XbSilo               1,01ms      query on / with `components/component/provides/firmware` limit=0 -> 587 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0016 XbSilo               0,54ms      query on / with `components/component/provides/firmware` limit=0 -> 587 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0019 XbSilo               1,48ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='2800f812-b7b4-2d4b-aca8-46e0ff65814c']/../..` limit=1 -> 0 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0019 XbSilo               0,17ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='0c4c1146-581e-5a06-9efb-ecd847f88c82']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0019 XbSilo               0,66ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='c94770ca-1773-592c-b20a-e87243bc7cd0']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0020 XbSilo               0,51ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='a86a3f07-b7d3-4545-9238-0274fc1ba682']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0021 XbSilo               1,40ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='20c968e3-b6e1-53f8-a4b0-f45577c5b1c5']/../..` limit=1 -> 0 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0026 XbSilo               4,52ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='707c63d2-e597-5c40-84db-9b1bb4c48d96']/../..|components/component/provides/firmware[@type='flashed'][text()='acfcd89b-105d-55b9-b85b-08bf8508f38c']/../..|components/component/provides/firmware[@type='flashed'][text()='568ffa1e-a0db-5287-9ea3-872b60f7730b']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0027 XbSilo               0,73ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='cd357cf1-40b2-5d87-b8df-bb2dd82774aa']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0028 XbSilo               0,84ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='8ceeeffd-51b6-580c-9b75-69143227aff8']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0029 XbSilo               0,94ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='89fec0b6-6b76-5008-b82c-5e5c6c164007']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0034 XbSilo               5,51ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4']/../..|components/component/provides/firmware[@type='flashed'][text()='b27d25f1-019d-5718-b41a-02ddaefe5577']/../..|components/component/provides/firmware[@type='flashed'][text()='ac5b774c-b49d-566b-9255-85f0f7f8a4ed']/../..` limit=1 -> 1 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0038 XbSilo               3,32ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='c6cd9ab0-8f20-512e-9e1f-1af55b8454b9']/../..|components/component/provides/firmware[@type='flashed'][text()='82741c78-f5dc-5c23-a152-00de5799edc8']/../..|components/component/provides/firmware[@type='flashed'][text()='7c0505be-b609-566e-bf94-561e441a9171']/../..` limit=1 -> 0 results
Nov 19 11:19:55 schneckenhaus fwupd[3971]: 10:19:55:0039 XbSilo               1,47ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='fed2efa4-7045-55a1-b4fc-b29283d59fe5']/../..|components/component/provides/firmware[@type='flashed'][text()='8b72a10c-1279-5f8e-a28a-34fb11a58240']/../..` limit=1 -> 0 results
mue@schneckenhaus:~>

**Additional questions**
- Operating system and version: opensuse leap 15.2
- Have you tried unplugging the dock or any peripherals from your machine? I unplugged the dock. Then I retried the update. There were no updates possible then as the only one left was the one for the dock. I replugged the dock and tried it again. Then the update worked.
- Have you tried to power cycle the dock from the AC adapter? No.
- Is this a regression? I do not understand the question.

mueihn, 2020-11-19:
I was able to solve it as described in the ticket.

closed 2020-11-19 by mueihn

superm1, 2020-11-19:
Thanks for sharing.  Some of the architecture of how the update works has changed in modern fwupd versions, so I expect it's less problematic there.  Glad you sorted it after a few tries though on the older version.